wlcrs/huawei_solar #856: [Bug]:  Abnormal value for "Battery Total charge"

JensenNick, 2024-11-24:
### Describe the issue

I'm getting an very abnormal initial reading of the value for ´sensor.batteries_total_charge´ . The reading currently id about 463 MWh which of course isn't correct. I also se a smal deviation of the reading for ´sensor.batteries_total_discharge´. Here are the readings I get in the Huawei Solar integration and the values I have in the FusionSolar app.
[image]

The issue is also described in this discussion https://github.com/wlcrs/huawei_solar/discussions/628
 
I have created e new instance of Home Assistant with a new install og the Huawei Solar integration. In order to get the initial value included in the debug log I have proceeded with these steps.
1. Installed the Huawei Solar integration
2. Enabled debug logging 
3. Reconfigured the integration
4. Downloaded the debug log named "…145Z.log"

In order to "track" the reading of the modbus data from the battery I have logged the discharged and charged of the battery.
6. Set up TOU to discharge from 13:32 until 13:42 and to charge from 13:45 to 13:50.
7. Enabled debug logging
8. Downloaded the debug log named "…270Z.log"

### Describe your Huawei Solar Setup

Inverter #1 Type: Huawei SUN2000 10KTL-M1
Inverter #1 Firmware version: V100R001C00SPC165
Inverter #2 Type: Huawei SUN2000 3KTL-L1
Inverter #2 Firmware version: V200R001C00SPC142
sDongle present: Yes
sDongle Type: sDongleA-05 (WiFi / Ethernet)
sDongle Connectivitiy: Ethernet
sDongle Firmware: V200R022C10SPC121
Power meter present: three phase
Optimizers Present: No
Battery: LUNA2000-S1 10kWh
Battery Firmware version: V100R002C00SPC624 
Huawei Solar integration version: 1.5.0a1

### How do you connect to the inverter?

Please select your connection method

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
[config_entry-huawei_solar-01JDF1SV35W5S3HFZ4PFK720ZP.json](https://github.com/user-attachments/files/17893403/config_entry-huawei_solar-01JDF1SV35W5S3HFZ4PFK720ZP.json)

[home-assistant_huawei_solar_2024-11-24T12-26-27.145Z.log](https://github.com/user-attachments/files/17893466/home-assistant_huawei_solar_2024-11-24T12-26-27.145Z.log)

[home-assistant_huawei_solar_2024-11-24T12-57-02.270Z.log](https://github.com/user-attachments/files/17893467/home-assistant_huawei_solar_2024-11-24T12-57-02.270Z.log)

### Upload your relevant debug logs

```text
Getting an error trying to copy/past the log here and I don't have an option to attach a file.
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2024-11-25:
The total charge is stored in registers 37780 and 37781. We retrieve this by reading 28 registers, starting from register 37760.

2024-11-24 13:25:13.783 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760 with length 28 from slave 1
2024-11-24 13:25:13.784 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x61 0x0 0x0 0x0 0x6 0x1 0x3 0x93 0x80 0x0 0x1c
2024-11-24 13:25:13.791 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x61 0x0 0x0 0x0 0x3b 0x1 0x3 0x38 0x3 0xca 0x0 0x0 0x0 0x2 0x1b 0x16 0x0 0x0 0xff 0xff 0xff 0xee 0x0 0x0 0x27 0x10 0x0 0x0 0x27 0x10 0x0 0x0 0x27 0x10 0x0 0x0 0x27 0x10 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 **0x2 0xc2 0xf9 0x34** 0x0 0x8 0x89 0x65 0x0 0x0 0x2 0xa4 0x0 0x0 0x0 0x1e old_data:  addr=None

| Register | Hex value |
| -------- | --------- |
| 37760    | 0x3 0xca  |
| 37761    | 0x0 0x0   |
| 37762    | 0x0 0x2   |
| 37763    | 0x1b 0x16 |
| 37764    | 0x0 0x0   |
| 37765    | 0xff 0xff |
| 37766    | 0xff 0xee |
| 37767    | 0x0 0x0   |
| 37768    | 0x27 0x10 |
| 37769    | 0x0 0x0   |
| 37770    | 0x27 0x10 |
| 37771    | 0x0 0x0   |
| 37772    | 0x27 0x10 |
| 37773    | 0x0 0x0   |
| 37774    | 0x27 0x10 |
| 37775    | 0x0 0x0   |
| 37776    | 0x0 0x0   |
| 37777    | 0x0 0x0   |
| 37778    | 0x0 0x0   |
| 37779    | 0x0 0x0   |
| 37780    | **0x2 0xc2**  |
| 37781    | **0xf9 0x34** |
| 37782    | 0x0 0x8   |
| 37783    | 0x89 0x65 |
| 37784    | 0x0 0x0   |
| 37785    | 0x2 0xa4  |
| 37786    | 0x0 0x0   |
| 37787    | 0x0 0x1e  |

0x02c2f934 = 46332212

The total discharge can be read from the two registers following that:
0x00088965 = 559461

My conclusion is that the integration is working as intended, as it correctly decodes the values reported by the inverter.

closed 2024-11-29 by wlcrs

torwag, 2024-11-29:
Hey @wlcrs does closing mean it's fixed, or does closing mean, there is nothing we can do on this side?

Just want to make sure before spending time debugging this.

wlcrs, 2024-11-29:
I've gone down to actual bits being sent via Modbus to investigate this issue. This integration correctly requests the relevant register and correctly decodes the values being sent back by the inverter. What can/should I fix according to you?

torwag, 2024-11-29:
Than it might simply be the cosmic ray, which flips that one bit....
nah all good, if you did this already, there is no reason to see if that can be fixed. I just wanted to make sure, not blaming anyone.

It's a strange bug, as Fusionsolar shows the right value but you can prove that you exactly read in the registers as it is documented.

JensenNick, 2024-11-30:
@wlcrs I have now received an answer from Huawei Support. According to Huawei Support the "correct" register to use would be 37066 and 37068. Below the full reply from Huawei Support. Could this be implemented in the integration?

[image]

There has not been any other battery connected, so their assumption is incorrect and again it is impossible that any battery would charge the abnormal amount of kWh which is registered.

JensenNick, 2024-12-03:
@wlcrs I'd appreciate your comment to the solution suggested by Huawei Support. They wan't to close the issue. I've asked them to wait for my (your) feedback.