wlcrs/huawei_solar #861: [Bug]: Unable to import component: No module named 'huawei_solar' with Home Assistant 2024.12.0

virtualzone, 2024-12-05:
### Describe the issue

Thanks for this great integration! I've been using it for about a year now and it has been working very well.

However, since upgrading Home Assistant to 2024.12.0, the integration (installed as a custom component) is not loaded anymore. Home Assistant reports:

```
ModuleNotFoundError: No module named 'huawei_solar'
```

### Describe your Huawei Solar Setup

Inverter Type: 
Inverter Firmware version: 
sDongle present: Yes / No
sDongle Type: sDongleA-05 (WiFi / Ethernet) / SDongleA-03 (4G)
sDongle Connectivitiy: WiFi / Ethernet / 4G
sDongle Firmware: 
Power meter present: three phase / single phase / no
Optimizers Present: Yes / No
Battery: LUNA2000-SO xxkWh / LUNA2000-S1 xxkWh (Released 2024) / LG RESU xxkWh / None
Battery Firmware version: 
Huawei Solar integration version: 

### How do you connect to the inverter?

Via the SDongle, wired connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
2024-12-05 06:27:24.480 ERROR (MainThread) [homeassistant.setup] Setup failed for custom integration 'huawei_solar': Unable to import component: No module named 'huawei_solar'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1034, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1014, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1074, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1022, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/huawei_solar/__init__.py", line 20, in <module>
    from huawei_solar import (
    ...<4 lines>...
    )
ModuleNotFoundError: No module named 'huawei_solar'
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

dragosmd, 2024-12-05:
Upgrade to 1.5.0a1

virtualzone, 2024-12-05:
Same issue with 1.5.0a1

virtualzone, 2024-12-05:
Something seems to be broken regarding the installation of Python packages for custom integrations in 2024.12.0, see: 
https://github.com/home-assistant/core/issues/127966

The following workaround works for me:
I'm running Home Assistant in a Docker container. Manually setting the container's environment variable ```PYTHONPATH``` to ```/config/deps``` resolves the issue.

wlcrs, 2024-12-05:
Thank you for referencing the issue in HA core.

Closing as not an integration bug.

closed 2024-12-05 by wlcrs

exetico, 2024-12-07:
For other people visiting this: Upgrade to 2024.12.1, and you're good to go :-)