wlcrs/huawei_solar #867: [Bug]: Abnormal value for "Battery Total charge"

JensenNick, 2024-12-10:
### Describe the issue

With reference to the issue described in https://github.com/wlcrs/huawei_solar/issues/856 Huawei Support has reported back and suggested a solution. According to Huawei Support the correct modbus registers to use is 37066 and 37068 (se the answer from Huawei Support below). I'd appreciate a reply / comment to the solution suggested by Huawei Support. Thanks!

Reply from Huawei:
[image]

 

### Describe your Huawei Solar Setup

Inverter Type (master): Huawei SUN2000 10KTL-M1
Inverter 1 (master) Firmware version: V100R001C00SPC165
Inverter Type (slave): Huawei SUN2000 3KTL-L1
Inverter (slave) Firmware version: V200R001C00SPC142
sDongle present: Yes
sDongle Type: sDongleA-05 (WiFi / Ethernet)
sDongle Connectivitiy: Ethernet
sDongle Firmware: V200R022C10SPC121
Power meter present: three phase
Optimizers Present: No
Battery: LUNA2000-S1 10kWh
Battery Firmware version: V100R002C00SPC624
Huawei Solar integration version: 1.5.0a1

### How do you connect to the inverter?

Via the SDongle, wired connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
Diagnostics file included in #856.

### Upload your relevant debug logs

```text
Debug logs included in #856.
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

Roving-Ronin, 2025-01-06:
@JensenNick Hi Nick,

As Thijs has commented the intergration uses these registers, which as per the name is 'Storage' total Dis/Charge, i.e. Storage refers to the total for ALL connected batteries.

rn.STORAGE\_TOTAL\_CHARGE: U32Register("kWh", 100, 37780),  
rn.STORAGE\_TOTAL\_DISCHARGE: U32Register("kWh", 100, 37782),

and these that Huawei is point to as a cause. These are the registers for the individual battery storage unit/tower number 1.

rn.STORAGE\_UNIT\_1\_TOTAL\_CHARGE: U32Register("kWh", 100, 37066),  
rn.STORAGE\_UNIT\_1\_TOTAL\_DISCHARGE: U32Register("kWh", 100, 37068),

This would appear that support has dodged the issue and tried to point you to using the registers that are specific to the individual battery stacks (i.e. each 5/10/15kW "tower"), which in fact the integration alrady does. IMHO I'd be flagging with Huawei that issue appears to he hardware/softare related upon their side of things and getting them solve their issue.

---

BTW here a site running 1.5.1a1 of 2 x 5kW L1's with a LUNA S0 15kWh stack attached to Inverter 1 (approx 4-5 years old).

The BATTERIES sensors:

[image]

BATTERY 1 sensors:

[image]

Both appear fine.

JensenNick, 2025-01-14:
@Roving-Ronin Thank you. Your help is really appreciated.