wlcrs/huawei_solar #870: [Bug]: The connection only works once after a restart

psi33, 2024-12-25:
### Describe the issue

First of all, many thanks for this ingenious integration. Unfortunately, it only works once for me and is not stable. After restarting the integration, all data is correctly retrieved from the inverter via the network. 

If I try to connect again, the values are no longer updated. This is clearly visible in the log.

A second home server also accesses the same interface via the network. This shows that the connection to the Modbus interface is working.

For a test, I ended this second query so that the Home Assistant has unrestricted sole access. Unfortunately, this did not result in any improvement and the values are only read in once after a restart. Why does the integration crash? 

Thank you for your help

### Describe your Huawei Solar Setup

Inverter Type: SUN2000-10KTL-M1
Inverter Firmware version: V100R001C00SPC162
sDongle present: Yes
sDongle Type: sDongleA-05 (WiFi / Ethernet)
sDongle Connectivitiy: Ethernet
sDongle Firmware: V100R001C00SPC335
Power meter present: three phase
Optimizers Present: No
Battery:  None
Huawei Solar integration version: None

### How do you connect to the inverter?

Via the SDongle, wired connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
[config_entry-huawei_solar-01JFN1QA6R2EV2RNH153ECHFY9.json](https://github.com/user-attachments/files/18246641/config_entry-huawei_solar-01JFN1QA6R2EV2RNH153ECHFY9.json)
[Log.txt](https://github.com/user-attachments/files/18246644/Log.txt)

### Upload your relevant debug logs

```text
[Log.txt](https://github.com/user-attachments/files/18246644/Log.txt)
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

NBOnGithub, 2024-12-28:
I'm seeing the same behaviour for the past week. Any restart of home assistant gives me a single snapshot value for the battery status and inferior power, etc. and nothing further. Command actions still seem to take effect as I can manage to change the charge from grid settings via automations. 
Previous to this week the integration has been working flawlessly. 
I've tried resetting the solar system by turning off the power to it but this has not fixed it either

Mav70, 2025-01-01:
@psi33 I'm also seeing the same issue for the past couple of days. Any update on a resolution?

NBOnGithub, 2025-01-01:
I eventually removed the integration and also removed the download from hacs. Then I rebooted and reinstalled and it started working again. 
It might be getting something corrupt on disk that it is then failing to reload properly.
Note that I did need to do a second reboot to fix the automation warnings.

psi33, 2025-01-02:
Thx @NBOnGithub 
I have deleted the integration and also deleted it in HACS. Then restarted HA. Added the integration (1.5.0a1) again. It has improved and works for about 10 queries, but even after that the integration crashes. Does anyone have an idea where this could be

Mav70, 2025-01-02:
@psi33 @NBOnGithub I've also tried to delete and reinstall but with the same results as before. What I have notcied though is that sometimes the inverter isn't responding to pings (I connect via wifi) although I can access it fine through the app.  The wifi connection at the inverter side seems to go into a "sleep" mode.

psi33, 2025-01-03:
@NBOnGithub und @Mav70 
I connect via a fixed network cable. I have just tested it again. After restarting the integration (1.5.0a1), I get 20 measuring points from the inverter. Then the integration crashes without a log entry and does not send any further data. In all this time I have not lost a single ping. So I think we can rule out the inverter and the network connection. What else can we test or adjust? Does anyone have an idea?

psi33, 2025-01-06:
Unfortunately, the new HA version 2025.1 does not bring any improvement. I think only the developer can help us here. I hope for feedback

fehlerteufel, 2025-01-10:
I have a similar issue (worked fine before updating HA and the integration yesterday). Data from the inverter is updated normally, but data from the additionally connected power meter is only updated once after restarting the integration.

WILHER, 2025-01-20:
I have the same Problem. After reboot ist works für a couple of seconds than it crash

fehlerteufel, 2025-01-20:
Reverting the Huawei Solar Integration back to version 1.3.3 solved the problem for now. I'm happy to help in resolving the problem with version 1.4.1, please ping me if I can support in some way.

psi33, 2025-01-20:
Unfortunately, installing version 1.3.3 does not help me. The data is updated for 5 minutes and then the interface crashes.
I also hope for a solution soon and will gladly support you in testing and reporting.