wlcrs/huawei_solar #871: [Bug]: Unable to connect to Inverter SUN2000-10K-MAP0

robgalea, 2024-12-29:
### Describe the issue

Hi,

I have been previously been using this integration with inverter SUN2000-10K-M1, infact with the help of this integration I found out that the installer scammed me by supplying the wrong inverter type for the panels I have.

I have since been given the SUN2000-10K-MAP0 however the integration is not connecting to the inverter. I have tried both ports 6607 and 502 as well as both User IDs

Is this expected? This would be ideal to support as to my knowledge this is the new model to support higher amperage panels

### Describe your Huawei Solar Setup

Inverter Type: SUN2000-10K-MAP0
Inverter Firmware version: V200R024C00SPC100
sDongle present: Yes
sDongle Type: sDongleA-05 (WiFi / Ethernet)
sDongle Connectivitiy: Ethernet
sDongle Firmware: V100R001C00SPC335
Power meter present: three phase
Optimizers Present:No
Battery: None
Huawei Solar integration version: 1.4.1 but also tried version1.5.0a1

### How do you connect to the inverter?

Please select your connection method

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
2024-12-29 17:35:56.751 INFO (MainThread) [huawei_solar.huawei_solar] Waiting for connection ...
2024-12-29 17:36:01.752 ERROR (MainThread) [huawei_solar.huawei_solar] Timeout while waiting for connection. Reconnecting...
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 157, in _communication_lock
2024-12-29 17:36:01.752 ERROR (MainThread) [huawei_solar.huawei_solar] Aborting client creation due to error.
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 199, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 118, in _initialize
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 271, in get
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 308, in get_multiple
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 444, in _read_registers
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 157, in _communication_lock
2024-12-29 17:36:05.258 INFO (MainThread) [huawei_solar.huawei_solar] Waiting for connection ...
2024-12-29 17:36:10.259 ERROR (MainThread) [huawei_solar.huawei_solar] Timeout while waiting for connection. Reconnecting...
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 157, in _communication_lock
2024-12-29 17:36:10.260 ERROR (MainThread) [huawei_solar.huawei_solar] Aborting client creation due to error.
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 199, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 118, in _initialize
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 271, in get
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 308, in get_multiple
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 444, in _read_registers
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 157, in _communication_lock
2024-12-29 17:36:14.482 DEBUG (MainThread) [huawei_solar.modbus] Waiting for 1500 milliseconds after connection before performing operations
2024-12-29 17:36:14.483 INFO (MainThread) [huawei_solar.huawei_solar] Waiting for connection ...
2024-12-29 17:36:15.983 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 43006 with length 1 from slave 0
2024-12-29 17:36:55.989 ERROR (MainThread) [huawei_solar.huawei_solar] Aborting client creation due to error.
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 199, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 118, in _initialize
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 271, in get
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 308, in get_multiple
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 451, in _read_registers
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 399, in _do_read
2024-12-29 17:36:58.959 DEBUG (MainThread) [huawei_solar.modbus] Waiting for 1500 milliseconds after connection before performing operations
2024-12-29 17:36:58.959 INFO (MainThread) [huawei_solar.huawei_solar] Waiting for connection ...
2024-12-29 17:37:00.460 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 43006 with length 1 from slave 1
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

Roving-Ronin, 2025-01-06:
@robgalea Update your sDongle to firmware: V200R022C10SPC110

https://github.com/wlcrs/huawei_solar/wiki/Confirmed-firmware-versions#three-phase---sun2000-map0-hybrid

robgalea, 2025-01-21:
Hi, @Roving-Ronin  tried to do this however V200R022C10SPC110 is not available to download, i instead installed V200R022C10SPC121 but still did not manage to work. The integration specifically works with V200R022C10SPC110 only?

Roving-Ronin, 2025-01-21:
It isn't "tied" to using SPC110 it was just that SPC110 with a MAP0 was the version that other people have working with their similiary setup and have confirmed on the Wiki that it works.  SPC121 should also work, only time I've seen people give a firmware a thumbs down, is not because it failed to work at all, but more likely its connection dropped in and out a few times a day.

I assume you don't have anything else connecting to the inverters using modbus (and app or even the EV charger, that uses the single modbus connection the dongle supports), and that you have enabled Modbus access (unrestricted is simplest and not to a specific IP) as per the setup instructions?

robgalea, 2025-01-21:
I see, from the logs I can see the errors have changed, now it reports the following: Read exception while connecting via ModbusTCP

To my knowledge I am not using modbus for anything else. Currently I only have the inverter and dongle, no EV Charger and the battery system should be arriving next week. Regrading apps, I am only using the phone Fusion Solar app and the Fusion Solar App on home assistant (temporarily) using the Kiosk feature. 

The Modbus access is indeed set to unrestricted and the inverter has a static IP configured

Roving-Ronin, 2025-01-21:
Have you updated the Modbus firmware also v330 or v335 iirc are available. You update on the SolarFusion portal, same as like updating dongle / inverter, just choose the MBUS option.

Of that doesn't work I'd be swapping the dongle for another and/or flagging with Huawei support to fix.