wlcrs/huawei_solar #873: [Bug]: Setup failed for custom integration 'huawei_solar': Unable to import component: No module named 'pymodbus.message'

screamer3003, 2025-01-03:
### Describe the issue

After the Update to 2025.1.0 I Get this error

Logger: homeassistant.setup
Quelle: setup.py:334
Erstmals aufgetreten: 21:28:29 (1 Vorkommnisse)
Zuletzt protokolliert: 21:28:29

Setup failed for custom integration 'huawei_solar': Unable to import component: No module named 'pymodbus.message'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1035, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1015, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/huawei_solar/__init__.py", line 9, in <module>
    from huawei_solar import (
    ...<9 lines>...
    )
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/__init__.py", line 4, in <module>
    from .bridge import (
    ...<6 lines>...
    )
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/bridge.py", line 34, in <module>
    from .huawei_solar import (
    ...<5 lines>...
    )
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 19, in <module>
    from pymodbus.message.rtu import MessageRTU
ModuleNotFoundError: No module named 'pymodbus.message'

### Describe your Huawei Solar Setup

Inverter Type: 
Inverter Firmware version: 
sDongle present: Yes / No
sDongle Type: sDongleA-05 (WiFi / Ethernet) / SDongleA-03 (4G)
sDongle Connectivitiy: WiFi / Ethernet / 4G
sDongle Firmware: 
Power meter present: three phase / single phase / no
Optimizers Present: Yes / No
Battery: LUNA2000-SO xxkWh / LUNA2000-S1 xxkWh (Released 2024) / LG RESU xxkWh / None
Battery Firmware version: 
Huawei Solar integration version: 

### How do you connect to the inverter?

Via the SDongle, wired connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
Logger: homeassistant.setup
Quelle: setup.py:334
Erstmals aufgetreten: 21:28:29 (1 Vorkommnisse)
Zuletzt protokolliert: 21:28:29

Setup failed for custom integration 'huawei_solar': Unable to import component: No module named 'pymodbus.message'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1035, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1015, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/huawei_solar/__init__.py", line 9, in <module>
    from huawei_solar import (
    ...<9 lines>...
    )
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/__init__.py", line 4, in <module>
    from .bridge import (
    ...<6 lines>...
    )
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/bridge.py", line 34, in <module>
    from .huawei_solar import (
    ...<5 lines>...
    )
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 19, in <module>
    from pymodbus.message.rtu import MessageRTU
ModuleNotFoundError: No module named 'pymodbus.message'
```

### Please confirm the following:

- [ ] I'm running the latest release of Home Assistant.
- [ ] I'm running the latest release of this integration.
- [ ] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2025-01-03:
> [!WARNING]
> Users that use the native modbus-integration alongside this integration should not upgrade to Home Assistant 2025.1, as it contains an upgrade to an incompatible version 3.7.4 of the pyModbus library. Significant work has already been done to upgrade this integration to support v3.8.0 of this library. We expect this to release alongside Home Assistant 2025.2, which is expected to also upgrade to that latest pyModbus version. This will solve the compatibility issue.

Criaga, 2025-01-04:
@wlcrs 
Hi, sorry but I'm not very expert. So you don't have to update to the new version 2025.1?

xm4rcell0x, 2025-01-04:
Second this bug.
Hard to revert to 2024.12.5 and is working well.

jwcappeln, 2025-01-04:
Yes, i downgrade core as well without restore a backup.
`ha core update --version 2024.12.5`

jomach, 2025-01-04:
same here. I just rollback the upgrade without any issues

ablyes, 2025-01-04:
Same problem for me. 
I have modbus for my hitachi heat pump and this plugin for the huawei converter. 
Is it possible to have a quick fix ?

jomach, 2025-01-04:
@ablyes  downgrade with the command: "ha core update --version 2024.12.5"

ablyes, 2025-01-04:
I don’t have any backup. 
Is this command working without ? With any type of installation ? I have a virtual machine under a synology.

[image]

ablyes, 2025-01-04:
Your command worked. 
Reading this command makes me understand that it doesn't rely on any backup. Just reach the asked version and install it. 
Thank you so much. I have my HA in a good state. 
I'll wait for an update of the HACS.

luzik, 2025-01-05:
Sorry for hate but this is the only integration I ever used that ask me for a downgrade. HA 2025.1 is also related to security updates. Now I can't downgrade because of others updated integration that requires ha 2025.1
You ask me to check all installed integration issues before HA upgrade? or only huawei_solar ? That's stupid!

Aljosha89, 2025-01-05:
> Sorry for hate but this is the only integration I ever used that ask me for a downgrade. HA 2025.1 is also related to security updates. Now I can't downgrade because of others updated integration that requires now HA 2025.1 You ask me to check all installed integration README before HA upgrade? or only huawei_solar ? That's stupid!

wlcrs is making huge effort to support this integration, your comment is untimely. If you do not like how the integration is handled you should not use it.

mariusnauj, 2025-01-05:
@wlcrs if it works with 2025.01 version of HA if I am using network type integration via inverter wi-fi?

luzik, 2025-01-05:
Don't get me wrong, I love this integration, just wanted to show that some things are important for some people.
This is how I'me using it
[image]

netmb, 2025-01-05:
Did you pay a penny for the software? Probably not. Accordingly, you have no claims whatsoever against the author. If the software is so important to you, program it yourself. Some people's sense of entitlement really pisses me off. Remember: the author created this in his free time, so we don't have to stress him out. If you can do better: Let's go.

mariusnauj, 2025-01-05:
Helio,

Maybe somebody who tryied can answer:

if it works with 2025.01 version of HA if I am using network type integration via inverter wi-fi?

luzik, 2025-01-05:
> Did you pay a penny for the software? Probably not. Accordingly, you have no claims whatsoever against the author. If the software is so important to you, program it yourself. Some people's sense of [...]

So I’ve got an idea to contribute a bit.
@wlcrs, if you upgrade any branch to include the already updated [huawei-solar-lib](https://github.com/wlcrs/huawei-solar-lib), which supports the new pymodbus, just add the requirement "homeassistant": "2025.2.0b0" in hacs.json. I’ll update my HA to the 2025.2 beta dev version and test the integration with my setup as much as my hardware allows.

luzik, 2025-01-05:
> Helio,
> 
> Maybe somebody who tryied can answer:
> 
> if it works with 2025.01 version of HA if I am using network type integration via inverter wi-fi?

I believe that it does not work

mimusato, 2025-01-05:
> Hello,
> 
> Maybe somebody who tryied can answer:
> 
> if it works with 2025.01 version of HA if I am using network type integration via inverter wi-fi?

For me it fails with:

Error setting up entry SUN2000-4KTL-L1 for huawei_solar
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 640, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/huawei_solar/__init__.py", line 207, in async_setup_entry
    await hass.config_entries.async_forward_entry_setups(entry, PLATFORMS)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2278, in async_forward_entry_setups
    await integration.async_get_platforms(platforms)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1185, in async_get_platforms
    import_future.result()
    ~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1185, in async_get_platforms
    import_future.result()
    ~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1185, in async_get_platforms
    import_future.result()
    ~~~~~~~~~~~~~~~~~~~~^^
  [Previous line repeated 1 more time]
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1173, in async_get_platforms
    platforms.update(self._load_platforms(platform_names))
                     ~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1098, in _load_platforms
    platform_name: self._load_platform(platform_name)
                   ~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1272, in _load_platform
    cache[full_name] = self._import_platform(platform_name)
                       ~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1304, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
           ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/huawei_solar/sensor.py", line 16, in <module>
    from homeassistant.const import (
    ...<8 lines>...
    )
ImportError: cannot import name 'ENERGY_KILO_WATT_HOUR' from 'homeassistant.const' (/usr/src/homeassistant/homeassistant/const.py)

---

Note: a similar error has been fixed in another repo - https://github.com/tggm/rointe-hacs/pull/15/files

mimusato, 2025-01-05:
Never mind, I was running an old version. After upgrading to 1.4.1 it seems to work well.

mariusnauj, 2025-01-05:
> Never mind, I was running an old version. After upgrading to 1.4.1 it seems to work well.

What version of HA, and what type of integration?

mimusato, 2025-01-05:
#What version of HA, and what type of integration?#

HA version 2025.1.0. Using integration via inverter wi-fi.

0x442E472E, 2025-01-05:
huawei_solar 1.5.0b1 seems to work great for me with HA 2025.1.0! Thank you, @wlcrs :)

Criaga, 2025-01-05:
Good evening, sorry but you have to be patient 😅
 I read that it works for some and for others it doesn't. 
 I'm connected via the Dongle's IP and I would like to know if I can update to version 2025.1
Thk

Barbarul, 2025-01-05:
> Yes, i downgrade core as well without restore a backup. `ha core update --version 2024.12.5`

Rolled back with your command, and not it works fine. I will skip this core version.

ThomDietrich, 2025-01-07:
Thanks for all your work on this component! 🎉 

To contribute and for potential troubleshooting. I am on HA 2025.01 and using the latest 1.5.0b2, this is the error I am faced with:

```
Setup failed for custom integration 'huawei_solar': Unable to import component: No module named 'pymodbus.pdu.register_message'

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1035, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1015, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/huawei_solar/__init__.py", line 21, in <module>
    from huawei_solar import (
    ...<9 lines>...
    )
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/__init__.py", line 4, in <module>
    from .bridge import (
    ...<6 lines>...
    )
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/bridge.py", line 34, in <module>
    from .huawei_solar import (
    ...<5 lines>...
    )
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 20, in <module>
    from pymodbus.pdu.register_message import (
    ...<2 lines>...
    )
ModuleNotFoundError: No module named 'pymodbus.pdu.register_message'
```

mariusnauj, 2025-01-07:
After upgrading HA to 2025.1.1 integration not working:

Setup failed for custom integration 'huawei_solar': Unable to import component: cannot import name 'ModbusRequest' from 'pymodbus.pdu' (/usr/local/lib/python3.13/site-packages/pymodbus/pdu/__init__.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1015, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/huawei_solar/__init__.py", line 20, in <module>
    from huawei_solar import (
    ...<4 lines>...
    )
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/__init__.py", line 3, in <module>
    from .bridge import HuaweiSolarBridge
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/bridge.py", line 28, in <module>
    from .huawei_solar import (
    ...<5 lines>...
    )
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 17, in <module>
    from pymodbus.message.rtu import MessageRTU
  File "/usr/local/lib/python3.13/site-packages/pymodbus/message/__init__.py", line 7, in <module>
    from pymodbus.message.message import Message, MessageType
  File "/usr/local/lib/python3.13/site-packages/pymodbus/message/message.py", line 15, in <module>
    from pymodbus.message.rtu import MessageRTU
  File "/usr/local/lib/python3.13/site-packages/pymodbus/message/rtu.py", line 12, in <module>
    from pymodbus.factory import ClientDecoder
  File "/usr/local/lib/python3.13/site-packages/pymodbus/factory.py", line 15, in <module>
    from pymodbus import bit_read_message as bit_r_msg
  File "/usr/local/lib/python3.13/site-packages/pymodbus/bit_read_message.py", line 15, in <module>
    from pymodbus.pdu import ModbusRequest, ModbusResponse
ImportError: cannot import name 'ModbusRequest' from 'pymodbus.pdu' (/usr/local/lib/python3.13/site-packages/pymodbus/pdu/__init__.py)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1035, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1027, in async_get_component
    comp = self._get_component()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[... 4 lines cut ...]
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/huawei_solar/__init__.py", line 20, in <module>
    from huawei_solar import (
    ...<4 lines>...
    )
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/__init__.py", line 3, in <module>
    from .bridge import HuaweiSolarBridge
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/bridge.py", line 28, in <module>
    from .huawei_solar import (
    ...<5 lines>...
    )
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 17, in <module>
    from pymodbus.message.rtu import MessageRTU
  File "/usr/local/lib/python3.13/site-packages/pymodbus/message/__init__.py", line 7, in <module>
    from pymodbus.message.message import Message, MessageType
  File "/usr/local/lib/python3.13/site-packages/pymodbus/message/message.py", line 15, in <module>
    from pymodbus.message.rtu import MessageRTU
  File "/usr/local/lib/python3.13/site-packages/pymodbus/message/rtu.py", line 12, in <module>
    from pymodbus.factory import ClientDecoder
  File "/usr/local/lib/python3.13/site-packages/pymodbus/factory.py", line 15, in <module>
    from pymodbus import bit_read_message as bit_r_msg
  File "/usr/local/lib/python3.13/site-packages/pymodbus/bit_read_message.py", line 15, in <module>
    from pymodbus.pdu import ModbusRequest, ModbusResponse
ImportError: cannot import name 'ModbusRequest' from 'pymodbus.pdu' (/usr/local/lib/python3.13/site-packages/pymodbus/pdu/__init__.py)

mariusnauj, 2025-01-07:
I am using native pymodbus integration for my Komfovent device. SO after installed 2025.1.1 Huawei integration stop working

m3rsok3, 2025-01-07:
I suppose there will be a update so it works woth 2025 core soon?

luzik, 2025-01-08:
[PR](https://github.com/home-assistant/core/pull/134809) has been merged, new HA dev image has been build, but it still messing with modbus integration:

```
Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/modbus/sensor.py", line 114, in async_update
    raw_result = await self._hub.async_pb_call(
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._slave, self._address, self._count, self._input_type
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/components/modbus/modbus.py", line 409, in async_pb_call
    result = await self.low_level_pb_call(unit, address, value, use_call)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/modbus/modbus.py", line 374, in low_level_pb_call
    result: ModbusPDU = await entry.func(address, value, **kwargs)
                              ~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: ModbusClientMixin.read_input_registers() takes 2 positional arguments but 3 positional arguments (and 1 keyword-only argument) were given
```

JavanXD, 2025-01-10:
> Users that use the native modbus-integration alongside this integration should not upgrade to Home Assistant 2025.1, as it contains an upgrade to an incompatible version 3.7.4 of the pyModbus [...]

looks like we have to wait and downgrade first

ThomDietrich, 2025-01-10:
Maybe someone can help me here. I did downgrade to  2024.12.5, now I am seeing this new issue:

```
Logger: homeassistant.setup
Quelle: setup.py:334
Erstmals aufgetreten: 11:14:18 (1 Vorkommnisse)
Zuletzt protokolliert: 11:14:18

Setup failed for custom integration 'huawei_solar': Unable to import component: cannot import name 'FramerRTU' from 'pymodbus.framer' (/usr/local/lib/python3.13/site-packages/pymodbus/framer/__init__.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1014, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1074, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1022, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/huawei_solar/__init__.py", line 21, in <module>
    from huawei_solar import (
    ...<9 lines>...
    )
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/__init__.py", line 4, in <module>
    from .bridge import (
    ...<6 lines>...
    )
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/bridge.py", line 34, in <module>
    from .huawei_solar import (
    ...<5 lines>...
    )
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 18, in <module>
    from pymodbus.framer import FramerRTU
ImportError: cannot import name 'FramerRTU' from 'pymodbus.framer' (/usr/local/lib/python3.13/site-packages/pymodbus/framer/__init__.py). Did you mean: 'Framer'?

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1034, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1026, in async_get_component
    comp = self._get_component()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1074, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1022, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/huawei_solar/__init__.py", line 21, in <module>
    from huawei_solar import (
    ...<9 lines>...
    )
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/__init__.py", line 4, in <module>
    from .bridge import (
    ...<6 lines>...
    )
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/bridge.py", line 34, in <module>
    from .huawei_solar import (
    ...<5 lines>...
    )
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 18, in <module>
    from pymodbus.framer import FramerRTU
ImportError: cannot import name 'FramerRTU' from 'pymodbus.framer' (/usr/local/lib/python3.13/site-packages/pymodbus/framer/__init__.py). Did you mean: 'Framer'?
```

Timbolot, 2025-01-13:
I have the same issue with 2025.02

Logger: homeassistant.setup
Quelle: setup.py:334
Erstmals aufgetreten: 09:56:01 (1 Vorkommnisse)
Zuletzt protokolliert: 09:56:01

Setup failed for custom integration 'huawei_solar': Unable to import component: No module named 'pymodbus.message'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1035, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1015, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/huawei_solar/__init__.py", line 20, in <module>
    from huawei_solar import (
    ...<4 lines>...
    )
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/__init__.py", line 3, in <module>
    from .bridge import HuaweiSolarBridge
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/bridge.py", line 28, in <module>
    from .huawei_solar import (
    ...<5 lines>...
    )
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 17, in <module>
    from pymodbus.message.rtu import MessageRTU
ModuleNotFoundError: No module named 'pymodbus.message'

MrNickIE, 2025-01-14:
Issue still exists with 2025.02.

Sorry, and thanks for the hard work!

ThomDietrich, 2025-01-16:
Did anyone find a workaround for this? The downgrade didn't resolve the issue for me (see above) and I have no visibility on my solar system for more than a week. This is especially bad for long-term data collection and analysis in Grafana :-/

@wlcrs wrote:

> Significant work has already been done to upgrade this integration to support v3.8.0 of this library. We expect this to release alongside Home Assistant 2025.2, which is expected to also upgrade to that latest pyModbus version. This will solve the compatibility issue.

Is there any news on this? Could someone please reference the relevant PR?

luzik, 2025-01-16:
For me HA 2025.2 dev working fine. If you have extra modbus devices then here is a fix that waits for merge https://github.com/home-assistant/core/pull/135108

MrNickIE, 2025-01-17:
How would you go back to the non-dev version after?  I do not see that in the above instructions.

MrNickIE, 2025-01-17:
Also, the Dev build did not work for me, and now I have a different error:
Setup failed for custom integration 'huawei_solar': Unable to import component: No module named 'pymodbus.message'

Zbiku91, 2025-01-17:
Thanks @luzik for the hint, ver `2025.2.0.dev202501170226` works for me. I'm curious if this will be merged to the official release 2025.1.3

ThomDietrich, 2025-01-17:
> How would you go back to the non-dev version after? I do not see that in the above instructions.

```
ha supervisor options --channel stable
```

MrNickIE, 2025-01-17:
> Thanks [@luzik](https://github.com/luzik) for the hint, ver `2025.2.0.dev202501170226` works for me. I'm curious if this will be merged to the official release 2025.1.3

Really odd, it did not work for me.

ThomDietrich, 2025-01-17:
@MrNickIE for your comparison:

My system:
- Core 2025.2.0.dev202501160226
-  Supervisor 2025.01.0.dev1703
-  Operating System 14.1 
- huawei_solar v1.5.0b2
- additional modbus sensors using yaml config

MrNickIE, 2025-01-17:
Thanks, I have the same setup as you.

I have downgraded to the 2024 version now and have things back to normal.

joergwi123, 2025-01-17:
Ich habe jetzt auch wieder auf 2024.12.5 zurück gespielt, Huawei funktioniert wieder, aber dafür ist kein Modbussensor mehr vorhanden, auch die Integration ist nicht mehr vorhanden.
Ich hoffe, das der Fehler bald behoben ist.
Danke für die Arbeit.

m3rsok3, 2025-01-17:
I have never done a role back before and are new to ha. But should I just wait for a update from hacs? I had mayor issues to setup the Huawei application.

MrNickIE, 2025-01-18:
If you can wait.  Wait.  

I had to go back as I was adding a new HRV system to HA today and had to make it work (I didn't as it happens) 😆

bluenobbe, 2025-01-21:
Has someone testet Home Assistant Core Update 2025.1.3 ?? is this update fixing the problem regarding pymodbus.message

sakalauskas, 2025-01-21:
@bluenobbe still the same on Home Assistant Core Update 2025.1.3
```
Aborting client creation due to error
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 178, in create
    client = AsyncHuaweiSolarModbusTcpClient(host, port, timeout)
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/modbus.py", line 65, in __init__
    super().__init__(host, port=port, timeout=timeout, reconnect_delay=RECONNECT_DELAY)
    ~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/modbus.py", line 28, in __init__
    super().__init__(*args, **kwargs, trace_connect=self._trace_connect)  # forward all unused arguments
    ~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: AsyncModbusTcpClient.__init__() got an unexpected keyword argument 'trace_connect'
```

MrNickIE, 2025-01-21:
Confirmed still broken in 2025.1.3

> Has someone testet Home Assistant Core Update 2025.1.3 ?? is this update fixing the problem regarding pymodbus.message

Confirmed still broken in 2025.1.3

Lyngaae, 2025-01-21:
> Confirmed still broken in 2025.1.3
> 
> > Has someone testet Home Assistant Core Update 2025.1.3 ?? is this update fixing the problem regarding pymodbus.message

Have you tried to update to the beta version 1.5.0?

bsergei, 2025-01-22:
I have HA Core 2025.1 and Huawei Solar 1.5.0b2. It worked somehow for couple weeks. Suddenly after reboot I am getting:

`Setup failed for custom integration 'huawei_solar': Unable to import component: No module named 'pymodbus.pdu.register_message'`.

Update to HA Core 2025.1.3 doesn't help.

jomach, 2025-01-23:
Is someone on ha core actively working on this? Seems to block a lot of people now 2025.1.3 is out now

wlcrs, 2025-01-23:
My PR's that [upgrade pyModbus to v3.8.3](https://github.com/home-assistant/core/pull/134809) and [fix some compatibility issues](https://github.com/home-assistant/core/pull/135108) due to pyModbus API changes in the core modbus-integration have been accepted in HA Core yesterday. These code changes will be released with HA 2025.2 . Do not expect to see any fixes in 2025.1.x releases.

This will fix the compatibility issue between this integration (starting from v1.5.0) and the core modbus-integration: both will use the same pyModbus version 3.8.3, fixing the crashes due to an incompatible pyModbus version being present.

Please note that if you are using another custom integration which relies on an incompatible pyModbus version (ie. 3.7.x), then that integration will need to upgrade it's pyModbus dependency to 3.8.3 to restore compatibility with the versions which HA Core and this integration expect.

Side note: my contributions on HA Core have brought me into contact with some core maintainers. I have raised the issue of pyModbus version changes potentially breaking this and other integrations. While there is easy solution to this - it's a limitation of how Python libraries are installed - they are at least aware of the implications. Having a direct line will hopefully allow me to coordinate better to prevent issues like this one.

Lyngaae, 2025-01-23:
> My PR's that [upgrade pyModbus to v3.8.3](https://github.com/home-assistant/core/pull/134809) and [fix some compatibility issues](https://github.com/home-assistant/core/pull/135108) due to pyModbus [...]

Thank you for your with this issue and this integration. It is much appreciated from my side.

joergwi123, 2025-01-23:
Vielen Dank für deine Arbeit. Dann hoffen wir mal, dass 2025.2 blad kommt.

ThomDietrich, 2025-01-23:
Thanks @wlcrs for going through all this trouble and resolving the issues.

I can confirm that with the latest dev release from today both my huawei_solar as well as yaml defined modbus sensors work as expected again! Thank you!

iBLtz, 2025-01-24:
Thank you for the work  behind :)

Joeknx, 2025-01-29:
> My PR's that [upgrade pyModbus to v3.8.3](https://github.com/home-assistant/core/pull/134809) and [fix some compatibility issues](https://github.com/home-assistant/core/pull/135108) due to pyModbus [...]

I'm on HA 2025.2.0b0 and 1.5.0b2 but it still does not work.

wlcrs, 2025-01-31:
v1.5.0 of this integration has been released. This version depends on pyModbus v3.8, restoring compatibility with the native `modbus`-integration in HA 2025.2.

I will close this issue once HA 2025.2 is officially released next week.

> [!IMPORTANT]
> **If you are using another (custom) integration which also relies on pyModbus** then the compatibility issues will persist until the other integration is also upgraded to pyModbus v3.8.
> Our policy is to keep our pyModbus version compatible with that used by the `modbus`-integration in HA Core. We expect other (custom) integration maintainers to do the same.
>
> So if your HA installation is still broken after upgrading to HA 2025.2 and v1.5.0 of this integration due to the error `Unable to import component: No module named 'pymodbus.<xyz>'`:
> 1. Determine which other (custom) integration uses pyModbus
> 2. Contact the maintainers and ask them to upgrade to pyModbus v3.8
> 3. Wait for a new release of that integration to restore overall compatibility .

> [!CAUTION]
> Note that the compatibility issues with other (custom) integrations can seem to be intermittent, as the pyModbus version will always be determined by the last integration with a pyModbus-dependency to be installed on your HA.
> Please refrain from adding comments to this issue / creating new issues with messages like 'it started working again after upgrading to vX.Y.Z', or 'it broke again after upgrading to HA vX.Y' and follow the steps listed above instead.

mariusrga, 2025-01-31:
Is this version supposed to work with HA 2025.1.4 and newer?

MWagnerOE5AOO, 2025-01-31:
After Update the Setup works. After restart HA the same issue comes again.

[8 more replies]

closed 2025-02-06 by wlcrs