wlcrs/huawei_solar #877: [Bug]: 1.5.0b1 set TOU not working

jfberry, 2025-01-06:
### Describe the issue

I have been using an automation to change my TOU to activate a charge from grid for a long while now.  On update to HA 2025.1.0 and 1.5.0b1 this change is not applying.  Looking at the inverter entities they look ok

[image]

### Describe your Huawei Solar Setup

Inverter Type: SUN2000-6KTL-L1
Inverter Firmware version: V200R001C00SPC142
sDongle present: Yes 
sDongle Type: sDongleA-05 (WiFi / Ethernet)
sDongle Connectivitiy:  Ethernet 
sDongle Firmware: V200R022C10SPC114
Power meter present: single phase
Optimizers Present: No
Battery: LUNA2000-5KW-C0
Battery Firmware version: V100R002C00SPC624
Huawei Solar integration version: 1.5.0b1

### How do you connect to the inverter?

Via the SDongle, wired connection

### Upload your Diagnostics File

[config_entry-huawei_solar-964602c05dd1a5676457e18a422f7858.json](https://github.com/user-attachments/files/18320810/config_entry-huawei_solar-964602c05dd1a5676457e18a422f7858.json)

### Upload your relevant debug logs

```text
alarm_3, pv_01_voltage, pv_01_current, pv_02_voltage, pv_02_current
2025-01-06 14:49:30.329 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000 with length 20 from slave 1
2025-01-06 14:49:30.329 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x5c 0x0 0x0 0x0 0x6 0x1 0x3 0x7d 0x0 0x0 0x14
2025-01-06 14:49:31.873 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x5c 0x0 0x0 0x0 0x2b 0x1 0x3 0x28 0x0 0x6 0x0 0x2 0x0 0x7 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x81 0xb4 0xa 0xf 0x0 0x1a 0xc 0xaa 0x0 0x1a old_data:  addr=None
2025-01-06 14:49:31.873 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x5c 0x0 0x0 0x0 0x2b 0x1 0x3 0x28 0x0 0x6 0x0 0x2 0x0 0x7 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x81 0xb4 0xa 0xf 0x0 0x1a 0xc 0xaa 0x0 0x1a
2025-01-06 14:49:31.873 DEBUG (MainThread) [pymodbus.logging] decoded PDU function_code(3 sub -1) -> ReadHoldingRegistersResponse(dev_id=0, transaction_id=0, address=0, count=0, bits=[], registers=[6, 2, 7, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 33204, 2575, 26, 3242, 26], status=1) 
2025-01-06 14:49:31.873 DEBUG (MainThread) [pymodbus.logging] Frame advanced, resetting header!!
2025-01-06 14:49:31.873 DEBUG (MainThread) [huawei_solar.bridge] Batch update of the following registers: input_power, day_active_power_peak, active_power, efficiency, device_status, startup_time, shutdown_time, accumulated_yield_energy, total_dc_input_power, hourly_yield_energy, daily_yield_energy
2025-01-06 14:49:31.874 DEBUG (MainThread) [huawei_solar.huawei_solar] Sleeping for 0.049538 seconds before making next call
2025-01-06 14:49:31.924 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064 with length 52 from slave 1
2025-01-06 14:49:31.924 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x5d 0x0 0x0 0x0 0x6 0x1 0x3 0x7d 0x40 0x0 0x34
2025-01-06 14:49:33.736 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x5d 0x0 0x0 0x0 0x6b 0x1 0x3 0x68 0x0 0x0 0x0 0x88 0x9 0x76 0x0 0x0 0x0 0x0 0x4 0xb3 0x0 0x3 0x0 0x0 0x0 0x0 0x2 0x96 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x13 0x90 0x0 0x0 0x0 0x88 0x0 0x0 0x0 0x2 0x3 0xe8 0x13 0x7c 0x27 0x10 0x0 0xfc 0xb 0xb8 0x2 0x0 0x0 0x0 0x67 0x7b 0x9b 0xdf 0xff 0xff 0xff 0xff 0x0 0x0 0x0 0x87 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xff 0xff 0xff 0xfe 0x0 0xd 0x46 0x29 0x0 0x1d 0xc7 0xeb 0x0 0x13 0xb5 0xf7 0x67 0x7b 0xed 0x77 0x0 0x0 0x0 0x40 0x0 0x0 0x5 0x42 old_data:  addr=None
2025-01-06 14:49:33.736 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x5d 0x0 0x0 0x0 0x6b 0x1 0x3 0x68 0x0 0x0 0x0 0x88 0x9 0x76 0x0 0x0 0x0 0x0 0x4 0xb3 0x0 0x3 0x0 0x0 0x0 0x0 0x2 0x96 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x13 0x90 0x0 0x0 0x0 0x88 0x0 0x0 0x0 0x2 0x3 0xe8 0x13 0x7c 0x27 0x10 0x0 0xfc 0xb 0xb8 0x2 0x0 0x0 0x0 0x67 0x7b 0x9b 0xdf 0xff 0xff 0xff 0xff 0x0 0x0 0x0 0x87 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xff 0xff 0xff 0xfe 0x0 0xd 0x46 0x29 0x0 0x1d 0xc7 0xeb 0x0 0x13 0xb5 0xf7 0x67 0x7b 0xed 0x77 0x0 0x0 0x0 0x40 0x0 0x0 0x5 0x42
2025-01-06 14:49:33.736 DEBUG (MainThread) [pymodbus.logging] decoded PDU function_code(3 sub -1) -> ReadHoldingRegistersResponse(dev_id=0, transaction_id=0, address=0, count=0, bits=[], registers=[0, 136, 2422, 0, 0, 1203, 3, 0, 0, 662, 0, 0, 0, 0, 0, 5008, 0, 136, 0, 2, 1000, 4988, 10000, 252, 3000, 512, 0, 26491, 39903, 65535, 65535, 0, 135, 0, 0, 0, 0, 0, 65535, 65534, 13, 17961, 29, 51179, 19, 46583, 26491, 60791, 0, 64, 0, 1346], status=1) 
2025-01-06 14:49:33.736 DEBUG (MainThread) [pymodbus.logging] Frame advanced, resetting header!!
2025-01-06 14:49:33.737 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching TA2270269646_inverter_data_update_coordinator data in 3.409 seconds (success: True)
2025-01-06 14:49:33.739 DEBUG (MainThread) [huawei_solar.bridge] Batch update of the following registers: storage_state_of_capacity, storage_running_status, storage_bus_voltage, storage_bus_current, storage_charge_discharge_power, storage_total_charge, storage_total_discharge, storage_current_day_charge_capacity, storage_current_day_discharge_capacity
2025-01-06 14:49:33.739 DEBUG (MainThread) [huawei_solar.huawei_solar] Sleeping for 0.047660 seconds before making next call
2025-01-06 14:49:33.788 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760 with length 28 from slave 1
2025-01-06 14:49:33.788 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x5e 0x0 0x0 0x0 0x6 0x1 0x3 0x93 0x80 0x0 0x1c
2025-01-06 14:49:34.401 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x5e 0x0 0x0 0x0 0x3b 0x1 0x3 0x38 0x0 0x5a 0x0 0x0 0x0 0x2 0x10 0x8e 0xff 0xff 0x0 0x0 0x0 0x0 0x0 0x0 0x3a 0x98 0x0 0x0 0x3a 0x98 0x0 0x0 0x34 0xbc 0x0 0x0 0x34 0xbc 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x12 0x1f 0xc8 0x0 0x11 0xca 0x61 0x0 0x0 0x5 0x7 0x0 0x0 0x4 0xdd old_data:  addr=None
2025-01-06 14:49:34.401 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x5e 0x0 0x0 0x0 0x3b 0x1 0x3 0x38 0x0 0x5a 0x0 0x0 0x0 0x2 0x10 0x8e 0xff 0xff 0x0 0x0 0x0 0x0 0x0 0x0 0x3a 0x98 0x0 0x0 0x3a 0x98 0x0 0x0 0x34 0xbc 0x0 0x0 0x34 0xbc 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x12 0x1f 0xc8 0x0 0x11 0xca 0x61 0x0 0x0 0x5 0x7 0x0 0x0 0x4 0xdd
2025-01-06 14:49:34.401 DEBUG (MainThread) [pymodbus.logging] decoded PDU function_code(3 sub -1) -> ReadHoldingRegistersResponse(dev_id=0, transaction_id=0, address=0, count=0, bits=[], registers=[90, 0, 2, 4238, 65535, 0, 0, 0, 15000, 0, 15000, 0, 13500, 0, 13500, 0, 0, 0, 0, 0, 18, 8136, 17, 51809, 0, 1287, 0, 1245], status=1) 
2025-01-06 14:49:34.401 DEBUG (MainThread) [pymodbus.logging] Frame advanced, resetting header!!
2025-01-06 14:49:34.402 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching TA2270269646_battery_data_update_coordinator data in 3.002 seconds (success: True)
2025-01-06 14:49:36.010 INFO (MainThread) [custom_components.huawei_solar.services] Got the following bridge and update_coordinators in get_battery_bridge: <huawei_solar.bridge.HuaweiSUN2000Bridge object at 0x7fb9cdd963c0>, <custom_components.huawei_solar.update_coordinator.HuaweiSolarUpdateCoordinator object at 0x7fb9cf1e0c30>
2025-01-06 14:49:36.010 ERROR (MainThread) [homeassistant.components.automation.new_cheap_rate_set_tou_to_charge] New cheap rate, set TOU to charge: Error executing script. Unexpected error for call_service at pos 1: Wrong number of registers to write
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 526, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 764, in _async_call_service_step
    response_data = await self._async_run_long_action(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<9 lines>...
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 727, in _async_run_long_action
    return await long_task
           ^^^^^^^^^^^^^^^
[... 50 lines cut ...]
  File "/usr/src/homeassistant/homeassistant/core.py", line 2795, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2838, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/huawei_solar/services.py", line 553, in set_tou_periods
    await bridge.set(
    ...<2 lines>...
    )
[... cut ...]
huawei_solar.exceptions.WriteException: Wrong number of registers to write
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2025-01-06:
Thank you for your bug report. Can you test again with 1.5.0 beta 2? Please also validate if the battery charges/discharges as expected: due to certain functionality being removed from pyModbus, I had to re-implement it myself. Always dangerous for unexpected bugs to pop up.

lauri33141, 2025-01-21:
Hi,

Im using EMMA and 1.5.0.b2. Hence EMMA (device?) is not listed as battery device I cant call service/action for TOU periods. 

Failed to perform the action huawei_solar.set_tou_periods. Not a valid 'Connected Energy Storage' device for dictionary value @ data['device_id']. Got None

wlcrs, 2025-01-21:
No reaction from the original Issue author after 2 weeks. So I assume that my fix in the latest beta version works.

closed 2025-01-21 by wlcrs

jfberry, 2025-01-21:
Sorry I missed the request for feedback since you closed this issue. I confirm it solves my particular issue.