wlcrs/huawei_solar #895: [Bug]: HA 2025.1.4 Huwei Solar 1.4.1

MWagnerOE5AOO, 2025-01-25:
### Describe the issue

Since an update of the huawei solar the configuration of the Huawei Solar 1.4.1 cause in an error.
"Der Konfigurationsfluss konnte nicht geladen werden:(message: Invalid handler specified)
Rollback to an older version not working.
Inverter Sun2000 10KTL with wifi Dongle. No Powermeter, Batterie or Optimzier

### Describe your Huawei Solar Setup

Inverter Type: sun2000 10 HKT
Inverter Firmware version: 
sDongle present: Yes 
sDongle Type: sDongleA-05 WiFi  
sDongle Connectivitiy: WiFi 
sDongle Firmware: 
Power meter present: three phase / single phase / no
Optimizers Present: No
Battery Firmware version: 
Huawei Solar integration version: 1.4.1

### How do you connect to the inverter?

Via the SDongle, wireless connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
Where i can get this
```

### Please confirm the following:

- [x] I'm running the latest release of Home Assistant.
- [x] I'm running the latest release of this integration.
- [x] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

hexadmin, 2025-01-25:
Same problem here. Sun2000 6KTL with wifi Dongle

denmaus, 2025-01-26:
My setup with SUN2000-8KTL-M1 (Firmware Version V100R001C00SPC168) with SDongle via LAN works fine.

I had connection problems after Update SDongle to Firmware Version V200R022C10SPC200.
After Downgrade to V200R022C10SPC126 no issues with connection.

Maybe you have some firmware issues too?

cybermailer84, 2025-01-26:
Same problem with 1.4.1 and beta2 version.

For testing I reinstalled the integratration and update to latest 1.5.x.b2 and tried to reinstall. Reinstall integrated does not find and install Huawei integration.

MWagnerOE5AOO, 2025-01-27:
The Firmware was always the same version

cybermailer84, 2025-01-27:
In my case i do not have changed the firmware of any device (EMMA, MAP0). With old Home Assistant 2024.12 no problem. I went back to old version everything is fine. Strange was with installed integration i was able to update to new HA version and some value i have seen, but not all. therefore i deinstalled. But now a new installation with new HA Version is not possible.

inigoml, 2025-01-28:
Same problem here. Broken after 2025.1.4 update.

inigoml, 2025-01-28:
Jumping to 1.5 rc3 fixes my problem.

cybermailer84, 2025-01-28:
in my case probably not because of using modbus tcp also for nibe heatpump

closed 2025-01-31 by wlcrs

MWagnerOE5AOO, 2025-01-31:
After the update to.1.5.1 the configuration Dialog work again. After restart the same Problem gain