wlcrs/huawei_solar #901: [Bug]: Integration 1.5.0. does not start with serial connection

nurzen, 2025-01-31:
### Describe the issue

When i update this integration from 1.4.1 to 1.5.0 i get an error at startup crashing the integration.

### Describe your Huawei Solar Setup

Inverter Type: SUN2000 20KTL-M2; SUN2000 40KTL-M3
Inverter Firmware version:  V100R001C00SPC168;  V100R001C20SPC144
sDongle present: No
Power meter present: no
Optimizers Present: No
Battery:  None
Battery Firmware version: 
Huawei Solar integration version: 1.5.0

### How do you connect to the inverter?

Over serial, with a serial-to-USB stick

### Upload your Diagnostics File

no diagnostics available since no connection can be made.

### Upload your relevant debug logs

```text
2025-01-31 10:43:56.635 ERROR (MainThread) [huawei_solar.huawei_solar] Aborting client creation due to error
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 206, in create_rtu
    client = AsyncHuaweiSolarModbusSerialClient(
        port,
    ...<2 lines>...
        **serial_kwargs,
    )
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/modbus.py", line 57, in __init__
    super().__init__(self, port=port, baudrate=baudrate, timeout=timeout, **serial_kwargs)
    ~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/modbus.py", line 28, in __init__
    super().__init__(*args, **kwargs, trace_connect=self._trace_connect)  # forward all unused arguments
    ~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: AsyncModbusSerialClient.__init__() got multiple values for argument 'port'
```

### Please confirm the following:

- [x] I'm running the latest release of Home Assistant.
- [x] I'm running the latest release of this integration.
- [x] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [x] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2025-01-31:
Fixed in v1.5.1

I have pulled release v1.5.0 as this is a serious bug.

nurzen, 2025-01-31:
Nope, didnt work
2025-01-31 12:10:11.931 ERROR (MainThread) [huawei_solar.huawei_solar] Aborting client creation due to error
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 206, in create_rtu
    client = AsyncHuaweiSolarModbusSerialClient(
        port,
    ...<2 lines>...
        **serial_kwargs,
    )
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/modbus.py", line 57, in __init__
    super().__init__(self, port, baudrate=baudrate, timeout=timeout, **serial_kwargs)
    ~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/modbus.py", line 28, in __init__
    super().__init__(*args, **kwargs, trace_connect=self._trace_connect)  # forward all unused arguments
    ~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: AsyncModbusSerialClient.__init__() takes 2 positional arguments but 3 positional arguments (and 3 keyword-only arguments) were given

wlcrs, 2025-01-31:
Let's try that again with v1.5.2 😅

nurzen, 2025-01-31:
Perfect. Now it works.
Thank you

closed 2025-01-31 by wlcrs

firebowl, 2025-01-31:
Not Working for me:

Logger: homeassistant.config_entries
Quelle: config_entries.py:640
Erstmals aufgetreten: 13:02:09 (2 Vorkommnisse)
Zuletzt protokolliert: 13:03:30
Error setting up entry SUN2000-5KTL-M1

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 640, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/huawei_solar/__init__.py", line 246, in async_setup_entry
    await hass.config_entries.async_forward_entry_setups(entry, PLATFORMS)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2278, in async_forward_entry_setups
    await integration.async_get_platforms(platforms)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1124, in async_get_platforms
    if platform := self._get_platform_cached_or_raise(platform_name):
                   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1215, in _get_platform_cached_or_raise
    raise ModuleNotFoundError(
    ...<2 lines>...
    )
ModuleNotFoundError: Platform huawei_solar.number not found