wlcrs/huawei_solar #902: [Bug]: Échec de la configuration, nouvel essai: Unsupported product model 'EDF ESS-3K-Mono'

avidflyer17, 2025-01-31:
### Describe the issue

Since last update (1.5), my product isn't working anymore ...

Any advice ?

### Describe your Huawei Solar Setup

Inverter Type: ESS 3k Mono
Inverter Firmware version: V200LV200R001C00SPC144
sDongle present: Yes / No
sDongle Type: sDongleA-05 (WiFi / Ethernet)
sDongle Connectivitiy: WiFi
sDongle Firmware: V200R022C10SPC118
Power meter present: single phase
Optimizers Present: Yes 
Battery: None
Battery Firmware version: N/A
Huawei Solar integration version: 1.5 (was working fine on 1.4.x)

### How do you connect to the inverter?

Via the SDongle, wireless connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
Cette erreur provient d'une intégration personnalisée

Enregistreur: custom_components.huawei_solar.config_flow
Source: custom_components/huawei_solar/config_flow.py:508
intégration: Huawei Solar (documentation, problèmes)
S'est produit pour la première fois: 10:33:26 (2 occurrences)
Dernier enregistrement: 10:35:17

Unexpected exception while connecting via ModbusTCP
Traceback (most recent call last):
  File "/config/custom_components/huawei_solar/config_flow.py", line 508, in async_step_setup_network
    info = await validate_network_setup_auto_slave_discovery(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<3 lines>...
    )
    ^
  File "/config/custom_components/huawei_solar/config_flow.py", line 100, in validate_network_setup_auto_slave_discovery
    bridge = await create_tcp_bridge(
             ^^^^^^^^^^^^^^^^^^^^^^^^
    ...<3 lines>...
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/bridge.py", line 653, in create_tcp_bridge
    return await _create(await AsyncHuaweiSolar.create(host, port, slave_id), slave_id)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/bridge.py", line 679, in _create
    product_info = await HuaweiSolarProductInfo.retrieve_from_device(client, slave_id)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/bridge.py", line 68, in retrieve_from_device
    ) = await client.get_multiple(
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<8 lines>...
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 292, in get_multiple
    response = await self._read_registers(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<3 lines>...
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 429, in _read_registers
    return await _do_read()
           ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 377, in _do_read
    response = await self._client.read_holding_registers(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<3 lines>...
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 155, in execute
    raise ModbusIOException(txt)
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] No response received after 3 retries, continue with next request
```

### Please confirm the following:

- [x] I'm running the latest release of Home Assistant.
- [x] I'm running the latest release of this integration.
- [x] I did not find an existing issue describing this problem.
- [x] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [x] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

avidflyer17, 2025-01-31:
Thanks to automatic saves, I was able to get in working back on previous working version : 

[image]

Let me know if I can help solving issue in 1.5.0

wlcrs, 2025-01-31:
I haven't seen this model type before.

What inverter (and other) are you using? Is this a Huawei-inverter that has been rebadged by EDF? Can you provide me with some photos of the information stickers on your hardware?

avidflyer17, 2025-01-31:
It is a new inverter rebadged by EDF yes !

[image]

[image]

Is this OK ?

avidflyer17, 2025-01-31:
I only have access to the dongle wifi because the inverter is connected to their hardware via the yuze box on this photo : 

[image]

When I use fusion solar application, it says inverter is offline.

As I can get what I want via dongle, I don't care : 

[image]

I was able to download diagrams as it's working, hope it can help...

[config_entry-utility_meter-01JHNM9QQJWYXHQ48BNBWEV8FC.json](https://github.com/user-attachments/files/18616721/config_entry-utility_meter-01JHNM9QQJWYXHQ48BNBWEV8FC.json)

wlcrs, 2025-01-31:
I have added support for auto-discovery of these rebranded inverters by EDF with v1.5.1, this solves the error from the subject of this issue.

However, the debug log that you posted is just a generic read timeout, which should be unrelated to that issue?

If the still have problems with v1.5.1, please add the following lines to your `configuration.yaml`:

```
logger:
  logs:
    pymodbus: debug
    huawei_solar: debug
    homeassistant.components.huawei_solar: debug
```

Then restart HA and post your full logs, available from: [image]

avidflyer17, 2025-01-31:
I confirm this is OK !

Thanks for your work/time et the awesome support you have provided to me.

Regards,

[image]

Criaga, 2025-01-31:
@wlcrs 

Hi
! Why don't I get the notification to update to version 1.5.0 or 1.5.1? I'm currently on version 1.4.1.

Thk

closed 2025-01-31 by wlcrs

avidflyer17, 2025-01-31:
You need to use the refresh button in HACS.