wlcrs/huawei_solar #906: [Bug]: After updating to 1.5.0 (and 1.5.1 and 1.5.2) the integration fails to set up

tde01065, 2025-01-31:
### Describe the issue

After updating to 1.5.0 (and subsequently to 1.5.1 and finally to 1.5.2), the integration is no longer able to set up correctly. The error message states "Unsupported product model" (see screenshot).

[image]

### Describe your Huawei Solar Setup

Inverter Type: Svea Solar PowerShifter 10 (Huawei SUN2000)
Inverter Firmware version: V100R001C00SPC165
sDongle present: Yes
sDongle Type: sDongleA-05 (WiFi / Ethernet) / SDongleA-03 (4G)
sDongle Connectivitiy: WiFi / Ethernet / 4G
sDongle Firmware: 
Power meter present: three phase
Optimizers Present: No
Battery: LUNA2000
Battery Firmware version: V100R002C00SPC123
Huawei Solar integration version: 1.4.1

### How do you connect to the inverter?

Via the `SUN2000-<serial_no>` WiFi

### Upload your Diagnostics File

Unable to extract diagnostics, the option is not available (I'm guessing due to the fact that the integration is not loaded).

### Upload your relevant debug logs

```text
2025-01-31 12:54:38.045 DEBUG (MainThread) [pymodbus.logging] Connecting to 192.168.200.1:6607.
2025-01-31 12:54:38.045 DEBUG (MainThread) [pymodbus.logging] Connecting comm
2025-01-31 12:54:38.071 DEBUG (MainThread) [pymodbus.logging] Connected to comm
2025-01-31 12:54:38.071 DEBUG (MainThread) [huawei_solar.modbus] Waiting for 1500 milliseconds after connection before performing operations
2025-01-31 12:54:38.106 DEBUG (MainThread) [pymodbus.logging] -> transport: received eof
2025-01-31 12:54:38.107 DEBUG (MainThread) [pymodbus.logging] Connection lost comm due to None
2025-01-31 12:54:38.108 DEBUG (MainThread) [pymodbus.logging] Wait comm 1000000 ms before reconnecting.
2025-01-31 12:54:38.177 INFO (MainThread) [huawei_solar.huawei_solar] Waiting for connection
2025-01-31 12:54:39.573 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 30000 with length 65 from slave 0
2025-01-31 12:54:39.573 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x1 0x0 0x0 0x0 0x6 0x0 0x3 0x75 0x30 0x0 0x41
2025-01-31 12:54:39.589 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x1 0x0 0x0 0x0 0x85 0x0 0x3 0x82 0x50 0x6f 0x77 0x65 0x72 0x73 0x68 0x69 0x66 0x74 0x65 0x72 0x20 0x31 0x30 0x0 0x30 0x31 0x30 0x37 0x34 0x33 0x31 0x31 0x2d 0x30 0x30 0x38 0x0 0x0 0x48 0x56 0x32 0x33 0x39 0x30 0x32 0x34 0x31 0x31 0x36 0x38 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x31 0x30 0x37 0x34 0x33 0x31 0x31 0x2d 0x30 0x30 0x37 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x56 0x31 0x30 0x30 0x52 0x30 0x30 0x31 0x2d 0x30 0x32 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x56 0x31 0x30 0x30 0x52 0x30 0x30 0x31 0x43 0x30 0x30 0x53 0x50 0x43 0x31 0x36 0x35 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 old_data:  addr=None
2025-01-31 12:54:39.589 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x1 0x0 0x0 0x0 0x85 0x0 0x3 0x82 0x50 0x6f 0x77 0x65 0x72 0x73 0x68 0x69 0x66 0x74 0x65 0x72 0x20 0x31 0x30 0x0 0x30 0x31 0x30 0x37 0x34 0x33 0x31 0x31 0x2d 0x30 0x30 0x38 0x0 0x0 0x48 0x56 0x32 0x33 0x39 0x30 0x32 0x34 0x31 0x31 0x36 0x38 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x31 0x30 0x37 0x34 0x33 0x31 0x31 0x2d 0x30 0x30 0x37 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x56 0x31 0x30 0x30 0x52 0x30 0x30 0x31 0x2d 0x30 0x32 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x56 0x31 0x30 0x30 0x52 0x30 0x30 0x31 0x43 0x30 0x30 0x53 0x50 0x43 0x31 0x36 0x35 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2025-01-31 12:54:39.590 DEBUG (MainThread) [pymodbus.logging] decoded PDU function_code(3 sub -1) -> ReadHoldingRegistersResponse(dev_id=0, transaction_id=0, address=0, count=0, bits=[], registers=[20591, 30565, 29299, 26729, 26228, 25970, 8241, 12288, 12337, 12343, 13363, 12593, 11568, 12344, 0, 18518, 12851, 14640, 12852, 12593, 13880, 0, 0, 0, 0, 12337, 12343, 13363, 12593, 11568, 12343, 0, 0, 0, 0, 22065, 12336, 21040, 12337, 11568, 12800, 0, 0, 0, 0, 0, 0, 0, 0, 0, 22065, 12336, 21040, 12337, 17200, 12371, 20547, 12598, 13568, 0, 0, 0, 0, 0, 0], status=1)
2025-01-31 12:54:39.590 DEBUG (MainThread) [pymodbus.logging] Frame advanced, resetting header!!
```

### Please confirm the following:

- [x] I'm running the latest release of Home Assistant.
- [x] I'm running the latest release of this integration.
- [x] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [x] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

anowicki8422, 2025-01-31:
I also have this problem.
I noticed that HA does not make a copy after a software update and I have no way to go back to the previous version!!!

totovio, 2025-01-31:
The same issue. The error log:

Logger: huawei_solar.huawei_solar
Source: /usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py:146
First occurred: 2:27:16 PM (2 occurrences)
Last logged: 2:28:44 PM

Timeout while waiting for connection. Reconnecting
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/tasks.py", line 507, in wait_for
    return await fut
           ^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/locks.py", line 213, in wait
    await fut
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 141, in _communication_lock
    await asyncio.wait_for(
    ...<2 lines>...
    )
  File "/usr/local/lib/python3.13/asyncio/tasks.py", line 506, in wait_for
    async with timeouts.timeout(timeout):
               ~~~~~~~~~~~~~~~~^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/timeouts.py", line 116, in __aexit__
    raise TimeoutError from exc_val
TimeoutError

anowicki8422, 2025-01-31:
in 1.5.2 everything works. thank you.

totovio, 2025-01-31:
I have 1.5.2 and it doesn't work

tde01065, 2025-01-31:
@anowicki8422 
> I also have this problem. I noticed that HA does not make a copy after a software update and I have no way to go back to the previous version!!!

Go into HACS, find the integration. Under the three dots at the top right, click download again and select specific version. I reverted to 1.4.1 where my integration is working again (but with warnings at startup).

totovio, 2025-01-31:
After the inverter restart, the connection restored. The integrations is working. Thank you

maxbengtzen, 2025-01-31:
I'm having the same issue. Also a rebranded Huawei SUN2000 inverter from Svea Solar.

Reverting from 1.5.2 to 1.4.1 fixes the problem.

`unsupported product model 'Powershifter 601074315-008'`

wlcrs, 2025-01-31:
One of the few sources that mention this PowerShifter device is [this PDF](https://www.energiforetagen.se/globalassets/medlemsportalen-oppet/rikta-ratt/rikta_ratt.pdf) which probably lists the devices which were certified for usage on the Swedish electrical grid?
Looking to the firmware version numbers, it seems like there are some other rebadged SUN2000 and LUNA2000 devices on there: from Haier and Sigenergy.

I'm going to fallback to issuing a warning + trying to initialize devices as SUN2000 devices when I don't recognize the model name returned by the device. This should fix the initialization logic for all the rebadged inverters that were already working with v1.4 of this integration.

wlcrs, 2025-01-31:
Fixed with https://github.com/wlcrs/huawei-solar-lib/commit/edfe3aa1e57c4d0a0812206243384f5183691172 , which was released with v1.5.3 .

closed 2025-01-31 by wlcrs

maxbengtzen, 2025-01-31:
> Fixed with [wlcrs/huawei-solar-lib@edfe3aa](https://github.com/wlcrs/huawei-solar-lib/commit/edfe3aa1e57c4d0a0812206243384f5183691172) , which was released with v1.5.3 .

Thank you! That solved it for me! Happy weekend 🍻

[image]

tde01065, 2025-01-31:
Yes, I can also confirm that 1.5.3 solved the issue for me. Thanks!