wlcrs/huawei_solar #909: [Bug]: No longer working after latest upgrade today

Grovertje, 2025-01-31:
### Describe the issue

Hi ,
After performiung the upgrade today the integration is no longer working...

The logging says:

Logger: huawei_solar.huawei_solar
Bron: /usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py:146
Eerst voorgekomen: 15:06:12 (21 gebeurtenissen)
Laatst gelogd: 15:26:47

Timeout while waiting for connection. Reconnecting
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/tasks.py", line 507, in wait_for
    return await fut
           ^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/locks.py", line 213, in wait
    await fut
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 141, in _communication_lock
    await asyncio.wait_for(
    ...<2 lines>...
    )
  File "/usr/local/lib/python3.13/asyncio/tasks.py", line 506, in wait_for
    async with timeouts.timeout(timeout):
               ~~~~~~~~~~~~~~~~^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/timeouts.py", line 116, in __aexit__
    raise TimeoutError from exc_val
TimeoutError

Any solutions?

[image]

### Describe your Huawei Solar Setup

Inverter Type: SUN2000-5KTL-M101074320-006
Inverter Firmware version: V100R001C00SPC165
sDongle present: Yes
sDongle Type: sDongleA-05 (WiFi / Ethernet)
sDongle Connectivitiy: WiFi 
sDongle Firmware: V200R022C10SPC118
Power meter present:  no
Optimizers Present: Yes 
Battery: None
Battery Firmware version: 
Huawei Solar integration version: 

### How do you connect to the inverter?

Via the SDongle, wireless connection

### Upload your Diagnostics File

No diagnostics download possible...

### Upload your relevant debug logs

```text
Logger: huawei_solar.huawei_solar
Bron: /usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py:146
Eerst voorgekomen: 15:06:12 (21 gebeurtenissen)
Laatst gelogd: 15:26:47

Timeout while waiting for connection. Reconnecting
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/tasks.py", line 507, in wait_for
    return await fut
           ^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/locks.py", line 213, in wait
    await fut
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 141, in _communication_lock
    await asyncio.wait_for(
    ...<2 lines>...
    )
  File "/usr/local/lib/python3.13/asyncio/tasks.py", line 506, in wait_for
    async with timeouts.timeout(timeout):
               ~~~~~~~~~~~~~~~~^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/timeouts.py", line 116, in __aexit__
    raise TimeoutError from exc_val
TimeoutError
```

### Please confirm the following:

- [x] I'm running the latest release of Home Assistant.
- [x] I'm running the latest release of this integration.
- [ ] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

mariusrga, 2025-01-31:
I also updated to latest, it didn't worked with HA 2025 1.4, so I rolled back also HA and Huawei, but not it's not working. (at some reboot Huawei integrations works, but my other modbus sensor not, then after another reboot, modbus sensors work and not Huawei)

===
Setup failed for custom integration 'huawei_solar': Unable to import component: No module named 'pymodbus.message'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1035, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1015, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/huawei_solar/__init__.py", line 20, in <module>
    from huawei_solar import (
    ...<4 lines>...
    )
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/__init__.py", line 3, in <module>
    from .bridge import HuaweiSolarBridge
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/bridge.py", line 28, in <module>
    from .huawei_solar import (
    ...<5 lines>...
    )
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 17, in <module>
    from pymodbus.message.rtu import MessageRTU
ModuleNotFoundError: No module named 'pymodbus.message'
===

Grovertje, 2025-01-31:
Issue solved for me...

After using the reconfiguring option under the 3 dots in the integration and REissuing the IP nummer, it started working again..
Nu more issues at the moment...

closed 2025-01-31 by Grovertje

mariusrga, 2025-01-31:
> Issue solved for me...
> 
> After using the reconfiguring option under the 3 dots in the integration and REissuing the IP nummer, it started working again.. Nu more issues at the moment...

What HA version are you using?

Grovertje, 2025-01-31:
> > Issue solved for me...
> > After using the reconfiguring option under the 3 dots in the integration and REissuing the IP nummer, it started working again.. Nu more issues at the moment...
> 
> What HA version are you using?

I'm using 2025.1.4