wlcrs/huawei_solar #911: [Bug]: Time Of Use (TOU) periods unavailable

Dringard2, 2025-01-31:
### Describe the issue

With the new version 1.5.3 the TOU periods are unavailable

### Describe your Huawei Solar Setup

Inverter Type: 
Inverter Firmware version: 
sDongle present: Yes / No
sDongle Type: sDongleA-05 (WiFi )
sDongle Connectivitiy: WiFi 
sDongle Firmware: 
Power meter present: three phase / single phase / no
Optimizers Present: Yes / No
Battery: LUNA2000-SO xxkWh / LUNA2000-S1 xxkWh (Released 2024) / LG RESU xxkWh / None
Battery Firmware version: 
Huawei Solar integration version: 

### How do you connect to the inverter?

Via the SDongle, wireless connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics F
[home-assistant_huawei_solar_2025-01-31T17-06-58.025Z.log](https://github.com/user-attachments/files/18622119/home-assistant_huawei_solar_2025-01-31T17-06-58.025Z.log)

ile here.

### Upload your relevant debug logs

```text
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2025.1.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.13.1",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Paris",
    "os_name": "Linux",
    "os_version": "6.6.62-haos-raspi",
    "supervisor": "2024.12.3",
    "host_os": "Home Assistant OS 14.2",
    "docker_version": "27.2.0",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "daily_min_max": {
      "documentation": "https://github.com/philsson/HomeAssistantCustomComponents",
      "version": "1.0",
      "requirements": []
    },
    "hacs": {
      "documentation": "https://hacs.xyz/docs/use/",
      "version": "2.0.5",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "apiMareeInfo": {
      "documentation": "",
      "version": "1.1.7",
      "requirements": []
    },
    "start_time": {
[... 618 lines cut ...]
      },
      "14": {
        "__type": "<class 'huawei_solar.files.OptimizerRealTimeData'>",
        "repr": "OptimizerRealTimeData(optimizer_address=14, output_power=1.7, voltage_to_ground=489.2, alarm=[], output_voltage=24.9, output_current=0.07, input_voltage=30.6, input_current=0.09, temperature=12.2, running_status=<OptimizerRunningStatus.RUNNING: 4>, accumulated_energy_yield=732.127)"
      },
      "15": {
        "__type": "<class 'huawei_solar.files.OptimizerRealTimeData'>",
        "repr": "OptimizerRealTimeData(optimizer_address=15, output_power=1.3, voltage_to_ground=617.2, alarm=[], output_voltage=26.0, output_current=0.05, input_voltage=30.7, input_current=0.08, temperature=12.5, running_status=<OptimizerRunningStatus.RUNNING: 4>, accumulated_energy_yield=739.714)"
      },
      "16": {
        "__type": "<class 'huawei_solar.files.OptimizerRealTimeData'>",
        "repr": "OptimizerRealTimeData(optimizer_address=16, output_power=3.7, voltage_to_ground=591.7, alarm=[], output_voltage=29.0, output_current=0.13, input_voltage=30.9, input_current=0.15, temperature=11.5, running_status=<OptimizerRunningStatus.RUNNING: 4>, accumulated_energy_yield=699.825)"
      }
    }
  }
}
```

### Please confirm the following:

- [x] I'm running the latest release of Home Assistant.
- [x] I'm running the latest release of this integration.
- [x] I did not find an existing issue describing this problem.
- [x] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [x] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2025-02-01:
I don't see any issues in the logs and the diagnostics file indicates that the TOU entries are correctly read and available.

Please provide more information about what the issue is: add screenshots, add logs containing relevant errors, etc

chalkstorm, 2025-02-01:
[image]

Dringard2, 2025-02-01:
With the version 1.4.1, we can extract the different periods of the sensor battery_time_of_use_periods, but with the version 1.5.3 the sensor is unavailable.
See my screen shoot of my version 1.4.1

[image]
[image]

wlcrs, 2025-02-02:
Due to a renaming of an internal identifier to allow to distinguish between TOU periods set in the inverter or in the EMMA device, HA thinks a new sensor has been created, and the old one is not available anymore:

In my case, the new sensor is called `sensor.batteries_tou_charging_and_discharging_periods`

[image]

I will not change this back as this would introduce issues for new installations which contain an EMMA device. Please use the new sensor instead. You can explicitly delete the old sensor from your HA.

closed 2025-02-02 by wlcrs

Dringard2, 2025-02-02:
Good morning,Thank you for your response and your responsiveness, indeed the new sensor corrects the problem.Sincerely,Envoyé de mon iPad

image.png (view on web)

chalkstorm, 2025-02-02:
Perfect, thanks wlcrs :)