wlcrs/huawei_solar #912: [Bug]: Not possible to set max feed to grid greater then 11000W (11Kw) 2 Inverters in Cascade.

hedkvist, 2025-01-31:
### Describe the issue

I'm using 2 Inverters in cascade. (11+11Kw)
In the Huawei GUI i can set 14.5kw maximum feed to grid under 
Dongle -> Configuration -> Active Power Control -> Max grid feed in power. 

When i'm trying from HA, i cant set greater then 11000. (11kw) 

### Describe your Huawei Solar Setup

Inverter Type: Huawei Sun2000-10KTL-M1 and Sun2000-10KTL-M0
Inverter Firmware version: V100R001C00SPC167
sDongle present: Yes
sDongle Type: sDongleA-05  WIFI/Ethernet 
sDongle Connectivitiy: Ethernet
sDongle Firmware: V200R022C10SPC126
Power meter present: three phase 
Optimizers Present:  No
Battery: LUNA2000-SO None
Battery Firmware version:  N/A
Huawei Solar integration version: 1.4.1

### How do you connect to the inverter?

Please select your connection method

### Upload your Diagnostics File

[config_entry-huawei_solar-743c45dd48b53cc263c0bae5948c3ef3.json](https://github.com/user-attachments/files/18623655/config_entry-huawei_solar-743c45dd48b53cc263c0bae5948c3ef3.json)

### Upload your relevant debug logs

```text
File "/config/custom_components/huawei_solar/services.py", line 458, in set_maximum_feed_grid_power
    power = await _validate_power_value(service_call.data[DATA_POWER], bridge, rn.P_MAX)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/huawei_solar/services.py", line 274, in _validate_power_value
    raise ValueError(f"Power cannot be more than {maximum_active_power}W")
ValueError: Power cannot be more than 11000W
```

### Please confirm the following:

- [x] I'm running the latest release of Home Assistant.
- [x] I'm running the latest release of this integration.
- [x] I did not find an existing issue describing this problem.
- [x] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [x] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2025-02-01:
You are trying to change a different setting than in FusionSolar. This integration does not provide dongle values/settings, only inverters and EMMA device are supported. 

I don't own an SDongle, and don't want to add code I cannot test. If you are willing to contribute, you can add SDongle detection logic and registers to the underlying huawei-solar Python library. We can then add these SDongle sensors in this HA integration.

If you or anybody else wants to help: please let me know and I will provide some pointers to get you started.

hedkvist, 2025-02-01:
I would love to help, but don't think i got the knowledge to do so. I'm not so good at Python.
If i can help with something that isn't to advanced, I'm happy to do so.

By the way: If i set feed to grid to e.g. 4000w. the value  appears in the dongel. 
But a value greater than 11000 isn't possible from HA. therefore I thought it would be enough to remove the 11000w limit in HA if you have e.g. 2 inverters. See pictures. 

I looked in the code how to remove the 11000W limit but wasent sure how to do so. 

`async def set_maximum_feed_grid_power(
    hass: HomeAssistant, service_call: ServiceCall
) -> None:
    """Set Active Power Control to 'Power-limited grid connection' with the given wattage."""
    bridge, uc = get_inverter_bridge(hass, service_call)
    power = await _validate_power_value(service_call.data[DATA_POWER], bridge, rn.P_MAX)

    await bridge.set(rn.MAXIMUM_FEED_GRID_POWER_WATT, power)
    await bridge.set(
        rn.ACTIVE_POWER_CONTROL_MODE,
        rv.ActivePowerControlMode.POWER_LIMITED_GRID_CONNECTION_WATT,
    )

    await uc.async_refresh()`

[image]
[image]

Jasimo, 2025-03-10:
Also love to have the SDongle registers from the Huawei powermeter available.
Especially the modbus register h37119 (Positive_active_electricity_kWh) would be fine, or is it somewhere and it is just to early for me to find it.

BadP1t, 2025-05-27:
What is the register h37119 actually for and what are you trying to achieve by having it available ?

dettofatto, 2025-06-08:
Do you have any news about this? I would like reduce this value for preserve my grid...
I