wlcrs/huawei_solar #914: [Feature Request]: Support of new SUN5000 series (e.g SUN5000 MAP0)

flinterm, 2025-01-31:
### Describe your feature request

Hi,

It appears that the new SUN5000 MAP0 series is not yet supported.
Technically though they are quite similar / if not even the same as the SUN2000 MAP0 series with respect to the MODBUS Comms.

I am currently running EMMA together with the SUN5000 MAP0 12 but it is not possible to read the information from the inverter as slave (1).
Emma gets integrated normally.

Thanks for your help.

BR

### Proper usage

- [x] I confirm that this is not a bug report or support request
- [x] I confirm that this feature request is within the stated scope of the integration

wlcrs, 2025-02-01:
Please provide me with the modbus definitions pdf for this inverter. You will typically find it in the zip-file with firmware updates.

flinterm, 2025-02-01:
Hi,

Thanks alot for your quick reply.

Let me know if there is anything I can help with

[SUN2000MA V200R024C00SPC102 Modbus Interface Definitions(V3.0).pdf](https://github.com/user-attachments/files/18627896/SUN2000MA.V200R024C00SPC102.Modbus.Interface.Definitions.V3.0.pdf)

wlcrs, 2025-02-01:
Can you please provide logs of you trying to add the integration so that I can see what fails? 

This should just work

flinterm, 2025-02-02:
Please find attached the error logs

wlcrs, 2025-02-02:
Are you using the `AUTO` function to discover the slave ids? 

The logs show that the integration probes slave 1, which returns a general slave failure (ie. a slave with that number does not exist).

larrybml, 2025-02-03:
I had the same problem with SUN5000-12K-MAP0, but I downgraded HA to 2024.12.5 and installed version 1.4.1. of Huawei Solar and all is working. Now version 1.5.3. should work with SUN5000 model ? But I think that HA should still be kept to older version, right?