wlcrs/huawei_solar #915: [Feature Request]: add some extra sensors

tranzitwww, 2025-01-31:
### Describe your feature request

Hi,
First great job for this cool tool!

Can you please add the following Modbus sensors that are measuring the current month/year energy yield (which are now available thru FusionSolar app/website)?
They have the registers addresses 32116 and 32118 (both uint32)

Also add a option to stop querying the inverter during the night (ex: inverter_inverter_state = standby) for sensors that will only be available during the day (ex: inverter_input_power, inverter_active_power, inverter_pv_1_current, etc...)

```
#/homeassistant/modbus.yaml
- type: tcp
  host: "1.2.3.4"  #Huawei dongle
  port: 502           #Default Modbus TCP port (new is TCP 6607)
  name: "sun2000_huawei_inverter"
  delay: 0.1              # Delay between requests (in seconds)
  timeout: 3              # Timeout for responses (in seconds)

  sensors:
    # Current month energy yield
    - name: "Inverter - Current month energy yield"
      unique_id: "inverter_current_month_energy_yield"
      unit_of_measurement: "kWh"
      slave: 1
      address: 32116
      data_type: uint32
      scan_interval: 3600
      scale: 0.01
      precision: 2
      device_class: energy

    # Current year energy yield
    - name: "Inverter - Current year energy yield"
      unique_id: "inverter_current_year_energy_yield"
      unit_of_measurement: "kWh"
      slave: 1
      address: 32118
      data_type: uint32
      scan_interval: 3600
      scale: 0.01
      precision: 2
      device_class: energy
```

It will be nice to have the following ones. Now I create them thru templates.yaml
There were some issues when combining two sensors in one because the new one will have a larger amount of values (triggered each time a source sensor values is changed / and is multiplied by the source sensors number) and some of them might not be from real-time data. I've tried to fix that using "last_updated" time values.

1) House Power consumption - Right now there is no easy way to know how much power is consumed by the house utilities
2) Inverter Available power - How much power you still have available to consume from PV (which normally goes to the grid)
3) Inverter PV 1 and 2 power - By default you have voltage and amps and you need to multiply them to find out how much power is produced by each string. :)

```
#/homeassistant/templates.yaml
- sensor:
    - name: "House Power consumption"
      unique_id: house_power_consumption
      unit_of_measurement: "W"
      device_class: power
      state: >-
        {% set inverter_time = states.sensor.inverter_active_power.last_updated %}
        {% set meter_time = states.sensor.power_meter_active_power.last_updated %}
        
        {% set ac_production = states('sensor.inverter_active_power') | float(0) %}
        {% set meter_power = states('sensor.power_meter_active_power') | float(0) %}
        
        {% if ac_production == 0 %}
          {{ (meter_power | abs) | round(0) }}
        {% else %}
          {% if (inverter_time - meter_time).total_seconds() | abs <= 60 %}
            {% set power_consumption = (ac_production - meter_power) | round(0) %}
            {% if power_consumption >= 0 %}
              {{ power_consumption }}
            {% endif %}
          {% endif %}
        {% endif %}
      availability: >-
        {{ states('sensor.inverter_active_power') not in ['unknown', 'unavailable', 'none'] and
           states('sensor.power_meter_active_power') not in ['unknown', 'unavailable', 'none'] }}

    - name: "Inverter Available power"
      unique_id: inverter_available_power
      unit_of_measurement: "W"
      device_class: power
      state: >-
        {% set meter_power = states('sensor.power_meter_active_power') | float(0) %}
        {{ (meter_power if meter_power > 0 else 0) | round(0) }}
      availability: >-
        {{ states('sensor.power_meter_active_power') not in ['unknown', 'unavailable', 'none'] }}

    - name: "Inverter PV 1 power"
      unique_id: inverter_pv_1_power
      unit_of_measurement: "W"
      device_class: power
      state: >-
        {% set voltage = states('sensor.inverter_pv_1_voltage') | float(0) %}
        {% set current = states('sensor.inverter_pv_1_current') | float(0) %}
        {{ (voltage * current) | round(0) }}
      availability: >-
        {{ states('sensor.inverter_pv_1_voltage') not in ['unknown', 'unavailable', 'none'] and 
           states('sensor.inverter_pv_1_current') not in ['unknown', 'unavailable', 'none'] }}

    - name: "Inverter PV 2 power"
      unique_id: inverter_pv_2_power
      unit_of_measurement: "W"
      device_class: power
      state: >-
        {% set voltage = states('sensor.inverter_pv_2_voltage') | float(0) %}
        {% set current = states('sensor.inverter_pv_2_current') | float(0) %}
        {{ (voltage * current) | round(0) }}
      availability: >-
        {{ states('sensor.inverter_pv_2_voltage') not in ['unknown', 'unavailable', 'none'] and 
           states('sensor.inverter_pv_2_current') not in ['unknown', 'unavailable', 'none'] }}

```

### Proper usage

- [x] I confirm that this is not a bug report or support request
- [x] I confirm that this feature request is within the stated scope of the integration

wlcrs, 2025-02-01:
As stated at the top of the feature request form:

> **Scope of this integration**
> This integration aims to expose the information and functions made available by Huawei Solar inverters directly over Modbus in Home Assistant.
> 
> It does **NOT** do any interpretation of - or calculations with - this data.

I will thus only consider adding the weekly/monthly sensors provided by the inverter.

I'm not going to implement a lot of logic to pause querying for no good reason. It's bound to create more bugs/support requests because of edge cases.

Providing calculated sensors is also bound to get people complaining that it doesn't match their expectations/add up in calculations. I'd rather have the end user know which values to trust completely (=all of them) instead of having to guess.

tranzitwww, 2025-02-01:
Thanks for your answer.
I'll keep the calculation thru templates.yaml :)

[image]

commit 488c6f3a (2025-02-02): Add monthly and yearly yield sensors

Fixes #915
closed 2025-02-02 by wlcrs