wlcrs/huawei_solar #918: [Bug]: Configuration flow could not be loaded: {"message":"An invalid handler was specified

iulianmitea, 2025-02-01:
### Describe the issue

After you released huawai solar 1.5.3, I upgraded to si Core 2025.1.4, but the integration does not work, it gave me this message "Configuration flow could not be loaded: {"message":"An invalid handler was specified "}". the integration function in version 1.4.1 and core 2024.12.5., with 2 8 kw inverters connected via waier in Sdongle and the second 3 kw connected via wifi, no problems. I thought it was a conflict with Modbuss where I have a hitachi PDC mbuss via TCP, but after deleting it completely the message is the same

### Describe your Huawei Solar Setup

Inverter Type: huawai 2000
Inverter Firmware version:
sDongle present: Yes yes for inverter 8kw
sDongle Type: sDongleA-05 Ethernet - inverter 2000 8 kw
Power meter present: three phase and single phase
Huawei Solar integration version: 1.5.3

### How do you connect to the inverter?

Please select your connection method

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
-
```

### Please confirm the following:

- [ ] I'm running the latest release of Home Assistant.
- [ ] I'm running the latest release of this integration.
- [ ] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

anowicki8422, 2025-02-01:
[image]

cristianmitea, 2025-02-01:
yes lock same to me

closed 2025-02-01 by wlcrs

anowicki8422, 2025-02-01:
yesterday on 1.5.2 it worked.

In the evening I updated to 1.5.3 and it doesn't work. So the problem is now.