wlcrs/huawei_solar #922: [Feature Request]: Prevent unavailable values

gimmel, 2025-02-02:
### Describe your feature request

Although my inverter is located directly next to a WLAN access point, all values are often undefined for a short time. I would like to have the option of bridging such data pauses: Instead of undefined, the last value should then be output.

Or am I the only one with this "undefined" phases?

### Proper usage

- [x] I confirm that this is not a bug report or support request
- [x] I confirm that this feature request is within the stated scope of the integration

wlcrs, 2025-02-02:
If the value cannot be retrieved due to a bad connection, then the only correct value is 'unavailable'. Returning the last value is factually incorrect and would prevent users from detecting that the value is outdated.

closed 2025-02-02 by wlcrs

gimmel, 2025-02-03:
From a device perspective, you are of course absolutely right. But from the user interface point of view, it's not nice when the solar values jump to “unavailable” in between and destroy the clean UI.
I assumed that this was a bug in the Huawei inverters.

But of course this also happens with other devices and sometimes I create a template sensor (sensor.value_last) for this. With the Huawei integration, however, there are a lot of values that are then saved twice.

However, it would probably make more sense to incorporate this at HomeAssistant level so that sensors can be used for user interfaces in such a way that they always output the last valid value, perhaps with a checkbox.