wlcrs/huawei_solar #925: [Bug]: 1.5.3 constantly no connection / data from Inverters

ogghi, 2025-02-03:
### Describe the issue

HI there, since the latest 1.5.3 I often get disconnects.
Anything that changed here in relation to this?

### Describe your Huawei Solar Setup

Inverter Type:   SUN2000-10KTL-M1 +  SUN2000-12KTL-M2
Inverter Firmware version:  V100R001C00SPC165 + V100R001C00SPC165
sDongle present: Yes 
sDongle Type: sDongleA-05 ( Ethernet)
sDongle Connectivitiy:  Ethernet 
sDongle Firmware: V200R022C10SPC118
Power meter present: three phase
Optimizers Present: Yes
Battery: LUNA2000-5KW-C0 10kWh
Battery Firmware version: V100R002C00SPC624 (both)
Huawei Solar integration version: 1.5.3

### How do you connect to the inverter?

Via the SDongle, wired connection

### Upload your Diagnostics File

[config_entry-huawei_solar-d6c2f6c8f4f6ac800852eb7bec68550c.json](https://github.com/user-attachments/files/18642887/config_entry-huawei_solar-d6c2f6c8f4f6ac800852eb7bec68550c.json)
[config_entry-huawei_solar-d648a372145c4d8fb0ee41c971ea9909.json](https://github.com/user-attachments/files/18642888/config_entry-huawei_solar-d648a372145c4d8fb0ee41c971ea9909.json)
Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
2025-01-31 17:39:46.874 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.solax_inverter (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using state class 'measurement' which is impossible considering device class ('energy') it is using; expected None or one of 'total_increasing', 'total'; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2025-01-31 17:39:46.874 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.solarpanels (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using state class 'measurement' which is impossible considering device class ('energy') it is using; expected None or one of 'total_increasing', 'total'; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2025-01-31 17:39:47.638 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter
NoneType: None
2025-01-31 17:39:47.638 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter)
2025-01-31 17:39:47.782 ERROR (MainThread) [homeassistant.components.mqtt.device_trigger] Config for device trigger shellydimmer2-C45BBE473648-input-0 button_release conflicts with existing device trigger, cannot set up trigger, got: {'qos': 0, 'type': 'button_short_release', 'device': {'connections': [['mac', 'c4:5b:be:47:36:48']], 'sw_version': '20221027-104444/v1.12.1-ga9117d3', 'hw_version': 'gen1 (SHDM-2)', 'model': 'Shelly Dimmer 2', 'name': 'Shelly Dimmer 2 C45BBE473648', 'manufacturer': 'Allterco Robotics', 'configuration_url': 'http://192.168.1.16/', 'identifiers': []}, 'topic': 'shellies/shellydimmer2-C45BBE473648/input/0', 'automation_type': 'trigger', 'subtype': 'button_1', 'payload': '0', 'encoding': 'utf-8', 'value_template': None}
2025-01-31 17:39:47.784 ERROR (MainThread) [homeassistant.components.mqtt.device_trigger] Config for device trigger shellydimmer2-C45BBE473648-input-0 button_long_press conflicts with existing device trigger, cannot set up trigger, got: {'qos': 0, 'type': 'button_long_press', 'value_template': '{{value_json.event}}', 'device': {'connections': [['mac', 'c4:5b:be:47:36:48']], 'sw_version': '20221027-104444/v1.12.1-ga9117d3', 'hw_version': 'gen1 (SHDM-2)', 'model': 'Shelly Dimmer 2', 'name': 'Shelly Dimmer 2 C45BBE473648', 'manufacturer': 'Allterco Robotics', 'configuration_url': 'http://192.168.1.16/', 'identifiers': []}, 'topic': 'shellies/shellydimmer2-C45BBE473648/input_event/0', 'automation_type': 'trigger', 'payload': 'L', 'subtype': 'button_1', 'encoding': 'utf-8'}
2025-01-31 17:39:47.785 ERROR (MainThread) [homeassistant.components.mqtt.device_trigger] Config for device trigger shellydimmer2-C45BBE473648-input-0 button_short_press conflicts with existing device trigger, cannot set up trigger, got: {'qos': 0, 'type': 'button_short_press', 'value_template': '{{value_json.event}}', 'device': {'connections': [['mac', 'c4:5b:be:47:36:48']], 'sw_version': '20221027-104444/v1.12.1-ga9117d3', 'hw_version': 'gen1 (SHDM-2)', 'model': 'Shelly Dimmer 2', 'name': 'Shelly Dimmer 2 C45BBE473648', 'manufacturer': 'Allterco Robotics', 'configuration_url': 'http://192.168.1.16/', 'identifiers': []}, 'topic': 'shellies/shellydimmer2-C45BBE473648/input_event/0', 'automation_type': 'trigger', 'payload': 'S', 'subtype': 'button_1', 'encoding': 'utf-8'}
2025-01-31 17:39:47.785 ERROR (MainThread) [homeassistant.components.mqtt.device_trigger] Config for device trigger shellydimmer2-C45BBE473648-input-1 button_release conflicts with existing device trigger, cannot set up trigger, got: {'qos': 0, 'type': 'button_short_release', 'device': {'connections': [['mac', 'c4:5b:be:47:36:48']], 'sw_version': '20221027-104444/v1.12.1-ga9117d3', 'hw_version': 'gen1 (SHDM-2)', 'model': 'Shelly Dimmer 2', 'name': 'Shelly Dimmer 2 C45BBE473648', 'manufacturer': 'Allterco Robotics', 'configuration_url': 'http://192.168.1.16/', 'identifiers': []}, 'topic': 'shellies/shellydimmer2-C45BBE473648/input/1', 'automation_type': 'trigger', 'subtype': 'button_2', 'payload': '0', 'encoding': 'utf-8', 'value_template': None}
2025-01-31 17:39:47.786 ERROR (MainThread) [homeassistant.components.mqtt.device_trigger] Config for device trigger shellydimmer2-C45BBE473648-input-1 button_long_press conflicts with existing device trigger, cannot set up trigger, got: {'qos': 0, 'type': 'button_long_press', 'value_template': '{{value_json.event}}', 'device': {'connections': [['mac', 'c4:5b:be:47:36:48']], 'sw_version': '20221027-104444/v1.12.1-ga9117d3', 'hw_version': 'gen1 (SHDM-2)', 'model': 'Shelly Dimmer 2', 'name': 'Shelly Dimmer 2 C45BBE473648', 'manufacturer': 'Allterco Robotics', 'configuration_url': 'http://192.168.1.16/', 'identifiers': []}, 'topic': 'shellies/shellydimmer2-C45BBE473648/input_event/1', 'automation_type': 'trigger', 'payload': 'L', 'subtype': 'button_2', 'encoding': 'utf-8'}
2025-01-31 17:39:47.788 ERROR (MainThread) [homeassistant.components.mqtt.device_trigger] Config for device trigger shellydimmer2-C45BBE473648-input-1 button_short_press conflicts with existing device trigger, cannot set up trigger, got: {'qos': 0, 'type': 'button_short_press', 'value_template': '{{value_json.event}}', 'device': {'connections': [['mac', 'c4:5b:be:47:36:48']], 'sw_version': '20221027-104444/v1.12.1-ga9117d3', 'hw_version': 'gen1 (SHDM-2)', 'model': 'Shelly Dimmer 2', 'name': 'Shelly Dimmer 2 C45BBE473648', 'manufacturer': 'Allterco Robotics', 'configuration_url': 'http://192.168.1.16/', 'identifiers': []}, 'topic': 'shellies/shellydimmer2-C45BBE473648/input_event/1', 'automation_type': 'trigger', 'payload': 'S', 'subtype': 'button_2', 'encoding': 'utf-8'}
2025-01-31 17:39:48.639 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter
NoneType: None
2025-01-31 17:39:48.639 INFO (MainThread) [backoff] Backing off _do_read(...) for 2.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter)
2025-01-31 17:39:49.445 WARNING (MainThread) [aioesphomeapi.reconnect_logic] Can't connect to ESPHome API for kitchen-klima @ 192.168.1.89: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.1.89', port=6053))]: [Errno 113] Connect call failed ('192.168.1.89', 6053) (SocketAPIError)
2025-01-31 17:39:50.642 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter
NoneType: None
2025-01-31 17:39:50.642 INFO (MainThread) [backoff] Backing off _do_read(...) for 4.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter)
2025-01-31 17:39:55.117 WARNING (MainThread) [homeassistant.components.camera] Setup of camera platform amcrest is taking over 10 seconds.
2025-01-31 17:40:00.126 ERROR (MainThread) [pymodbus.logging] Cancel send, because not connected!
[... cut ...]
2025-01-31 18:15:35.381 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter
NoneType: None
```

### Please confirm the following:

- [x] I'm running the latest release of Home Assistant.
- [x] I'm running the latest release of this integration.
- [x] I did not find an existing issue describing this problem.
- [x] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [x] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2025-02-03:
This issue seems to be caused by a connection failure or an unstable connection to the inverter. This integration already has already implemented a lot of logic to automatically recover from connection failures or unstable connections. The three most important ones are:
- automatic reconnects if the connection to the inverter failed;
- each Modbus request is retried for up to 6 times, with increasing waits between each attempt to allow the connection to recover;
- after 3 failed Modbus requests the connection is reset to fix hanging connections.

Please understand that I cannot help you solve your connection failure or unstable connection. Consider [an alternative connection method](https://github.com/wlcrs/huawei_solar/wiki/Connecting-to-the-inverter).
If you are convinced that there is no connectivity issue, then I invite you to debug and fix the problem yourself and submit a pull request to the underlying [huawei-solar Python library](https://gitlab.com/Emilv2/huawei-solar). Your input would be greatly appreciated.

As there is nothing actionable for me to do in this issue, I'm closing it. You are welcome to open a topic in [Discussions: Connecting to the inverter](https://github.com/wlcrs/huawei_solar/discussions/categories/connecting-to-the-inverter) to get community support for your problem.

closed 2025-02-03 by wlcrs

ogghi, 2025-02-03:
I think it's just odd that since the last (2?) updates this started to happen.
What's an easy way to downgrade the integration installed?

WILHER, 2025-02-04:
The error only occurs for me if "Enable query for changes" is activated under "System options".

If I turn this option off, it no longer updates but also no longer loses the connection.

Is the data perhaps being requested too often?

ogghi, 2025-02-04:
Even more odd is that it seems stable as of today!
Will keep an eye on it and try the workaround in case it's not fine.
Thanks

ogghi, 2025-03-18:
@wlcrs 
Any idea on this?

Gaggioso, 2025-04-09:
I have the same issue. I have 3 Huawei inverters, connected via cable using an Ethernet dongle, a Luna battery, and many optimizers. On the Huawei app, everything works perfectly without any problems, ever.

In HomeAssistant, the values come and go intermittently. Sometimes I don't see the values for one inverter, sometimes for two or three; occasionally it's one inverter, other times another... What works or doesn't seem completely random. I have the latest firmware versions for all devices and doubt I have network issues. I'm an IT technician and all my equipment is functioning properly. I have no packet loss, and all other devices connected to the network work flawlessly.

Have you discovered anything about this issue?

ogghi, 2025-06-19:
Can I still comment here? It seems.
The issue persists on my end.
I have the dongle connected via Ethernet, no ping loss, I have monitoring for that.

Anybody having an idea on why this could happen or on how to debug?