wlcrs/huawei_solar #927: [Bug]: Charge Form Grid not available

briberolles, 2025-02-03:
### Describe the issue

I'm using the version 1.5.3 and the feature "Charge from Grid" (Battery) is in grey...
All others features are OK
I had same problems with the feature "TOU periods" but after fighting a litlle bit i have it (but i have old one TOU periods in grey). Impossible to delete the bad one (the good one is "battery_tou_periods")

Thanks for your help

[image]

[image]
[image]

### Describe your Huawei Solar Setup

Inverter Type: SUN2000 10KTL-M1
Inverter Firmware version: V100R001C00SPC153
sDongle present: Yes
sDongle Type: sDongleA-05 (WiFi / Ethernet)
sDongle Connectivitiy: WiFi
sDongle Firmware: V100R001C00SPC125
Power meter present: three phase
Optimizers Present: Yes
Battery: LUNA2000-SO 10kWh
Battery Firmware version: V100R002C00SPC119
Huawei Solar integration version: 1.5.3

### How do you connect to the inverter?

Please select your connection method

### Upload your Diagnostics File

{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2025.1.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.13.1",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Madrid",
    "os_name": "Linux",
    "os_version": "6.6.73-haos",
    "supervisor": "2024.12.3",
    "host_os": "Home Assistant OS 14.2",
    "docker_version": "27.2.0",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "tuya_ble": {
      "documentation": "https://www.home-assistant.io/integrations/tuya_ble",
      "version": "0.1.8",
      "requirements": [
        "tuya-iot-py-sdk==0.6.6",
        "pycountry==22.3.5"
      ]
    },
    "daikin_residential": {
      "documentation": "https://github.com/rospogrigio/daikin_residential/",
      "version": "2.3.1",
      "requirements": [
        "oic==1.6.0"
      ]
    },
    "hacs": {
      "documentation": "https://hacs.xyz/docs/use/",
[... 659 lines cut ...]
      }
    }
  }
}

### Upload your relevant debug logs

```text
2025-02-03 16:01:00.729 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration huawei_solar which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-02-03 16:02:13.438 WARNING (MainThread) [homeassistant.bootstrap] Waiting on integrations to complete setup: {('huawei_solar', '26c25d926cd4fae526ebaf325d076172'): 522401.601512395}
2025-02-03 16:02:28.372 WARNING (MainThread) [homeassistant.components.sensor] Setup of sensor platform huawei_solar is taking over 10 seconds.
2025-02-03 16:06:04.382 WARNING (MainThread) [homeassistant.util.loop] Detected blocking call to listdir with args ('/config/deps/lib/python3.13/site-packages',) inside the event loop by custom integration 'huawei_solar' at custom_components/huawei_solar/diagnostics.py, line 31: "pymodbus_version": version("pymodbus"), (offender: /usr/local/lib/python3.13/importlib/metadata/__init__.py, line 744: return os.listdir(self.root or '.')), please create a bug report at https://github.com/wlcrs/huawei_solar/issues
2025-02-03 16:06:04.412 WARNING (MainThread) [homeassistant.util.loop] Detected blocking call to read_text with args (PosixPath('/usr/local/lib/python3.13/site-packages/pymodbus-3.8.3.dist-info/METADATA'),) inside the event loop by custom integration 'huawei_solar' at custom_components/huawei_solar/diagnostics.py, line 31: "pymodbus_version": version("pymodbus"), (offender: /usr/local/lib/python3.13/importlib/metadata/__init__.py, line 915: return self._path.joinpath(filename).read_text(encoding='utf-8')), please create a bug report at https://github.com/wlcrs/huawei_solar/issues
2025-02-03 16:06:04.430 WARNING (MainThread) [homeassistant.util.loop] Detected blocking call to open with args (PosixPath('/usr/local/lib/python3.13/site-packages/pymodbus-3.8.3.dist-info/METADATA'),) inside the event loop by custom integration 'huawei_solar' at custom_components/huawei_solar/diagnostics.py, line 31: "pymodbus_version": version("pymodbus"), (offender: /usr/local/lib/python3.13/pathlib/_abc.py, line 632: with self.open(mode='r', encoding=encoding, errors=errors, newline=newline) as f:), please create a bug report at https://github.com/wlcrs/huawei_solar/issues
2025-02-03 16:13:38.574 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching TA22C0176309_power_meter_data_update_coordinator data in 3.106 seconds (success: True)
2025-02-03 16:13:44.224 WARNING (MainThread) [async_upnp_client.profiles.dlna] Only InstanceID 0 is supported
```

### Please confirm the following:

- [x] I'm running the latest release of Home Assistant.
- [x] I'm running the latest release of this integration.
- [x] I did not find an existing issue describing this problem.
- [x] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [x] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2025-02-03:
This is intentional. You cannot toggle "charge from grid" when active capacity control is enabled.

It is in fact forcibly on when capacity control is enabled, but Home Assistant does not support it to express that state.

closed 2025-02-03 by wlcrs

briberolles, 2025-02-03:
Thanks a lot for your quick reply

But Active Control Capacity and Charge From Grid (to charge 100% the battery exceptionally) are not incompatible
You can charge the battery respecting the maxi of Capacity allow by Active Control Capacity
For example:
Control Capacity : 10kW
Actual Load : 3kW
So i can charge from grid 7 kW.... The Metter will manage it..

Thanks
Bruno