wlcrs/huawei_solar #940: [Bug]:

TopoDiFogna, 2025-02-09:
### Describe the issue

After updating Homeassisstant to 2025.2.1 from 2025.2.0 integration shows error and everything is unavailable. Everything was working perfectly before that moment.

When I updated Homeassistant to latest version was nightime so no solar power from PV, only from batteries.

[image]

### Describe your Huawei Solar Setup

Inverter Type: SUN2000-6KTL-L1
Inverter Firmware version: V200R001C00SPC130
sDongle present: Yes
sDongle Type: sDongleA-05 (WiFi / Ethernet)
sDongle Connectivitiy: WiFi
sDongle Firmware: 
Power meter present: three phase / single phase / no
Optimizers Present: Yes
Battery: LUNA2000-SO 2 x 5kWh
Battery Firmware version: V100R002C00SPC621
Huawei Solar integration version: 1.5.3

### How do you connect to the inverter?

Via the SDongle, wireless connection

### Upload your Diagnostics File

N/A

### Upload your relevant debug logs

```text
2025-02-09 18:51:40.301 DEBUG (MainThread) [pymodbus.logging] Connecting to 192.168.0.250:6607.
2025-02-09 18:51:40.301 DEBUG (MainThread) [pymodbus.logging] Connecting comm
2025-02-09 18:51:40.307 DEBUG (MainThread) [pymodbus.logging] Connected to comm
2025-02-09 18:51:40.307 DEBUG (MainThread) [huawei_solar.modbus] Waiting for 1500 milliseconds after connection before performing operations
2025-02-09 18:51:40.365 DEBUG (MainThread) [pymodbus.logging] -> transport: received eof
2025-02-09 18:51:40.366 DEBUG (MainThread) [pymodbus.logging] Connection lost comm due to None
2025-02-09 18:51:40.366 DEBUG (MainThread) [pymodbus.logging] Wait comm 1000000 ms before reconnecting.
2025-02-09 18:51:40.408 INFO (MainThread) [huawei_solar.huawei_solar] Waiting for connection
2025-02-09 18:51:41.808 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 30000 with length 65 from slave 0
2025-02-09 18:51:41.809 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x1 0x0 0x0 0x0 0x6 0x0 0x3 0x75 0x30 0x0 0x41
2025-02-09 18:51:41.809 ERROR (MainThread) [pymodbus.logging] Exception response 255 / 0
2025-02-09 18:51:41.836 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x1 0x0 0x0 0x0 0x85 0x0 0x3 0x82 0x53 0x55 0x4e 0x32 0x30 0x30 0x30 0x2d 0x36 0x4b 0x54 0x4c 0x2d 0x4c 0x31 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x54 0x41 0x32 0x32 0x43 0x30 0x33 0x37 0x35 0x31 0x37 0x32 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x31 0x30 0x37 0x35 0x33 0x34 0x35 0x2d 0x30 0x31 0x35 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x56 0x32 0x30 0x30 0x52 0x30 0x30 0x31 0x43 0x30 0x30 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x56 0x32 0x30 0x30 0x52 0x30 0x30 0x31 0x43 0x30 0x30 0x53 0x50 0x43 0x31 0x33 0x30 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 old_data:  addr=None
2025-02-09 18:51:41.836 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x1 0x0 0x0 0x0 0x85 0x0 0x3 0x82 0x53 0x55 0x4e 0x32 0x30 0x30 0x30 0x2d 0x36 0x4b 0x54 0x4c 0x2d 0x4c 0x31 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x54 0x41 0x32 0x32 0x43 0x30 0x33 0x37 0x35 0x31 0x37 0x32 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x31 0x30 0x37 0x35 0x33 0x34 0x35 0x2d 0x30 0x31 0x35 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x56 0x32 0x30 0x30 0x52 0x30 0x30 0x31 0x43 0x30 0x30 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x56 0x32 0x30 0x30 0x52 0x30 0x30 0x31 0x43 0x30 0x30 0x53 0x50 0x43 0x31 0x33 0x30 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2025-02-09 18:51:41.836 DEBUG (MainThread) [pymodbus.logging] decoded PDU function_code(3 sub -1) -> ReadHoldingRegistersResponse(dev_id=0, transaction_id=0, address=0, count=0, bits=[], registers=[21333, 20018, 12336, 12333, 13899, 21580, 11596, 12544, 0, 0, 0, 0, 0, 0, 0, 21569, 12850, 17200, 13111, 13617, 14130, 0, 0, 0, 0, 12337, 12343, 13619, 13365, 11568, 12597, 0, 0, 0, 0, 22066, 12336, 21040, 12337, 17200, 12288, 0, 0, 0, 0, 0, 0, 0, 0, 0, 22066, 12336, 21040, 12337, 17200, 12371, 20547, 12595, 12288, 0, 0, 0, 0, 0, 0], status=1)
2025-02-09 18:51:41.836 DEBUG (MainThread) [pymodbus.logging] Frame advanced, resetting header!!
2025-02-09 18:51:42.244 DEBUG (MainThread) [pymodbus.logging] Connecting comm
2025-02-09 18:51:42.249 DEBUG (MainThread) [pymodbus.logging] Connected to comm
2025-02-09 18:51:42.250 DEBUG (MainThread) [huawei_solar.modbus] Waiting for 1500 milliseconds after connection before performing operations
2025-02-09 18:51:42.292 DEBUG (MainThread) [pymodbus.logging] -> transport: received eof
2025-02-09 18:51:42.293 DEBUG (MainThread) [pymodbus.logging] Connection lost comm due to None
2025-02-09 18:51:42.293 DEBUG (MainThread) [pymodbus.logging] Wait comm 1000000 ms before reconnecting.
2025-02-09 18:51:47.118 DEBUG (MainThread) [pymodbus.logging] Connecting to 192.168.0.250:6607.
2025-02-09 18:51:47.118 DEBUG (MainThread) [pymodbus.logging] Connecting comm
2025-02-09 18:51:47.121 DEBUG (MainThread) [pymodbus.logging] Connected to comm
2025-02-09 18:51:47.121 DEBUG (MainThread) [huawei_solar.modbus] Waiting for 1500 milliseconds after connection before performing operations
2025-02-09 18:51:47.155 DEBUG (MainThread) [pymodbus.logging] -> transport: received eof
2025-02-09 18:51:47.156 DEBUG (MainThread) [pymodbus.logging] Connection lost comm due to None
2025-02-09 18:51:47.156 DEBUG (MainThread) [pymodbus.logging] Wait comm 1000000 ms before reconnecting.
2025-02-09 18:51:47.222 INFO (MainThread) [huawei_solar.huawei_solar] Waiting for connection
2025-02-09 18:51:48.622 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 30000 with length 65 from slave 0
2025-02-09 18:51:48.622 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x1 0x0 0x0 0x0 0x6 0x0 0x3 0x75 0x30 0x0 0x41
2025-02-09 18:51:48.622 ERROR (MainThread) [pymodbus.logging] Exception response 255 / 0
2025-02-09 18:51:48.696 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x1 0x0 0x0 0x0 0x85 0x0 0x3 0x82 0x53 0x55 0x4e 0x32 0x30 0x30 0x30 0x2d 0x36 0x4b 0x54 0x4c 0x2d 0x4c 0x31 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x54 0x41 0x32 0x32 0x43 0x30 0x33 0x37 0x35 0x31 0x37 0x32 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x31 0x30 0x37 0x35 0x33 0x34 0x35 0x2d 0x30 0x31 0x35 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x56 0x32 0x30 0x30 0x52 0x30 0x30 0x31 0x43 0x30 0x30 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x56 0x32 0x30 0x30 0x52 0x30 0x30 0x31 0x43 0x30 0x30 0x53 0x50 0x43 0x31 0x33 0x30 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 old_data:  addr=None
2025-02-09 18:51:48.696 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x1 0x0 0x0 0x0 0x85 0x0 0x3 0x82 0x53 0x55 0x4e 0x32 0x30 0x30 0x30 0x2d 0x36 0x4b 0x54 0x4c 0x2d 0x4c 0x31 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x54 0x41 0x32 0x32 0x43 0x30 0x33 0x37 0x35 0x31 0x37 0x32 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x31 0x30 0x37 0x35 0x33 0x34 0x35 0x2d 0x30 0x31 0x35 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x56 0x32 0x30 0x30 0x52 0x30 0x30 0x31 0x43 0x30 0x30 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x56 0x32 0x30 0x30 0x52 0x30 0x30 0x31 0x43 0x30 0x30 0x53 0x50 0x43 0x31 0x33 0x30 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2025-02-09 18:51:48.696 DEBUG (MainThread) [pymodbus.logging] decoded PDU function_code(3 sub -1) -> ReadHoldingRegistersResponse(dev_id=0, transaction_id=0, address=0, count=0, bits=[], registers=[21333, 20018, 12336, 12333, 13899, 21580, 11596, 12544, 0, 0, 0, 0, 0, 0, 0, 21569, 12850, 17200, 13111, 13617, 14130, 0, 0, 0, 0, 12337, 12343, 13619, 13365, 11568, 12597, 0, 0, 0, 0, 22066, 12336, 21040, 12337, 17200, 12288, 0, 0, 0, 0, 0, 0, 0, 0, 0, 22066, 12336, 21040, 12337, 17200, 12371, 20547, 12595, 12288, 0, 0, 0, 0, 0, 0], status=1)
2025-02-09 18:51:48.696 DEBUG (MainThread) [pymodbus.logging] Frame advanced, resetting header!!
2025-02-09 18:51:58.969 DEBUG (MainThread) [pymodbus.logging] Connecting to 192.168.0.250:6607.
2025-02-09 18:51:58.969 DEBUG (MainThread) [pymodbus.logging] Connecting comm
2025-02-09 18:51:58.981 DEBUG (MainThread) [pymodbus.logging] Connected to comm
2025-02-09 18:51:58.981 DEBUG (MainThread) [huawei_solar.modbus] Waiting for 1500 milliseconds after connection before performing operations
2025-02-09 18:51:59.030 DEBUG (MainThread) [pymodbus.logging] -> transport: received eof
2025-02-09 18:51:59.031 DEBUG (MainThread) [pymodbus.logging] Connection lost comm due to None
2025-02-09 18:51:59.031 DEBUG (MainThread) [pymodbus.logging] Wait comm 1000000 ms before reconnecting.
[... cut ...]
2025-02-09 18:52:00.527 DEBUG (MainThread) [pymodbus.logging] Frame advanced, resetting header!!
```

### Please confirm the following:

- [x] I'm running the latest release of Home Assistant.
- [x] I'm running the latest release of this integration.
- [x] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [x] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

mgdelafuente, 2025-02-09:
Same error in my case, waiting to see if it is resolved

voyagerft, 2025-02-09:
Same error after upgrade 2025.2.21

tmycgn, 2025-02-09:
Same here after Upgrade

machinchose, 2025-02-09:
same, tried downgrading back to 2025.2.0, still same issue

TopoDiFogna, 2025-02-09:
> same, tried downgrading back to 2025.2.0, still same issue

Did you tried also with "old" dependencies versions or you Just downgraded homeadsistant?

machinchose, 2025-02-09:
> > same, tried downgrading back to 2025.2.0, still same issue
> 
> Did you tried also with "old" dependencies versions or you Just downgraded homeassistant?

Hi just homeassistant (basically ha core update --version 2025.2.0)

EmilionDK, 2025-02-09:
Same problem.

[image]

2025-02-09 20:07:33.217 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 113] Connect call failed ('192.168.1.19', 502)
2025-02-09 20:07:38.319 ERROR (MainThread) [huawei_solar.huawei_solar] Timeout while waiting for connection. Reconnecting
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/tasks.py", line 507, in wait_for
    return await fut
           ^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/locks.py", line 213, in wait
    await fut
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 141, in _communication_lock
    await asyncio.wait_for(
    ...<2 lines>...
    )
  File "/usr/local/lib/python3.13/asyncio/tasks.py", line 506, in wait_for
    async with timeouts.timeout(timeout):
               ~~~~~~~~~~~~~~~~^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/timeouts.py", line 116, in __aexit__
    raise TimeoutError from exc_val
TimeoutError
2025-02-09 20:07:41.408 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 113] Connect call failed ('192.168.1.19', 502)`

davdl92, 2025-02-09:
Same issue here; I tried backup restore both partial (i.e. only HA) and full (including add-ons) back to 2025.1.9 and it's still not working

grievertime, 2025-02-09:
Same issue

antnic78, 2025-02-09:
Same issue. With downgrade to 2024.1.4 and Huawei Solar 1.4.1 all works fine

TopoDiFogna, 2025-02-09:
> Same issue. With downgrade to 2024.1.4 and Huawei Solar 1.4.1 all works fine

Good catch!

Homeassistant 2025.2.1 and integration 1.4.1 everything works fine!

antnic78, 2025-02-09:
> > Same issue. With downgrade to 2024.1.4 and Huawei Solar 1.4.1 all works fine
> 
> Good catch!
> 
> Homeassistant 2025.2.1 and integration 1.4.1 everything works fine!

Great!

Cofirm that with Homeassistant 2025.2.1 and integration 1.4.1 everything works fine!

emocabras, 2025-02-09:
I confirm that it works for me with 2025.2.1 and integration 1.4.1. thanks.

radumarinoiu, 2025-02-09:
Same issue on upgrading to integration 1.4.1 and HA Core 2025.2.1, inverter firmware V200R001C00SPC146, tried via WIFI AP (was working until now) and then tried enabling "Local O&M" and tried connecting via my own wifi, getting the same error:
`huawei_solar.exceptions.ReadException: Got error while reading from register 30000 with length 65: ExceptionResponse(dev_id=1, transaction_id=0, address=0, count=0, bits=[], registers=[], status=1)`
Mind you, I cannot connect via WIFI AP anymore since I enabled Local O&M, which is to be expected.

Reverted to HA Core 2025.2.0, no effect. Reverting integration to 1.4.1 fixed it, thanks for the tip!

machinchose, 2025-02-10:
can confirm that reverting to integration version 1.4.1 fixed it

adbjma, 2025-02-10:
I'm greatful for the work that you all do here. @wlcrs 
While we are waiting I tried to revert back to 1.4.1 on 2025.2.1 and the error loading goes away but all the entities are unawailible, any suggestion here, was hoping to get this to work until pyModbus issue was corrected.

olibos, 2025-02-10:
Hello, for me, the issue is fixed with version 2025.2.1 and the latest update of the integration (1.5.4). 🎉

mariusnauj, 2025-02-10:
Hello,

@wlcrs if this integration will work with 2025.2.1 and with other native modbus integration. Because you wrote this solar version have pymodbus 3.8.5, and HA 2025.2.1 still using 3.8.3 ?

closed 2025-02-10 by wlcrs