wlcrs/huawei_solar #943: [Bug]: Emma: Inverter and battery configuration registers unavailable

delefant, 2025-02-10:
### Describe the issue

On the inverter with a battery connected, the write-enabled configuration registers of the inverter as well as the configuration and the per-module sensors of the battery entities instantly become unavailable.

On the other inverter without a connected battery, these write enabled registers such as MPPT-Scan On/Off are working.

Both inverters are accessed via modbus-tcp on an EMMA, connected via a Smartguard 63A-T0

### Describe your Huawei Solar Setup

Inverter Type: 2 x SUN2000 10k-MAP0
Inverter Firmware version: V200R024C00SPC102
MBUS version: V100R001C00SPC339
Emma firmware version: SmartHEMS V100R024C10SPC100 
Smartguard firmware version: V100R024C00SPC101
sDongle present: No
Power meter present: three phase
Optimizers Present: No
Battery: LUNA2000-S1 21kWh
Battery Firmware version: V200R023C10SPC115
Huawei Solar integration version: 1.5.4
Home Assistant version: 2025.2.1

### How do you connect to the inverter?

Via the SDongle, wired connection

### Upload your Diagnostics File

[config_entry-huawei_solar-01JKN80S4DKY3DRAPMPTMSV2KN.json](https://github.com/user-attachments/files/18738299/config_entry-huawei_solar-01JKN80S4DKY3DRAPMPTMSV2KN.json)
Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
2025-02-10 18:45:30.794 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47954 with length 2 from slave 4
2025-02-10 18:45:30.794 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x25 0x0 0x0 0x0 0x6 0x4 0x3 0xbb 0x52 0x0 0x2
2025-02-10 18:45:30.797 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x25 0x0 0x0 0x0 0x3 0x4 0x83 0x3 old_data:  addr=None
2025-02-10 18:45:30.797 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x25 0x0 0x0 0x0 0x3 0x4 0x83 0x3
2025-02-10 18:45:30.797 ERROR (MainThread) [pymodbus.logging] Exception response 131 / 0
2025-02-10 18:45:30.798 DEBUG (MainThread) [pymodbus.logging] Frame advanced, resetting header!!
2025-02-10 18:45:30.798 ERROR (MainThread) [custom_components.huawei_solar] Error fetching BT24A0032021_config_data_update_coordinator data: Could not update BT24A0032021 values: Got error while reading from register 47954 with length 2: ExceptionResponse(dev_id=4, transaction_id=37, address=0, count=0, bits=[], registers=[], status=1)
2025-02-10 18:45:30.798 DEBUG (MainThread) [huawei_solar.bridge] Batch update of the following registers: storage_state_of_capacity, storage_running_status, storage_bus_voltage, storage_bus_current, storage_charge_discharge_power, storage_total_charge, storage_total_discharge, storage_current_day_charge_capacity, storage_current_day_discharge_capacity
2025-02-10 18:45:30.799 DEBUG (MainThread) [huawei_solar.huawei_solar] Sleeping for 0.049070 seconds before making next call
```

### Please confirm the following:

- [x] I'm running the latest release of Home Assistant.
- [x] I'm running the latest release of this integration.
- [x] I did not find an existing issue describing this problem.
- [x] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [x] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2025-02-10:
I have no EMMA device, so I cannot easily debug this.

I need someone to compile me a list of entities which are incompatible with an EMMA device.

Please follow these steps:
- disable all but one configuration entities of the affected inverter and battery
- force an entity update with the `homeassistant.update_entity` action . You can do this in the "Actions" tab of the Developer tools. 
- Check if the entity becomes available again after a few seconds

Continue enabling entities and forcing an update until you see that all the entities become unavailable. If that is the case, add the entity to the list of incompatible entities. Disable the incompatible entity and continue enabling the next entity to check.

Once I receive this list, I will be able to remove those incompatible entities from the integration.

dragosmd, 2025-02-14:
@wlcrs Hi again :)
Is one of the errors I've reported from our first tests on Emma.

2025-02-14 21:09:25.229 DEBUG (MainThread) [huawei_solar.bridge] Storage capacity control as it is not supported by slave 123
2025-02-14 21:09:27.156 DEBUG (MainThread) [custom_components.huawei_solar.select] Storage capacity control is not supported by inverter HV2290299408
2025-02-14 21:09:27.860 DEBUG (MainThread) [custom_components.huawei_solar.number] Capacity control not supported on slave HV2290299408. Skipping capacity control number entities
2025-02-14 21:09:36.562 WARNING (SyncWorker_8) [homeassistant.components.huawei_lte] lan_host_info apparently not supported by device, excluding from future updates
2025-02-14 21:12:53.859 WARNING (MainThread) [custom_components.huawei_solar.config_flow] Skipping slave 131 with model SmartSwitch. Only SUN2000 inverters are supported as secondary slaves
2025-02-14 21:12:53.859 WARNING (MainThread) [custom_components.huawei_solar.config_flow] Skipping slave 122 with model SmartGuard-63A-S0. Only SUN2000 inverters are supported as secondary slaves
2025-02-14 21:12:54.164 DEBUG (MainThread) [huawei_solar.bridge] Storage capacity control as it is not supported by slave 123
2025-02-14 21:12:54.327 WARNING (MainThread) [custom_components.huawei_solar.config_flow] Skipping slave 127 with model SmartLoad. Only SUN2000 inverters are supported as secondary slaves
2025-02-14 21:12:54.819 DEBUG (MainThread) [huawei_solar.bridge] Storage capacity control as it is not supported by slave 123
**2025-02-14 21:12:54.984 DEBUG (MainThread) [custom_components.huawei_solar.select] Storage capacity control is not supported by inverter HV2290299408
2025-02-14 21:12:55.307 DEBUG (MainThread) [custom_components.huawei_solar.number] Capacity control not supported on slave HV2290299408. Skipping capacity control number entities**
2025-02-14 21:17:25.998 DEBUG (MainThread) [huawei_solar.bridge] Storage capacity control as it is not supported by slave 123
2025-02-14 21:17:26.161 DEBUG (MainThread) [custom_components.huawei_solar.select] Storage capacity control is not supported by inverter HV2290299408
2025-02-14 21:17:26.485 DEBUG (MainThread) [custom_components.huawei_solar.number] Capacity control not supported on slave HV2290299408. Skipping capacity control number entities

[home-assistant_huawei_solar_2025-02-14T19-18-13.153Z.log](https://github.com/user-attachments/files/18803756/home-assistant_huawei_solar_2025-02-14T19-18-13.153Z.log)

Inverter:
[image]
Batteries:

[image]
Lastest firmwares, latest integration.

My personal theory this is related to "Elevated permissions", all this entries are usually available with admin rights and since Emma ignores the authenitcation .... entires are detected but unavailable. Just a guess.
@delefant same issues as my pictures?

delefant, 2025-02-17:
Hi again,
@dragosmd Thanks for your reply.
As far as I can tell, the configuration entries only become available when "Elevated permissions" are set. According to https://github.com/wlcrs/huawei-solar-lib/commit/27dfdb429f525b8626a5bdcd9e6728ca5521735a EMMA should always have write access, which does seem to be the case, as both the inverter and the subordinate battery do not throw error messages, when changing (available) configuration parameters.

As to why the battery entities influence some inverter entities: I guess a generic check for write permissions may go wrong, when an expected register of the battery is not writable as intended, thus disabling the "write_permission" for the entire device, including the inverter as it is the parent device.

Our errors do not seem to be the same, heres a screenshot from my current (working) battery entities for comparison:

[image]

When enabling End-of-discharge SOC and Charge from grid, all configuration entities in the battery and the parent inverter become available, the log showing errors in reading register 47954 or 47955 respectively.

If I am interpreting https://github.com/wlcrs/huawei-solar-lib/blob/bb22046de7554798f37890aa37bc1f6efef633a0/src/huawei_solar/registers.py#L1392 right, the register for "Charge from Grid" should be 47087 (STORAGE_CHARGE_FROM_GRID_FUNCTION), so I´m wondering why I do not get an error containing this register.
I could not find a register for "End-of-discharge SOC", but functionally, 47102 (STORAGE_BACKUP_POWER_STATE_OF_CHARGE) which I assume is used for "Backup Power SOC" seems almost the same.

dragosmd, 2025-02-17:
> "Elevated permissions" are set. According to wlcrs/huawei-solar-lib@
> 27dfdb4

Without eelevated permissions I have the same errors.
Our lregisters do not match cause you have Map0 with S1 and I"m running L1
with S0.
I"ve enabled 3xcess and Working mode... dead
With them off ... dead... only if I enable elevTed permissions they work.

As I said Map0 doesn"t work as L1. Also L1 always required elevated
permissions for writing some registers.

Cheers

Sent from Gmail Mobile

jacekkjw, 2025-03-26:
I have exactly the same situation. Disabling configuration entities:
•	storage_charge_from_grid_function
•	storage_discharging_cutoff_capacity
make the rest available. My configuration works with "elevated permission" enabled.
I made some test with newly enabled  entities. “Batteries_working_mode” and "Batteries_excess_pv_energy_use_in_tou" do not seem to work. I can change their value, but it has no effect. Changing “Working_mode” and “Excess_pv_energy_use_in_tou” only works from EMMA settings.

jacekkjw, 2025-05-03:
Ping

radumarinoiu, 2025-05-07:
I'm also encountering this issue with the following entities from the "Batteries" device.

`switch.batteries_charge_from_grid`
```
2025-05-07 12:12:24.542 ERROR (MainThread) [pymodbus.logging] Exception response 131 / 0
2025-05-07 12:12:24.542 ERROR (MainThread) [custom_components.huawei_solar] Error fetching HV2280618393_config_data_update_coordinator data: Could not update HV2280618393 values: Got error while reading from register 47954 with length 1: ExceptionResponse(dev_id=3, transaction_id=21, address=0, count=0, bits=[], registers=[], status=1)
```

`number.batteries_end_of_discharge_soc`
```
2025-05-07 12:15:37.752 ERROR (MainThread) [pymodbus.logging] Exception response 131 / 0
2025-05-07 12:15:37.752 ERROR (MainThread) [custom_components.huawei_solar] Error fetching HV2280618393_config_data_update_coordinator data: Could not update HV2280618393 values: Got error while reading from register 47955 with length 1: ExceptionResponse(dev_id=3, transaction_id=22, address=0, count=0, bits=[], registers=[], status=1)
```

[image]

Installed Equipment:
EMMA-A02 - FW: `SmartHEMS V100R024C00SPC100` - SN: `NS2471380947`
SUN2000-6KTL-L1 - FW: `V200R001C00SPC149` - SN: `HV2280618393` - Parent: EMMA-02 - Comm Device: EMMA-02
SmartGuard-63A-S0 - FW: `V100R023C10SPC105` - SN: `NS248G161351` - Parent: EMMA-02 - Comm Device: EMMA-02
LUNA2000-1 - FW: `V100R002C00SPC633` - SN: `TA2480365652` - Parent: SUN2000 - Comm Device: EMMA-02

Let me know what else I can provide in order to solve this, if there's any other logs I can provide or if there's any way for me to provide the correct registers from my EMMA installation and thanks for all your work!

dragosmd, 2025-05-07:
Scrie mai sus cum se rezolva... momentan.

radumarinoiu, 2025-05-07:
For @dragosmd : Salut, te rog sa ma ajuti sa inteleg la care comment te referi de mai sus, si daca te referi la faptul ca le pot dezactiva, deja fac asta, dar as vrea sa folosesc End-of-discharge si momentan nu pot.

I'd be interested in setting End-of-discharge SOC higher when bad weather is announced, as there's a higher probability of grid failure. I already have Elevated Permissions enabled. If you know of a way to make the integration work with End of discharge, please let me know.
Otherwise if you meant I can "disable them" for the integration to work, I'm already doing that and the rest work fine, it's just I'm interested in one of the not working options :)

dragosmd, 2025-05-07:
End of discharge ramane mic 1-5% si nu te intereseaza.

Ce cauti tu la vreme rea este number.batteries_backup_power_soc. Procentul sub care nu se descarca bateria decat daca ramai fara curent.

Charge from grid trebuie sa raman iar activat si scos din HA(cu el dezactivat dupa ce ajunge invertorul la backup Soc intra in standby)

[image]

radumarinoiu, 2025-05-09:
Mersi mult, ai dreptate, merge cu Backup Power SOC.

closed 2025-05-29 by wlcrs

giuseppe99barchetta, 2025-06-17:
Hi, I've tried everything in this issue but nothing work for me. Can't read any battery values.

```
Error fetching 210107380310K5000014_config_data_update_coordinator data: Could not update 210107380310K5000014 values: Got error while reading from register 47299 with length 1: ExceptionResponse(dev_id=0, transaction_id=15, address=0, count=0, bits=[], registers=[], status=1)
Error fetching 210107380310K5000014_config_data_update_coordinator data: Could not update 210107380310K5000014 values: Got error while reading from register 47242 with length 56: ExceptionResponse(dev_id=0, transaction_id=24, address=0, count=0, bits=[], registers=[], status=1)
Error fetching 210107380310K5000014_battery_data_update_coordinator data: Could not update 210107380310K5000014 values: Got error while reading from register 37760 with length 28: ExceptionResponse(dev_id=0, transaction_id=25, address=0, count=0, bits=[], registers=[], status=1)
Error fetching 210107380310K5000014_config_data_update_coordinator data: Could not update 210107380310K5000014 values: Got error while reading from register 47242 with length 56: ExceptionResponse(dev_id=0, transaction_id=23, address=0, count=0, bits=[], registers=[], status=1)
Error fetching 210107380310K5000014_battery_data_update_coordinator data: Could not update 210107380310K5000014 values: Got error while reading from register 37760 with length 28: ExceptionResponse(dev_id=0, transaction_id=24, address=0, count=0, bits=[], registers=[], status=1)
```

SUN2000L-4KTL
Firmware: V100R001C00SPC334

commit 3257136b (2025-05-29): Create new entities for installations that don't support capacity control (#1024)

Fixes #943