wlcrs/huawei_solar #955: [Feature Request]: EMMA external CT values do not show

Jeffkip74, 2025-02-22:
### Describe your feature request

Seems like I miss alot of values like the external ct current etc. 
Is it possible to activate the sensors into the integration?

[image]

### Proper usage

- [x] I confirm that this is not a bug report or support request
- [x] I confirm that this feature request is within the stated scope of the integration

Jeffkip74, 2025-03-25:
Still not even a comment or simple answer on my issue? Am i the only one having this problem or does nobody use the Emma?

floschad, 2025-05-16:
I just switched from a simple smart meter to EMMA, and I'm sad that all the readings aren't there. A registry list is available here, if that helps: https://support.huawei.com/enterprise/de/doc/EDOC1100387581?currentPartNo=k004&togo=content&section=k004

I would love to get all the readings!

wlcrs, 2025-05-24:
Please test the code in PR #1023 and let me know if it works or if it doesn't. Please include (error) logs if you encounter problems.

PR #1023 (2025-06-30): Add EMMA power meter sensors
Fixes #955 . Both internal and external power meter sensors have been added. As I don't have any way to detect which one is active, I went with the following approach:
- all external power meter sensors are disabled by default.
- only the most important internal power meter sensors are enabled.
- 
This code is untested as I don't own/have access to an EMMA installation.

I would be grateful if somebody would test this code and let me know if there are issues. Please also mention what type of power meter your setup uses, as I want to have covered both type of instal [... cut]
closed 2025-06-30 by wlcrs