wlcrs/huawei_solar #958: [Bug]: Failed to read register 30000 after 6 tries

SvenAbels, 2025-02-26:
### Describe the issue

I got a SUN2000-10KTL connected via the sDongle with a LAN cable. I'm using firmware V100R001C00SPC156 and the version 1.5.4 of the Huawei integration. Everything worked well for months but since this afternoon, no conection can be established by the extension. I already restarted Home Assistant but without success. In can still pink the dongle with the local IP and the values show up fine in fusionsolar.

When looking at the extension page, I can see the following error: "Failed to read register 30000 after 6 tries"

[image]

### Describe your Huawei Solar Setup

Inverter Type:  SUN2000-10KTL
Inverter Firmware version:  V100R001C00SPC156 
sDongle present: Yes 
sDongle Type: sDongleA-05 (WiFi / Ethernet) 
sDongle Connectivitiy: Ethernet
sDongle Firmware: V200R022C10SPC108
Power meter present: three phase 
Optimizers Present: No
Battery: LUNA2000
Battery Firmware version:  V100R002C00SPC123 
Huawei Solar integration version:  1.5.4

### How do you connect to the inverter?

Via the SDongle, wired connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

[home-assistant_huawei_solar_2025-02-26T21-59-58.873Z.log](https://github.com/user-attachments/files/18996138/home-assistant_huawei_solar_2025-02-26T21-59-58.873Z.log)

### Upload your relevant debug logs

```text
2025-02-26 22:48:27.958 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter)
2025-02-26 22:48:28.959 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter
NoneType: None
2025-02-26 22:48:28.960 INFO (MainThread) [backoff] Backing off _do_read(...) for 2.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter)
2025-02-26 22:48:30.962 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter
NoneType: None
2025-02-26 22:48:30.962 INFO (MainThread) [backoff] Backing off _do_read(...) for 4.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter)
2025-02-26 22:48:34.964 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter
NoneType: None
2025-02-26 22:48:34.964 INFO (MainThread) [backoff] Backing off _do_read(...) for 8.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter)
2025-02-26 22:48:42.966 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter
NoneType: None
2025-02-26 22:48:42.966 INFO (MainThread) [backoff] Backing off _do_read(...) for 16.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter)
2025-02-26 22:48:58.968 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter
NoneType: None
2025-02-26 22:48:58.968 ERROR (MainThread) [backoff] Giving up _do_read(...) after 6 tries (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter)
2025-02-26 22:49:10.560 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter
NoneType: None
2025-02-26 22:49:10.560 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter)
2025-02-26 22:49:11.561 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter
NoneType: None
2025-02-26 22:49:11.561 INFO (MainThread) [backoff] Backing off _do_read(...) for 2.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter)
2025-02-26 22:49:13.563 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter
NoneType: None
2025-02-26 22:49:13.563 INFO (MainThread) [backoff] Backing off _do_read(...) for 4.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter)
2025-02-26 22:49:17.564 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter
NoneType: None
2025-02-26 22:49:17.565 INFO (MainThread) [backoff] Backing off _do_read(...) for 8.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter)
2025-02-26 22:49:25.568 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter
NoneType: None
2025-02-26 22:49:25.569 INFO (MainThread) [backoff] Backing off _do_read(...) for 16.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter)
2025-02-26 22:49:41.570 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter
NoneType: None
2025-02-26 22:49:41.571 ERROR (MainThread) [backoff] Giving up _do_read(...) after 6 tries (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter)
2025-02-26 22:50:03.346 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter
[... 132 lines cut ...]
2025-02-26 23:00:39.725 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter)
2025-02-26 23:00:40.726 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter
NoneType: None
2025-02-26 23:00:40.726 INFO (MainThread) [backoff] Backing off _do_read(...) for 2.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter)
2025-02-26 23:00:42.728 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter
NoneType: None
2025-02-26 23:00:42.728 INFO (MainThread) [backoff] Backing off _do_read(...) for 4.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter)
2025-02-26 23:00:46.729 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter
NoneType: None
2025-02-26 23:00:46.730 INFO (MainThread) [backoff] Backing off _do_read(...) for 8.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter)
2025-02-26 23:00:54.738 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter
NoneType: None
2025-02-26 23:00:54.739 INFO (MainThread) [backoff] Backing off _do_read(...) for 16.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter)
2025-02-26 23:01:10.740 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter
NoneType: None
2025-02-26 23:01:10.741 ERROR (MainThread) [backoff] Giving up _do_read(...) after 6 tries (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter)
```

### Please confirm the following:

- [x] I'm running the latest release of Home Assistant.
- [x] I'm running the latest release of this integration.
- [x] I did not find an existing issue describing this problem.
- [x] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [x] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

SvenAbels, 2025-02-27:
After about 6 hours, this error has disappeared. I will monitor the situation and re-open the ticket if necessary.

closed 2025-02-27 by SvenAbels

wlcrs, 2025-02-27:
Hi Sven,

This was a connection problem: your inverter does not react to any request made by the integration. It did seem to be able to setup the actual TCP connection, but no actual traffic came back over it.

Please make sure that no other devices/services try to connect to the dongle. I would suggest that you also do a full power cycle of your inverter and SDongle if it happens again.

SvenAbels, 2025-02-27:
> This was a connection problem: your inverter does not react to any request made by the integration. It did seem to be able to setup the actual TCP connection, but no actual traffic came back over [...]

Thanks a lot for your help.  The connection is possible again now.

I will monitor this and if it happens again then I will perform a full power cycle as you suggest.   :-)

Best regards,

Sven