wlcrs/huawei_solar #960: [Bug]: Failed to Connect Huawei SUN2000 Inverter to Home Assistant using Modbus TCP

matt7py, 2025-02-27:
### Describe the issue

I have a Huawei SUN2000-4.5KTL-L1 inverter connected to a Huawei 10KW battery, a power sensor, and an SDongleA-05. 

I have activated unrestricted Modbus TCP mode in the inverter settings. 

My network has two IP addresses related to the inverter setup: 192.168.68.113, which I believe is for the inverter, and 192.168.68.133, which I believe is for the SDongleA-05 connected by cable to my router. 

My Home Assistant server is on the same network with IP 192.168.68.121.

I am trying to connect the inverter to Home Assistant using Modbus TCP on port 502 or 6607, but the connection fails with an error message. 

I need help identifying what is missing in my configuration to successfully connect the inverter to Home Assistant using the Huawei Solar integration.

[image]

[image]

### Describe your Huawei Solar Setup

Inverter Type: 
Inverter Firmware version: 
sDongle present: Yes / No
sDongle Type: sDongleA-05 (WiFi / Ethernet) / SDongleA-03 (4G)
sDongle Connectivitiy: WiFi / Ethernet / 4G
sDongle Firmware: 
Power meter present: three phase / single phase / no
Optimizers Present: Yes / No
Battery: LUNA2000-SO xxkWh / LUNA2000-S1 xxkWh (Released 2024) / LG RESU xxkWh / None
Battery Firmware version: 
Huawei Solar integration version: 

### How do you connect to the inverter?

Please select your connection method

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
.
```

### Please confirm the following:

- [ ] I'm running the latest release of Home Assistant.
- [ ] I'm running the latest release of this integration.
- [ ] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2025-02-27:
I cannot provide support for connectivity problems.

closed 2025-02-27 by wlcrs

HolgerMiara, 2025-02-27:
Have you tried other ports than 6607?

matt7py, 2025-02-28:
> Have you tried other ports than 6607?

Yes 502

AndyVRD, 2025-03-12:
You need port 502 with slave1 for network (LAN/WIFI) connected inverter (connected with sdonglea-05). Port 6607 with slave0 is for "ad-hoc" connection, meaning that you connected direct with the internal inverter AP with a WIFI-dongle connected to your Home Assistant instance (thin client, Server, EPI, ...).