wlcrs/huawei_solar #965: [Bug]: after update cannot instal

avick71, 2025-03-08:
### Describe the issue

Logger: homeassistant.util.package
Source: util/package.py:159
First occurred: 10:14:18 PM (3 occurrences)
Last logged: 10:14:28 PM

Unable to install package huawei-solar==2.4.4: error: Failed to fetch: `https://files.pythonhosted.org/packages/c1/f9/6c2fadb181492942cb0ab84163e726741cae39bd8fe76f069d3690ce5b2e/huawei_solar-2.4.4-py3-none-any.whl.metadata` Caused by: Could not connect, are you offline? Caused by: Request failed after 3 retries Caused by: error sending request for url (https://files.pythonhosted.org/packages/c1/f9/6c2fadb181492942cb0ab84163e726741cae39bd8fe76f069d3690ce5b2e/huawei_solar-2.4.4-py3-none-any.whl.metadata) Caused by: client error (Connect) Caused by: dns error: failed to lookup address information: Name does not resolve Caused by: failed to lookup address information: Name does not resolve

Logger: homeassistant.setup
Source: setup.py:275
First occurred: 10:14:28 PM (1 occurrences)
Last logged: 10:14:28 PM

Setup failed for custom integration 'huawei_solar': Requirements for huawei_solar not found: ['huawei-solar==2.4.4'].

### Describe your Huawei Solar Setup

Inverter Type: 
Inverter Firmware version: 
sDongle present: Yes / No
sDongle Type: sDongleA-05 (WiFi / Ethernet) / SDongleA-03 (4G)
sDongle Connectivitiy: WiFi / Ethernet / 4G
sDongle Firmware: 
Power meter present: three phase / single phase / no
Optimizers Present: Yes / No
Battery: LUNA2000-SO xxkWh / LUNA2000-S1 xxkWh (Released 2024) / LG RESU xxkWh / None
Battery Firmware version: 
Huawei Solar integration version: 

### How do you connect to the inverter?

Please select your connection method

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
Logger: homeassistant.setup
Source: setup.py:275
First occurred: 10:14:28 PM (1 occurrences)
Last logged: 10:14:28 PM

Setup failed for custom integration 'huawei_solar': Requirements for huawei_solar not found: ['huawei-solar==2.4.4'].

Logger: homeassistant.util.package
Source: util/package.py:159
First occurred: 10:14:18 PM (3 occurrences)
Last logged: 10:14:28 PM

Unable to install package huawei-solar==2.4.4: error: Failed to fetch: `https://files.pythonhosted.org/packages/c1/f9/6c2fadb181492942cb0ab84163e726741cae39bd8fe76f069d3690ce5b2e/huawei_solar-2.4.4-py3-none-any.whl.metadata` Caused by: Could not connect, are you offline? Caused by: Request failed after 3 retries Caused by: error sending request for url (https://files.pythonhosted.org/packages/c1/f9/6c2fadb181492942cb0ab84163e726741cae39bd8fe76f069d3690ce5b2e/huawei_solar-2.4.4-py3-none-any.whl.metadata) Caused by: client error (Connect) Caused by: dns error: failed to lookup address information: Name does not resolve Caused by: failed to lookup address information: Name does not resolve
```

### Please confirm the following:

- [ ] I'm running the latest release of Home Assistant.
- [ ] I'm running the latest release of this integration.
- [ ] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

pavelekg, 2025-03-09:
Had the same issue, with huawei_solar and frigate, check this solution - worked for me:

https://github.com/home-assistant/core/issues/135054

wlcrs, 2025-03-10:
The errors indicate general connectivity problems to the internet. Nothing to do here.

closed 2025-03-10 by wlcrs