wlcrs/huawei_solar #974: [Bug]: hybrid inverter curtailing when settings forcible charge or discharge or 0w

scruysberghs, 2025-03-22:
### Describe the issue

I have a SUN2000-10KTL-M1 hybrid inverter with a 30kwh Huawei battery attached.
I control the battery from home assistant using the forcible charge or discharge action. And set forcible charge to 0 w when I want the battery to do nothing. If I want to have the battery follow grid I send the action: huawei_solar.stop_forcible_charge en set mode to maximise self consumption. This worked fine until recently.

What I notice since the latest release(?) that when I set a forcible chrage or discharge power (action: huawei_solar.forcible_charge or action: huawei_solar.forcible_discharge ) solar production is limited as wel. 
This means I have to send a action: huawei_solar.set_maximum_feed_grid_power 10000 or action: huawei_solar.set_maximum_feed_grid_power_percent 100 to the inverter every time I send a forcible charge command to the batteries.

Anyone else notice this? 

### Describe your Huawei Solar Setup

Inverter Type: SUN2000-10KTL-M1
Inverter Firmware version: 
sDongle present: Yes / No
sDongle Type: sDongleA-05 (WiFi / Ethernet) / SDongleA-03 (4G)
sDongle Connectivitiy: WiFi / Ethernet / 4G
sDongle Firmware: 
Power meter present: three phase / single phase / no
Optimizers Present: Yes / No
Battery: LUNA2000-SO xxkWh / LUNA2000-S1 xxkWh (Released 2024) / LG RESU xxkWh / None
Battery Firmware version: 
Huawei Solar integration version: 1.5.4

### How do you connect to the inverter?

Please select your connection method

### Upload your Diagnostics File

Drag & Drop your Diagn

[config_entry-huawei_solar-01JBHBH9YJPS497HTMW9RCJ09B.json](https://github.com/user-attachments/files/19404460/config_entry-huawei_solar-01JBHBH9YJPS497HTMW9RCJ09B.json)

ostics File here.

### Upload your relevant debug logs

```text
can't test right as no solar but will upload when I get the chance
```

### Please confirm the following:

- [x] I'm running the latest release of Home Assistant.
- [x] I'm running the latest release of this integration.
- [x] I did not find an existing issue describing this problem.
- [x] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [x] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

scruysberghs, 2025-03-23:
Addng two images. Solar production now is 10kW. There are 2 inverters.
If I set de the hybrid inverter to max self consuption everything is ok, first screenshot.
Second screenshot I send a forcible charge charge command to the battery connected to the 10k inverter but that seems to limit the inverter to the power that is sent to the battery and it looks like part of the solar production is dropped. Or am I looking at the wrong values here?

[image]

[image]

scruysberghs, 2025-03-25:
Ok, so probably not a bug but intended behaviour of the sun2000 inverter. Looked a bit further and this curtailing of hybrid inverters when using forcible charge/disharge commands is expected. Using TOU mode in combination with battery_maximum_charging_power would be a better way to set battery charging without curtailment happening. Thijs (@wlcrs ), can you confirm this or did you work around this in the forcible (dis)charge actions?

Still find it weird that it wasn't happening before on my my system while still using the same automations to control the battery.

wlcrs, 2025-03-26:
I'm not an expert in such matter. My knowledge is limited to reading and writing to the modbus registers as published in the technical documentation by Huawei. 

Please don't abuse the issue tracker for support requests like this. I'm moving this to a discussion instead.