wlcrs/huawei_solar #985: [Bug]: integration stopped working: device not loaded ( SUN2000-2KTL-L1 )

peterdedecker83, 2025-04-07:
### Describe the issue

Hi,

The integration worked fine for several days. After a reboot of the inverter caused by recabling the AC-side, it stopped working. No settings were altered in the interver. Log is in attach.

The inverter still works fine in terms of both AC power generation as in connectivity and reporting through the FusionSolar webapp.

I tried reconfiguring the integration a few times, keeping all settings as they were. Installer password works fine when I use the app to connect.

[home-assistant_huawei_solar_2025-04-07T22-05-20.347Z.log](https://github.com/user-attachments/files/19639782/home-assistant_huawei_solar_2025-04-07T22-05-20.347Z.log)

### Describe your Huawei Solar Setup

Inverter Type: SUN2000-2KTL-L1
Inverter Firmware version: V200R001C00SPC146
Inverter Protocol version: P1.15-D5.0
sDongle present: No seperate dongle, just using built-in WiFi modem on my LAN which worked fine
sDongle Connectivitiy: WiFi
sDongle Firmware: ?
Power meter present: no
Optimizers Present: No
Battery: None

### How do you connect to the inverter?

Via the SDongle, wireless connection

### Upload your Diagnostics File

Unfortunately no diagnostics:

[image]

through the link:

[image]

### Upload your relevant debug logs

```text
See attach in main message
```

### Please confirm the following:

- [x] I'm running the latest release of Home Assistant.
- [x] I'm running the latest release of this integration.
- [x] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [x] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

Schnabulation, 2025-04-08:
Have you by any chance upgraded to Home Assistant 2025.4? It seems that the integration is not working properly with this version. Might be related to ModBus again.

peterdedecker83, 2025-04-08:
Indeed, version 2025.4.1 here. I think it worked before in the 2025.4 version, but not sure how to check, can't remember when I did the upgrade, would have to check logs which are somewhat bloated.

peterdedecker83, 2025-04-08:
Ok, I could track down my upgrade cycle through the backups.

[image]

[image]

kbrelins, 2025-04-09:
Hello, i can confirm the same issue. In my case, integration stopped working at the 3:36 (CET) 8.04.2025. I have to HA version 2025.4.1 which I installed 5.04.2025 at 12:54.

wlcrs, 2025-04-10:
Your device stops responding to queries after an initial few do get an answer. Do you have another system connecting to it?

Please provide 3 more log files of this problem where you first reboot the inverter (switch off ac and DC side for a few seconds), wait for 10 minutes to let the inverter start up and then restart the integration. This will let me check if the issue occurs at exactly the same moment each time.

kbrelins, 2025-04-11:
My integration started working on 10.04.2025 at 12:41 (CET), without any additional steps.

peterdedecker83, 2025-04-11:
Some more retries and a much longer log file.

Note that I'm also using the Solarman HACS integration for another inverter, so the pymodbus might also be for that one, but I guess you can distinguish that based on the registers logged by the huawei-solar module.

[home-assistant_huawei_solar_2025-04-11T21-32-02.280Z.log](https://github.com/user-attachments/files/19714303/home-assistant_huawei_solar_2025-04-11T21-32-02.280Z.log)

peterdedecker83, 2025-04-11:
OK, I got it working again.

I tried to perform a firmware upgrade on the inverter through the Fusion Solar app. Connected to the inverter WiFi to login, got prompted to set the installer password again. Weird, it must somewhat have forgotten the configured password from a few weeks ago during commissioning. Filling in that installer password resulted in an unknown error. After a few tries, I entered a new installer password, that got accepted. Afterwards, in the config, I could set that installer password back to the one it was before my involuntary change, reloaded the integration and that did the trick: working fine again. Weird the inverter got somewhat lost the installer password and didn't accept the one that was set...

Even though: it would be good to catch that kind of exceptions with a decent error message as well :-)

Thanks for the help and the great integration!

closed 2025-04-12 by wlcrs