wlcrs/huawei_solar #993: [Bug]:Unexpected error in installer login

AnKloss, 2025-04-19:
### Describe the issue

Hello,
sins version 1.5.x, i can't not more login to huawei grid inverter.
Today I update to 1.5.4 , but still the same Unexpected error in the installaller login.
- i open the app (old FusionHome app) on phone and see inverter is status : **Accessible**.
- I test login as installer and get in see all things, is ok and logout.
- open Huawei in HA , select Network ( connection over wifi) put in IP , select Advanced.
- take 3-5 sec, then Installer window pop up
- select installer and put password in. (click on see password ), all ok from input data. click submit
- on the phone app is show then "mobile phone connected"
- then on HA stop and show "Unexpected error" and staus on phone app go back to :"Accessible"

Note:
- until 1.4.1 is working, now HA is 15.x ,  1.4.1 not work any more.(Handel error)
- not any firmware update sins 5 years on this Interver. Huawei not support this type any more.

Inverter Type: Grid Inverter 5Kw
Inverter Firmware version: ???? Grid inverter is 5 years old, 
sDongle present:  No
Connectivitiy: WiFi n 
Power meter present: single phase 
Optimizers Present: No
Battery: None
Huawei Solar integration version: ????

### Describe your Huawei Solar Setup

Inverter Type: 
Inverter Firmware version: 
sDongle present: Yes / No
sDongle Type: sDongleA-05 (WiFi / Ethernet) / SDongleA-03 (4G)
sDongle Connectivitiy: WiFi / Ethernet / 4G
sDongle Firmware: 
Power meter present: three phase / single phase / no
Optimizers Present: Yes / No
Battery: LUNA2000-SO xxkWh / LUNA2000-S1 xxkWh (Released 2024) / LG RESU xxkWh / None
Battery Firmware version: 
Huawei Solar integration version: 

### How do you connect to the inverter?

Please select your connection method

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
Logger: custom_components.huawei_solar.config_flow
Source: custom_components/huawei_solar/config_flow.py:620
integration: Huawei Solar (documentation, issues)
First occurred: 10:01:45 (10 occurrences)
Last logged: 10:33:00

Unexpected exception while validating login parameters via ModbusTCP
Traceback (most recent call last):
  File "/config/custom_components/huawei_solar/config_flow.py", line 620, in async_step_network_login
    return await self._create_or_update_entry(self._inverter_info)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/huawei_solar/config_flow.py", line 680, in _create_or_update_entry
    self._abort_if_unique_id_configured(updates=data)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2967, in _abort_if_unique_id_configured
    raise data_entry_flow.AbortFlow(error)
homeassistant.data_entry_flow.AbortFlow: Flow aborted: already_configured
```

### Please confirm the following:

- [x] I'm running the latest release of Home Assistant.
- [x] I'm running the latest release of this integration.
- [x] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

fmjensen, 2025-04-19:
I have similar problem.

Fisoft83, 2025-04-24:
hello, same issue for me :-(

Criaga, 2025-04-26:
same issue for me

@wlcrs

eddieari, 2025-04-26:
Hi, i downgrade the ha core to 2025.4.2. This solved the problem on my system, may you try this.
`ha core update --version 2025.4.2`

AnKloss, 2025-04-29:
This error originated from a custom integration.

Logger: custom_components.huawei_solar.config_flow
Source: custom_components/huawei_solar/config_flow.py:620
integration: Huawei Solar (documentation, issues)
First occurred: 08:37:22 (2 occurrences)
Last logged: 08:37:48

Unexpected exception while validating login parameters via ModbusTCP
Traceback (most recent call last):
  File "/config/custom_components/huawei_solar/config_flow.py", line 620, in async_step_network_login
    return await self._create_or_update_entry(self._inverter_info)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/huawei_solar/config_flow.py", line 680, in _create_or_update_entry
    self._abort_if_unique_id_configured(updates=data)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2967, in _abort_if_unique_id_configured
    raise data_entry_flow.AbortFlow(error)
homeassistant.data_entry_flow.AbortFlow: Flow aborted: already_configured

AnKloss, 2025-04-29:
install on a new pi5 with last ha , no other run.  ModusTCP
Please fix it, thanks

felixhohlstein, 2025-05-18:
Ich habe auch ein ähliches Problem

Vincenzopica, 2025-05-25:
Ho lo stesso problema anche io

deepminds83, 2025-06-10:
Hello.
The problem seems to persist.
I have a new HA installation and the latest Fusion Solar version on my dongle and I can't connect.

This problem seems to have existed for some time. Is a solution currently being worked on?

gyldenro, 2025-06-29:
> Hi, i reinstall the package and it's Ok now. Thz
> 
> Le sam. 26 avr. 2025, 09:23, Criaga ***@***.***> a écrit :
> […](#)

I must admit i was skeptical, but reinstalling the integration in hacs got my setup running again. Does not really make sence to me. But as they say "if it works its not stupid"