wlcrs/huawei_solar #995: [Feature Request]: Setup a battery buffer to avoid losing PV input power

GiulioBen, 2025-04-21:
### Describe your feature request

Hello,
 I have
15kWp panels
10kW luna inverter 
30kWh batteries 

When the production Is high and the SOC Is low, i can use all the Power fron the PV (a part goes to batteries and the rest to grid/house)

When the battery Is charged the PV production Is capped to 10kw.

Isn't It possibile to set a buffer to avoid this waste? e.g. max SOC 60% but with the possibility to go up to 90% to buffer the input exceeding the 10kW of the inverter. Once the production decreases the SOC Will return to 60 exporting at 10kw..

Thank you for your suggestions.

Best regards

### Proper usage

- [x] I confirm that this is not a bug report or support request
- [x] I confirm that this feature request is within the stated scope of the integration

tudor0410, 2025-04-21:
You can use TOU for this.
Just setup in fusion solar one single discharge period from 00:00 to 23:59 and set charge from grid limit to 0.
Then use TOU with the 2 available modes.
Excess pv set to Charge or set to Fed to grid.
In the morning, set it to Fed to grid and it will only feed power to grid, no charging. When power când be fed to grid due to high grid voltage or the maximum output power of the invertor, it will charge the battery.
When you want to charge the battery again with all the power, just switch to charge.
Perhaps you can set a time based automation to enable fed to grid in the morning(not charging the battery early) and switch to charge after the input power is less than a threshold.

wlcrs, 2025-04-21:
Not in scope for this integration.

closed 2025-04-21 by wlcrs