wlcrs/huawei_solar #996: [Bug]: Optimizers not working with Modbus RTU to TCP Gateway

JannikBirn, 2025-04-21:
### Describe the issue

I used this integration together with a “USB to RS485 converter” successfully. But now I need to use the Converter for another connection to another slave (HA modbus integration). To do this, I tried two Modbus Gateways [mbusd ](https://github.com/3cky/mbusd) and [ModbusGW ](https://github.com/EHerzog76/ModbusGW) both ending in the same result with working connections and data retrieval except for the optimizers.

I reconfigured the Huawei Solar integration to work with the Network connection type and checked the **Advanced: elevate permissions** checkbox. I receive the inverter, battery, and smart meter data but not the data of the optimizers that I previously were able to receive with the serial connection.
Note: I didn't receive a popup to enter my installer credentials after setup, and I made sure Advanced: elevate permission was checked twice. I accepted this because I changed my installer credentials.

I also tried deleting the integration and reinstalling it with the same result, also the optimizer entities weren't created that time.

So why is that the Optimizer data is not retrieved, and I didn't need to enter my installer credentials?
(Not sure which part of the logs is relevant, but I found one with `nb_online_optimizers` and copied a snipped from that.)
Happy to provide more info if necessary.

### Describe your Huawei Solar Setup

Inverter Type: Sun200-8KTL-M1
Inverter Firmware version: V100R001C00SPC168
sDongle present: No
sDongle Type:
sDongle Connectivitiy:
sDongle Firmware: 
Power meter present: three phase
Optimizers Present: Yes / No
Battery: LUNA2000-SO 15kWh
Battery Firmware version: V100R002C00SPC624
Huawei Solar integration version: 

### How do you connect to the inverter?

Over serial, with a serial-to-USB stick

### Upload your Diagnostics File

[config_entry-huawei_solar-01J60EACZ38YWAYF9VG60N0QZP.json](https://github.com/user-attachments/files/19838620/config_entry-huawei_solar-01J60EACZ38YWAYF9VG60N0QZP.json)
Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
2025-04-21 20:36:33.477 DEBUG (MainThread) [huawei_solar.huawei_solar] Sleeping for 0.049703 seconds before making next call
2025-04-21 20:36:33.527 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064 with length 52 from slave 1
2025-04-21 20:36:33.527 DEBUG (MainThread) [pymodbus.logging] send: 0xd 0x3a 0x0 0x0 0x0 0x6 0x1 0x3 0x7d 0x40 0x0 0x34
2025-04-21 20:36:33.905 DEBUG (MainThread) [pymodbus.logging] recv: 0xd 0x3a 0x0 0x0 0x0 0x6b 0x1 0x3 0x68 0x0 0x0 0x0 0x0 0xf 0xa8 0xf 0xa6 0xf 0xa2 0x9 0xa 0x9 0x1 0x9 0xf 0x0 0x0 0x3 0x93 0x0 0x0 0x3 0x83 0x0 0x0 0x3 0x94 0x0 0x0 0xf 0x51 0x0 0x0 0x2 0x63 0x0 0x0 0x0 0x1 0x3 0xe8 0x13 0x88 0x27 0x10 0x1 0x49 0xff 0xff 0x2 0x0 0x0 0x0 0x68 0x5 0xf3 0x19 0xff 0xff 0xff 0xff 0x0 0x0 0x2 0x75 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x8 0x0 0x0 0x7 0x27 0x0 0x3 0xcb 0x47 0x0 0x4 0x32 0x45 0x68 0x6 0xb6 0xda 0x0 0x0 0x0 0xc 0x0 0x0 0x3 0xb9 old_data:  addr=None
2025-04-21 20:36:33.905 DEBUG (MainThread) [pymodbus.logging] Processing: 0xd 0x3a 0x0 0x0 0x0 0x6b 0x1 0x3 0x68 0x0 0x0 0x0 0x0 0xf 0xa8 0xf 0xa6 0xf 0xa2 0x9 0xa 0x9 0x1 0x9 0xf 0x0 0x0 0x3 0x93 0x0 0x0 0x3 0x83 0x0 0x0 0x3 0x94 0x0 0x0 0xf 0x51 0x0 0x0 0x2 0x63 0x0 0x0 0x0 0x1 0x3 0xe8 0x13 0x88 0x27 0x10 0x1 0x49 0xff 0xff 0x2 0x0 0x0 0x0 0x68 0x5 0xf3 0x19 0xff 0xff 0xff 0xff 0x0 0x0 0x2 0x75 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x8 0x0 0x0 0x7 0x27 0x0 0x3 0xcb 0x47 0x0 0x4 0x32 0x45 0x68 0x6 0xb6 0xda 0x0 0x0 0x0 0xc 0x0 0x0 0x3 0xb9
2025-04-21 20:36:33.905 DEBUG (MainThread) [pymodbus.logging] decoded PDU function_code(3 sub -1) -> ReadHoldingRegistersResponse(dev_id=0, transaction_id=0, address=0, count=0, bits=[], registers=[0, 0, 4008, 4006, 4002, 2314, 2305, 2319, 0, 915, 0, 899, 0, 916, 0, 3921, 0, 611, 0, 1, 1000, 5000, 10000, 329, 65535, 512, 0, 26629, 62233, 65535, 65535, 0, 629, 0, 0, 0, 0, 0, 0, 8, 0, 1831, 3, 52039, 4, 12869, 26630, 46810, 0, 12, 0, 953], status=1) 
2025-04-21 20:36:33.905 DEBUG (MainThread) [pymodbus.logging] Frame advanced, resetting header!!
2025-04-21 20:36:33.905 DEBUG (MainThread) [huawei_solar.bridge] Batch update of the following registers: nb_online_optimizers
2025-04-21 20:36:33.905 DEBUG (MainThread) [huawei_solar.huawei_solar] Sleeping for 0.049701 seconds before making next call
2025-04-21 20:36:33.956 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37201 with length 1 from slave 1
2025-04-21 20:36:33.956 DEBUG (MainThread) [pymodbus.logging] send: 0xd 0x3b 0x0 0x0 0x0 0x6 0x1 0x3 0x91 0x51 0x0 0x1
2025-04-21 20:36:33.997 DEBUG (MainThread) [pymodbus.logging] recv: 0xd 0x3b 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x0 old_data:  addr=None
2025-04-21 20:36:33.997 DEBUG (MainThread) [pymodbus.logging] Processing: 0xd 0x3b 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x0
2025-04-21 20:36:33.997 DEBUG (MainThread) [pymodbus.logging] decoded PDU function_code(3 sub -1) -> ReadHoldingRegistersResponse(dev_id=0, transaction_id=0, address=0, count=0, bits=[], registers=[0], status=1) 
2025-04-21 20:36:33.997 DEBUG (MainThread) [pymodbus.logging] Frame advanced, resetting header!!
2025-04-21 20:36:38.197 DEBUG (MainThread) [huawei_solar.bridge] Batch update of the following registers: power_meter_active_power, active_grid_power_factor, active_grid_frequency, grid_exported_energy, grid_accumulated_energy, active_grid_A_power, active_grid_B_power, active_grid_C_power
```

### Please confirm the following:

- [x] I'm running the latest release of Home Assistant.
- [x] I'm running the latest release of this integration.
- [x] I did not find an existing issue describing this problem.
- [x] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [x] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2025-04-23:
installer credentials are only necessary when using the Modbus TCP interface of your inverter. When connected via serial, you already have elevated permissions and no password is necessary.

Optimizer information should also be available over Modbus RTU. The small amount of logs you posted provide no information on why they are missing. Please upload logs showing the full initialization process of the integration.

JannikBirn, 2025-04-23:
Thanks for your reply, that makes perfectly sense. I also missed to mention that im using the newest version of the Huawei Solar integration v1.5.4
I disabled my modbus integration, so the logs only contain information about this integration.
There is indeed an error when starting the integration:

```
[36m2025-04-23 21:46:46.756 DEBUG (MainThread) [pymodbus.logging] Connecting to 192.168.178.47:502.[0m
[36m2025-04-23 21:46:46.756 DEBUG (MainThread) [pymodbus.logging] Connecting comm[0m
[36m2025-04-23 21:46:46.758 DEBUG (MainThread) [pymodbus.logging] Connected to comm[0m
[36m2025-04-23 21:46:46.758 DEBUG (MainThread) [huawei_solar.modbus] Waiting for 1500 milliseconds after connection before performing operations[0m
[32m2025-04-23 21:46:46.859 INFO (MainThread) [huawei_solar.huawei_solar] Waiting for connection[0m
[36m2025-04-23 21:46:48.260 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 30000 with length 65 from slave 1[0m
[36m2025-04-23 21:46:48.260 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x75 0x30 0x0 0x41[0m
[36m2025-04-23 21:46:48.720 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x1 0x0 0x0 0x0 0x85 0x1 0x3 0x82 0x53 0x55 0x4e 0x32 0x30 0x30 0x30 0x2d 0x38 0x4b 0x54 0x4c 0x2d 0x4d 0x31 0x0 0x30 0x31 0x30 0x37 0x34 0x33 0x31 0x34 0x2d 0x30 0x30 0x36 0x0 0x0 0x42 0x54 0x32 0x33 0x32 0x30 0x39 0x39 0x37 0x32 0x38 0x34 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x31 0x30 0x37 0x34 0x33 0x31 0x34 0x2d 0x30 0x30 0x36 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x56 0x31 0x30 0x30 0x52 0x30 0x30 0x31 0x2d 0x30 0x32 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x56 0x31 0x30 0x30 0x52 0x30 0x30 0x31 0x43 0x30 0x30 0x53 0x50 0x43 0x31 0x36 0x38 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 old_data:  addr=None[0m
[36m2025-04-23 21:46:48.720 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x1 0x0 0x0 0x0 0x85 0x1 0x3 0x82 0x53 0x55 0x4e 0x32 0x30 0x30 0x30 0x2d 0x38 0x4b 0x54 0x4c 0x2d 0x4d 0x31 0x0 0x30 0x31 0x30 0x37 0x34 0x33 0x31 0x34 0x2d 0x30 0x30 0x36 0x0 0x0 0x42 0x54 0x32 0x33 0x32 0x30 0x39 0x39 0x37 0x32 0x38 0x34 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x31 0x30 0x37 0x34 0x33 0x31 0x34 0x2d 0x30 0x30 0x36 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x56 0x31 0x30 0x30 0x52 0x30 0x30 0x31 0x2d 0x30 0x32 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x56 0x31 0x30 0x30 0x52 0x30 0x30 0x31 0x43 0x30 0x30 0x53 0x50 0x43 0x31 0x36 0x38 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0[0m
[36m2025-04-23 21:46:48.720 DEBUG (MainThread) [pymodbus.logging] decoded PDU function_code(3 sub -1) -> ReadHoldingRegistersResponse(dev_id=0, transaction_id=0, address=0, count=0, bits=[], registers=[21333, 20018, 12336, 12333, 14411, 21580, 11597, 12544, 12337, 12343, 13363, 12596, 11568, 12342, 0, 16980, 12851, 12848, 14649, 14130, 14388, 0, 0, 0, 0, 12337, 12343, 13363, 12596, 11568, 12342, 0, 0, 0, 0, 22065, 12336, 21040, 12337, 11568, 12800, 0, 0, 0, 0, 0, 0, 0, 0, 0, 22065, 12336, 21040, 12337, 17200, 12371, 20547, 12598, 14336, 0, 0, 0, 0, 0, 0], status=1) [0m
[36m2025-04-23 21:46:48.720 DEBUG (MainThread) [pymodbus.logging] Frame advanced, resetting header!![0m
[36m2025-04-23 21:46:48.720 DEBUG (MainThread) [huawei_solar.huawei_solar] Sleeping for 0.049753 seconds before making next call[0m
[36m2025-04-23 21:46:48.771 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 30071 with length 1 from slave 1[0m
[36m2025-04-23 21:46:48.771 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x2 0x0 0x0 0x0 0x6 0x1 0x3 0x75 0x77 0x0 0x1[0m
[36m2025-04-23 21:46:48.812 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x2 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x2 old_data:  addr=None[0m
[36m2025-04-23 21:46:48.812 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x2 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x2[0m
[36m2025-04-23 21:46:48.813 DEBUG (MainThread) [pymodbus.logging] decoded PDU function_code(3 sub -1) -> ReadHoldingRegistersResponse(dev_id=0, transaction_id=0, address=0, count=0, bits=[], registers=[2], status=1) [0m
[36m2025-04-23 21:46:48.813 DEBUG (MainThread) [pymodbus.logging] Frame advanced, resetting header!![0m
[36m2025-04-23 21:46:48.813 DEBUG (MainThread) [huawei_solar.huawei_solar] Sleeping for 0.049757 seconds before making next call[0m
[36m2025-04-23 21:46:48.863 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37200 with length 1 from slave 1[0m
[36m2025-04-23 21:46:48.863 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x3 0x0 0x0 0x0 0x6 0x1 0x3 0x91 0x50 0x0 0x1[0m
[36m2025-04-23 21:46:48.904 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x3 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x19 old_data:  addr=None[0m
[36m2025-04-23 21:46:48.904 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x3 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x19[0m
[36m2025-04-23 21:46:48.904 DEBUG (MainThread) [pymodbus.logging] decoded PDU function_code(3 sub -1) -> ReadHoldingRegistersResponse(dev_id=0, transaction_id=0, address=0, count=0, bits=[], registers=[25], status=1) [0m
[36m2025-04-23 21:46:48.904 DEBUG (MainThread) [pymodbus.logging] Frame advanced, resetting header!![0m
[36m2025-04-23 21:46:48.905 DEBUG (MainThread) [huawei_solar.huawei_solar] Sleeping for 0.049818 seconds before making next call[0m
[36m2025-04-23 21:46:48.955 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47000 with length 1 from slave 1[0m
[... cut ...] 5c 0x11 0x0 0x0 0x4 0x93 0x0 0x0 0x2 0xf3[0m
[36m2025-04-23 21:47:33.955 DEBUG (MainThread) [pymodbus.logging] decoded PDU function_code(3 sub -1) -> ReadHoldingRegistersResponse(dev_id=0, transaction_id=0, address=0, count=0, bits=[], registers=[850, 0, 2, 7677, 65529, 65535, 65045, 0, 15000, 0, 15000, 0, 15000, 0, 15000, 0, 0, 0, 0, 0, 1, 25130, 1, 23569, 0, 1171, 0, 755], status=1) [0m
[36m2025-04-23 21:47:33.955 DEBUG (MainThread) [pymodbus.logging] Frame advanced, resetting header!![0m

```

JannikBirn, 2025-04-23:
Just to be sure, I tested the setup process again and got the same exception.
I hope it is an easy fix, I'm happy to test anything when necessary. ^^

wlcrs, 2025-04-24:
I don't think that the error is related to the issue that you are describing here.

The integration fails to read this register:

[image]

For the moment, I can only say that your current setup seems unstable as such a basic call should not fail.

wlcrs, 2025-04-25:
The logs of your ModbusGateway provide the answer: Your Modbus-RTU-to-TCP gateway is mangling the response when processing the reply of a request of the custom "Uploading Files" format used by Huawei. cfr. section 6.3.7.1 in the Solar Inverter Modbus Interface Definitions.

TCP recv: `41 06 03 45 0000`
--> read frame 0x0000 of file 45

Serial write: `01 41 06 03 45 0000 9d40`
--> same request with RTU preamble and CRC suffix

Serial-TCP resp: `01 41 06 f9 45 0000 56313033086c0b5a000000000019000100000101ffff4a 5632`
--> inverter responds with data

TCP resp with failure: `11`
--> gateway returns error instead of response

Relevant sections from the documentation:

[image]

[image]

Closing this as it's not an issue with the integration.

closed 2025-04-25 by wlcrs

JannikBirn, 2025-05-04:
Thanks for the detailed answer.
I was able to create my own Modbus RTU - TCP Gateway using pymodbus and adding the custom function 0x41 to just pass between the RTU Client and TCP Server.
Here is the source code for anyone with the same problem: https://github.com/JannikBirn/PyModbusGateway