wlcrs/huawei_solar #999: [Bug]: Removing replaced optimizers

luzik, 2025-04-23:
### Describe the issue

I had to replace few of my optimizers. I don't see a way to remove old ones from integration. Please support manual removing them in HA. Without re-adding all again.

### Describe your Huawei Solar Setup

SUN2000-5KTL-M1GIN300-5KTL-M1

### How do you connect to the inverter?

Via the SDongle, wired connection

### Upload your Diagnostics File

[config_entry-huawei_solar-100850ce110ff7a4f10558f7b6d930d1.json](https://github.com/user-attachments/files/19863462/config_entry-huawei_solar-100850ce110ff7a4f10558f7b6d930d1.json)

### Upload your relevant debug logs

```text
No logs
```

### Please confirm the following:

- [x] I'm running the latest release of Home Assistant.
- [x] I'm running the latest release of this integration.
- [x] I did not find an existing issue describing this problem.
- [x] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [x] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2025-04-23:
If your inverter stops reporting about the existence of an optimizer, then this integration will also no longer provide information to HA about that optimizer. It will thus become "unavailable" in HA, and you will be able to remove it.

closed 2025-04-23 by wlcrs

luzik, 2025-04-23:
Fusion app reports 15 of them, Huawei_solar also reports 15 active. All others (extra 9 more) are ghosts. I can't find anything about them in FusionSolar. I believe that you can see them somehow via modbus. All sensors in one of old optimizer device HA view are unavailable, but whole device nor single sensors can't be removed. Do you know a way to remove those via FusionApp ?

Please rethink - if huawei_solar can see optimizer object, but all sensors are unavailable, then maybe whole optimizer device should be unavailable/removable also ? For example if any of my zigbee/z-wave device is offline I can remove it. In my example those 9 extra old/replaced optimizers are gone for about two years now, so that won't happened it self.