wlcrs/huawei_solar #1006: [Bug]: status became unknown

TopoAS, 2025-04-26:
### Describe the issue

At interval of a few hours the status of all devices in huawei integration became unknown.  This started after upgrading homeassista to 2024.4.3 version.
Please any help will be appreciated 

### Describe your Huawei Solar Setup

Inverter Type: huawei 5ktl-l1
Inverter Firmware version: V200R001C00SPC148
sDongle present: Yes 
sDongle Type: sDongleA-05 (WiFi / Ethernet) 
sDongle Connectivitiy: WiFi 
sDongle Firmware: 
Power meter present: single phase 
Optimizers Present: Yes 
Battery: LUNA2000-SO xxkWh / LUNA2000-S1
Battery Firmware version: V100R002C00SPC629
Huawei Solar integration version: 1.5.4

### How do you connect to the inverter?

Via the `SUN2000-<serial_no>` WiFi

### Upload your Diagnostics File

[error_log-2.txt](https://github.com/user-attachments/files/19922681/error_log-2.txt)
Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
2025-04-25 22:33:46.537 ERROR (MainThread) [custom_components.huawei_solar.update_coordinator] Unexpected error fetching TA22B0100235_optimizer_data_update_coordinator data
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 380, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/huawei_solar/update_coordinator.py", line 93, in _async_update_data
    return await self.bridge.get_latest_optimizer_history_data()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/bridge.py", line 578, in get_latest_optimizer_history_data
    end_time = (await self.client.get(rn.SYSTEM_TIME_RAW, self.slave_id)).value
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 252, in get
    return (await self.get_multiple([name], slave))[0]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 292, in get_multiple
    response = await self._read_registers(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<3 lines>...
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 429, in _read_registers
    return await _do_read()
           ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 377, in _do_read
    response = await self._client.read_holding_registers(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<3 lines>...
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 155, in execute
    raise ModbusIOException(txt)
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] No response received after 3 retries, continue with next request
```

### Please confirm the following:

- [x] I'm running the latest release of Home Assistant.
- [x] I'm running the latest release of this integration.
- [x] I did not find an existing issue describing this problem.
- [x] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [x] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2025-04-26:
This issue seems to be caused by a connection failure or an unstable connection to the inverter. This integration already has already implemented a lot of logic to automatically recover from connection failures or unstable connections. The three most important ones are:
- automatic reconnects if the connection to the inverter failed;
- each Modbus request is retried for up to 6 times, with increasing waits between each attempt to allow the connection to recover;
- after 3 failed Modbus requests the connection is reset to fix hanging connections.

Please understand that I cannot help you solve your connection failure or unstable connection. Consider [an alternative connection method](https://github.com/wlcrs/huawei_solar/wiki/Connecting-to-the-inverter).
If you are convinced that there is no connectivity issue, then I invite you to debug and fix the problem yourself and submit a pull request to the underlying [huawei-solar Python library](https://gitlab.com/Emilv2/huawei-solar). Your input would be greatly appreciated.

As there is nothing actionable for me to do in this issue, I'm closing it. You are welcome to open a topic in [Discussions: Connecting to the inverter](https://github.com/wlcrs/huawei_solar/discussions/categories/connecting-to-the-inverter) to get community support for your problem.

closed 2025-04-26 by wlcrs

koehntopp, 2025-05-15:
Same issue, since update to Sdongle firmware 210

Also mentioned in german PV forum by others:
https://www.photovoltaikforum.com/thread/187189-huawei-solar-integration-austausch-home-assistant/?postID=4287624&highlight=huawei%2Bfirmware#post4287624

I have disabled the integration for the time being, and switched to the evcc integration, which does not show the same issues (i.e. I have the SUN2000 configured in evcc via ModbusTCP, and consume the values from the evcc integration).