wlcrs/huawei_solar #1015: [Bug]: Some Parameters are gone

knothski2, 2025-05-10:
### Describe the issue

HI,

Some off the parameters are gone in HA, marked them with red in pictures. We are normal using them for management setup.

[image]

### Describe your Huawei Solar Setup

Inverter Type: SUN2000-3KTL-M1
Inverter Firmware version: V100R001C00SPC168
sDongle present: Yes 
sDongle Type: sDongleA-05 (WiFi / Ethernet)
sDongle Connectivitiy: WiFi
sDongle Firmware: V200R022C10SPC201
Power meter present: three phase
Optimizers Present: No
Battery: LUNA2000-5KW-C0 15KW total
Battery Firmware version: V100R002C00SPC633
Huawei Solar integration version: 1.5.4

### How do you connect to the inverter?

Via the SDongle, wireless connection

### Upload your Diagnostics File

[home-assistant_huawei_solar_2025-05-10T17-11-34.533Z.log](https://github.com/user-attachments/files/20140693/home-assistant_huawei_solar_2025-05-10T17-11-34.533Z.log)
Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2025.5.0",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.13.3",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Copenhagen",
    "os_name": "Linux",
    "os_version": "6.12.23-haos",
    "supervisor": "2025.04.1",
    "host_os": "Home Assistant OS 15.2",
    "docker_version": "28.0.4",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "hacs": {
      "documentation": "https://hacs.xyz/docs/use/",
      "version": "2.0.5",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "nordpool": {
      "documentation": "https://github.com/custom-components/nordpool/",
      "version": "0.0.17",
      "requirements": [
        "backoff"
      ]
    },
    "huawei_solar": {
      "documentation": "https://github.com/wlcrs/huawei_solar/wiki",
      "version": "1.5.4",
      "requirements": [
        "huawei-solar==2.4.4"
      ]
    },
    "remote_homeassistant": {
      "documentation": "https://github.com/custom-components/remote_homeassistant",
      "version": "4.5",
      "requirements": []
    },
    "energidataservice": {
      "documentation": "https://github.com/MTrab/energidataservice/blob/master/README.md",
      "version": "1.6.7",
      "requirements": [
        "async-retrying-ng==0.1.3",
        "CurrencyConverter==0.18.5"
      ]
    },
    "panasonic_cc": {
      "documentation": "https://github.com/sockless-coding/panasonic_cc/",
      "version": "2025.1.2",
      "requirements": [
        "aiohttp",
        "aio-panasonic-comfort-cloud==2025.1.2",
        "aioaquarea==0.7.2"
      ]
    },
    "wiheat": {
      "documentation": "https://github.com/kimjohnsson/wiheat",
      "version": "0.1.0",
      "requirements": []
    }
  },
  "integration_manifest": {
    "domain": "huawei_solar",
    "name": "Huawei Solar",
    "after_dependencies": [
      "usb"
    ],
    "config_flow": true,
    "documentation": "https://github.com/wlcrs/huawei_solar/wiki",
    "issue_tracker": "https://github.com/wlcrs/huawei_solar/issues",
    "requirements": [
      "huawei-solar==2.4.4"
    ],
    "codeowners": [
      "wlcrs"
    ],
    "iot_class": "local_polling",
    "version": "1.5.4",
    "loggers": [
      "huawei_solar",
      "pymodbus"
    ],
    "is_built_in": false,
    "overwrites_built_in": false
  },
  "setup_times": {
    "null": {
      "setup": 1.3899989426136017e-05
    },
    "01JP7VKRP18JS180CGXKJJSPVK": {
      "wait_import_platforms": -0.009259406942874193,
      "config_entry_setup": 3.988565810956061
    }
  },
  "data": {
    "config_entry_data": {
      "enable_parameter_configuration": false,
      "host": "10.1.8.57",
      "password": null,
      "port": 502,
      "slave_ids": [
        1
      ],
      "username": null
    },
    "pymodbus_version": "3.8.4",
    "slave_1": {
      "_type": "SUN2000",
      "model_name": "SUN2000-3KTL-M1\u0000V100R001C00B19",
      "firmware_version": "V100R001-02",
      "software_version": "V100R001C00SPC168",
      "pv_string_count": 2,
      "has_optimizers": 0,
      "battery_type": 2,
      "battery_1_type": 2,
      "battery_2_type": 0,
      "power_meter_type": 1,
      "supports_capacity_control": true
    },
    "slave_1_inverter_data": {
      "state_1": [
        [
          "Grid-Connected",
          "Grid-Connected normally"
        ],
        null
      ],
      "state_2": [
        [
          "Unlocked",
          "PV connected",
          "DSP data collection"
        ],
        null
      ],
      "state_3": [
        [
          "On-grid",
          "Off-grid switch disabled"
        ],
        null
      ],
      "alarm_1": [
        [],
        null
      ],
      "alarm_2": [
        [],
        null
      ],
      "alarm_3": [
        [],
        null
      ],
      "pv_01_voltage": [
        278.9,
        "V"
      ],
      "pv_01_current": [
        0.28,
        "A"
      ],
      "pv_02_voltage": [
        278.2,
        "V"
      ],
      "pv_02_current": [
        0.3,
        "A"
      ],
      "input_power": [
        156,
        "W"
      ],
      "day_active_power_peak": [
        3181,
        "W"
      ],
      "active_power": [
        1619,
        "W"
      ],
      "efficiency": [
        100.0,
        "%"
      ],
      "device_status": [
        "On-grid",
        null
      ],
      "startup_time": [
        "2025-05-10T05:29:52+00:00",
        null
      ],
      "shutdown_time": [
        null,
        null
      ],
      "accumulated_yield_energy": [
        1042.66,
        "kWh"
      ],
      "total_dc_input_power": [
        735.75,
        "kWh"
      ],
      "hourly_yield_energy": [
        0.41,
        "kWh"
      ],
      "daily_yield_energy": [
        5.64,
        "kWh"
      ]
    },
    "slave_1_power_meter_data": {
      "power_meter_active_power": [
        0,
        "W"
      ],
      "active_grid_power_factor": [
        1.0,
        null
      ],
      "active_grid_frequency": [
        50.0,
        "Hz"
      ],
      "grid_exported_energy": [
        433.39,
        "kWh"
      ],
      "grid_accumulated_energy": [
        960.39,
        "kWh"
      ],
      "active_grid_A_power": [
        0,
        "W"
      ],
      "active_grid_B_power": [
        0,
        "W"
      ],
      "active_grid_C_power": [
        0,
        "W"
      ]
    },
    "slave_1_battery_data": {
      "storage_state_of_capacity": [
        82.0,
        "%"
      ],
      "storage_running_status": [
        2,
        null
      ],
      "storage_bus_voltage": [
        611.7,
        "V"
      ],
      "storage_bus_current": [
        -2.4,
        "A"
      ],
      "storage_charge_discharge_power": [
        -1460,
        "W"
      ],
      "storage_total_charge": [
        652.94,
        "kWh"
      ],
      "storage_total_discharge": [
        635.47,
        "kWh"
      ],
      "storage_current_day_charge_capacity": [
        15.3,
        "kWh"
      ],
      "storage_current_day_discharge_capacity": [
        2.71,
        "kWh"
      ]
    }
  }
}
```

### Please confirm the following:

- [x] I'm running the latest release of Home Assistant.
- [x] I'm running the latest release of this integration.
- [x] I did not find an existing issue describing this problem.
- [x] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2025-05-24:
Your logs show an unstable connection and the inverter incorrectly answering to queries. No indication of a bug where the integration does something incorrect.

Are there other systems also trying to read from your solar installation? That is not supported.

closed 2025-05-24 by wlcrs

knothski2, 2025-05-24:
No only the Home Assistant, and it had worked fine until this.