wlcrs/huawei_solar #1030: [Feature Request]: How to set battery to "pause" mode?

MariuszWoszczynski, 2025-06-04:
### Describe your feature request

I would like to set the operation of the energy storage in such a way that in the morning, when the selling price of energy is higher, all the energy from the photovoltaic goes to the grid, and from the moment when the selling price is low, or even falls to zero (energy price entity or a specific hour), start charging the storage. How to do it? I do not see in the options such a mode of operation, so as not to charge the storage, but to give energy to the grid. If I turn on the mode "fully feed to grid" it discharges my storage, and I want to avoid this. Can you add this functionality?

### Proper usage

- [x] I confirm that this is not a bug report or support request
- [x] I confirm that this feature request is within the stated scope of the integration

jacekkjw, 2025-06-07:
It's already supported. There is dedicated mode "Time of use (TOU)" for that. You can set it from battery device and define "TOU charging and discharging periods" using a service (look https://github.com/wlcrs/huawei_solar/wiki/Time-of-Use-control for details). If you have EMMA type installation unfortunately service  is not working. You should use second metod, setting the "Maximum charging power" . Define it to zero and battery will stop to charge.  Set to initial value and battery will start charge again. Use it with "Working mode" set to "Maximum self consumption". 
It's work perfectly at my installation.

c0delama, 2025-06-10:
@jacekkjw Could you please give me a hint on how to change the maximum charging power?

I'm trying to set it using:
```
action: number.set_value
target:
  entity_id: number.batteries_maximum_charging_power
data:
  value: 1000
```

but it doesn't do anything. I've tried automations, the example is from the developer options and i've even used the input field that is offered from the integration. Nothing works however. The value itself is also reported as undefined, while its attributes make absolutely sense. Setting it from Fusion Solar also works fine, so the system must support it. I'm using EMMA, is that the problem? But why does the integration offer it then?

MariuszWoszczynski, 2025-06-10:
I used HA automation:

[image]

jacekkjw, 2025-06-10:
@c0delama Yes you are right it's EMMA related issue. It probably should be fixed at next release. Look at this: https://github.com/wlcrs/huawei_solar/issues/943#issuecomment-2918633926. 

Before the next version you can use the first method described by me. Define one discharging period for whole day and one charging period, 5 minutes long at night, using the fusion solar web or app. (service it's not working with EMMA). After that if you switch working mode to TOU battery will stop to charge. It really works. I used this method about 6 months.
Second solution, read carefully information about the bug. There is a simple workaround how to enable maximum charging with EMMA installations.

closed 2025-06-23 by wlcrs