wlcrs/huawei_solar #1037: [Bug]: SDongle with V200R022C10SPC300 firmware cannot add device with extended permissions

reboot1983, 2025-06-22:
### Describe the issue

I had to upgrade my SDongle firmware to the latest firmware because of instability issues. In HA the connection with my Huawei Sun2000-3.68KTL-L1 was more down than online :(

So, before the upgrade I was on V200R022C10SPC118, after the upgrade I'm at V200R022C10SPC300. The 300 firmware is not letting me add the integration with the advanced option checked. I get an error 'unexpected error'. Without the checkbox marked it works fine, but I cannot send command's to the Inverter. 

### Describe your Huawei Solar Setup

Inverter Type: Huawei Sun2000-3.68KTL-L1
Inverter Firmware version: V200R001C00SPC149
sDongle present: Yes 
sDongle Type: sDongleA-05 (WiFi / Ethernet)
sDongle Connectivitiy: WiFi
sDongle Firmware: V200R022C10SPC300
Power meter present: single phase
Optimizers Present: Yes
Battery: LUNA2000-SO 5 kWh
Battery Firmware version: V100R002C00SPC633 
Huawei Solar integration version: 1.5.4

### How do you connect to the inverter?

Via the SDongle, wireless connection

### Upload your Diagnostics File

[config_entry-huawei_solar-01JYCDD8MYRTV8WSBEK7ZAGD9Z.json](https://github.com/user-attachments/files/20854137/config_entry-huawei_solar-01JYCDD8MYRTV8WSBEK7ZAGD9Z.json)

### Upload your relevant debug logs

```text
2025-06-22 20:08:26.353 ERROR (MainThread) [pymodbus.logging] No response received after 3 retries, continue with next request
2025-06-22 20:08:26.353 ERROR (MainThread) [custom_components.huawei_solar.config_flow] Unexpected exception while connecting via ModbusTCP
Traceback (most recent call last):
  File "/config/custom_components/huawei_solar/config_flow.py", line 508, in async_step_setup_network
    info = await validate_network_setup_auto_slave_discovery(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<3 lines>...
    )
    ^
  File "/config/custom_components/huawei_solar/config_flow.py", line 100, in validate_network_setup_auto_slave_discovery
    bridge = await create_tcp_bridge(
             ^^^^^^^^^^^^^^^^^^^^^^^^
    ...<3 lines>...
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/bridge.py", line 659, in create_tcp_bridge
    return await _create(await AsyncHuaweiSolar.create(host, port, slave_id), slave_id)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/bridge.py", line 685, in _create
    product_info = await HuaweiSolarProductInfo.retrieve_from_device(client, slave_id)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/bridge.py", line 67, in retrieve_from_device
    ) = await client.get_multiple(
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<8 lines>...
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 292, in get_multiple
    response = await self._read_registers(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<3 lines>...
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 429, in _read_registers
    return await _do_read()
           ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[... 767 lines cut ...]
           ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 377, in _do_read
    response = await self._client.read_holding_registers(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<3 lines>...
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 155, in execute
    raise ModbusIOException(txt)
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] No response received after 3 retries, continue with next request
```

### Please confirm the following:

- [x] I'm running the latest release of Home Assistant.
- [x] I'm running the latest release of this integration.
- [x] I did not find an existing issue describing this problem.
- [x] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [x] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2025-06-23:
This issue seems to be caused by a connection failure or an unstable connection to the inverter. This integration already has already implemented a lot of logic to automatically recover from connection failures or unstable connections. The three most important ones are:
- automatic reconnects if the connection to the inverter failed;
- each Modbus request is retried for up to 6 times, with increasing waits between each attempt to allow the connection to recover;
- after 3 failed Modbus requests the connection is reset to fix hanging connections.

Please understand that I cannot help you solve your connection failure or unstable connection. Consider [an alternative connection method](https://github.com/wlcrs/huawei_solar/wiki/Connecting-to-the-inverter).
If you are convinced that there is no connectivity issue, then I invite you to debug and fix the problem yourself and submit a pull request to the underlying [huawei-solar Python library](https://gitlab.com/Emilv2/huawei-solar). Your input would be greatly appreciated.

As there is nothing actionable for me to do in this issue, I'm closing it. You are welcome to open a topic in [Discussions: Connecting to the inverter](https://github.com/wlcrs/huawei_solar/discussions/categories/connecting-to-the-inverter) to get community support for your problem.

closed 2025-06-23 by wlcrs

9joh, 2025-06-25:
I just add more inverter and had to use SDongle. Installer upgrade all FW after finish installed 2nd inverter (2 5kw inveters and 1 SDongle). 
Eversince after upgrade, my OpenWRT connect to SDongle ok (had to change port from 6607 to 502), but it will keep disconnected every once in a short while (like 15-20 mins except at night when it stay longer). 
I try to connect HA directly to SDongle (same IP that connect to my WIFI to update Fusion through internet with port 502; you can do that in the new FW), but behavior is pretty much the same. Not sure if it is disconnect when SDonger try to talk to Fusion or not. 

WIFI connection to both OpenWRT or directly to my Ruijie AP is pretty strong, it only  1.5 meters apart from each other; I have not pull CAT5 to my inverter yet.