wlcrs/huawei_solar #1052: [Feature Request]: Decode SOH calibration status and Working status sensors

Ivano62, 2025-07-13:
### Describe your feature request

I think it would be good if the numeric values were decoded with the corresponding strings like in FusionSolar

### Proper usage

- [x] I confirm that this is not a bug report or support request
- [x] I confirm that this feature request is within the stated scope of the integration

wlcrs, 2025-07-13:
This is not documented by Huawei in the documentation. Please provide the complete and correct mapping that can be used.

Ivano62, 2025-07-25:
Hi,
Do you think the request can be granted?
Thanks

Ivano62, 2025-08-25:
Is there any chance the request will be granted?
Thank you.

wlcrs, 2025-09-09:
As there is no way for me to know if those lists are complete, I will not implement it. The library would crash when an unexpected value is returned.

wlcrs, 2025-09-09:
Everything can be managed. But it is my choice to stick to what is officially documented by Huawei as much as possible.

You can manage it yourself by using a template entity in Home Assistant to transform those values yourself.