wlcrs/huawei_solar #1065: [Bug]: Integration fails after Home Assistant 2025.8 update

dherzog90, 2025-08-08:
### Describe the issue

The integration does not load after updating to HA 2025.8 - there seems to be an modbus error.

### Describe your Huawei Solar Setup

Inverter Type: Huawei SUN2000 10KTL M1
Inverter Firmware version: V100R001C00SPC172
sDongle present: Yes
sDongle Type: sDongleA-05 (Ethernet)
sDongle Connectivitiy: Ethernet
sDongle Firmware: V200R022C10SPC300
Power meter present: three phase
Optimizers Present: No
Battery: None
Battery Firmware version: 
Huawei Solar integration version: 1.5.5

### How do you connect to the inverter?

Via the SDongle, wired connection

### Upload your Diagnostics File

Cant download the diagnostics because the integration is not loading

### Upload your relevant debug logs

```text
Logs from HA: 
Logger: homeassistant.config_entries
Source: config_entries.py:749
First occurred: 09:22:00 (2 occurrences)
Last logged: 09:30:16

Error setting up entry SUN2000-10KTL-M101074311-0 for huawei_solar
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 749, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/huawei_solar/__init__.py", line 96, in async_setup_entry
    primary_bridge = await create_tcp_bridge(
                     ^^^^^^^^^^^^^^^^^^^^^^^^
    ...<3 lines>...
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/bridge.py", line 659, in create_tcp_bridge
    return await _create(await AsyncHuaweiSolar.create(host, port, slave_id), slave_id)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/bridge.py", line 685, in _create
    product_info = await HuaweiSolarProductInfo.retrieve_from_device(client, slave_id)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/bridge.py", line 67, in retrieve_from_device
    ) = await client.get_multiple(
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<8 lines>...
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 292, in get_multiple
    response = await self._read_registers(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<3 lines>...
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 429, in _read_registers
    return await _do_read()
           ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 377, in _do_read
    response = await self._client.read_holding_registers(
                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
        register,
        ^^^^^^^^^
        count=length,
        ^^^^^^^^^^^^^
        slave=slave or self.slave_id,
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
TypeError: ModbusClientMixin.read_holding_registers() got an unexpected keyword argument 'slave'
```

### Please confirm the following:

- [x] I'm running the latest release of Home Assistant.
- [x] I'm running the latest release of this integration.
- [x] I did not find an existing issue describing this problem.
- [x] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [x] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2025-08-08:
What other integrations do you use Modbus under the hood?

jagoman81, 2025-08-09:
I have the same problem.
Do we have to do something with our HA installation to solve the situation, or just wait for integration update or HA new release.

Thank you?

jagoman81, 2025-08-09:
Downgraded to HA Core 2025.7.3 and Integration runs well again. So problem is in Core 2025.8.0

rosenhei64-byte, 2025-08-10:
I'm experiencing the same problem on my installation.
I also have the stiebel_eltron_isg_component installed.
After downgrading to HA 2025.7.3, huawei_solar started working again, but the stiebel_eltron_isg component stopped with a similar error.

dherzog90, 2025-08-10:
I did some "hack" to fix the issue temporarily

- Local copy of https://gitlab.com/Emilv2/huawei-solar/-/tree/master/src/huawei_solar?ref_type=heads
- New folder in the integration "lib/pyhuaweisolar" where i copied the source code above
- Update the import statements in every python file from the integration (before updating i created a backup)
- In the manifest json file i removed the reference to "huawei_solar" python package

[image]

The integration is now running aigan

sky-future, 2025-08-11:
> I did some "hack" to fix the issue temporarily
> 
> * Local copy of https://gitlab.com/Emilv2/huawei-solar/-/tree/master/src/huawei_solar?ref_type=heads
> * New folder in the integration "lib/pyhuaweisolar" where i copied the source code above
> * renamed slave to device_id
> * Update the import statements in every python file from the integration (before updating i created a backup)
> * In the manifest json file i removed the reference to "huawei_solar" python package
> 
> <img alt="Image" width="373" height="1488" src="https://private-user-images.githubusercontent.com/1121857/476372785-3cacc6cd-e1a1-45fb-bd09-2bf7ea9b599d.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTQ5MTk2OTIsIm5iZiI6MTc1NDkxOTM5MiwicGF0aCI6Ii8xMTIxODU3LzQ3NjM3Mjc4NS0zY2FjYzZjZC1lMWExLTQ1ZmItYmQwOS0yYmY3ZWE5YjU5OWQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDgxMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA4MTFUMTMzNjMyWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NDUzMjA2NDY4OTkwZDY5OWZlOTVjZDQyNjZmZDJkNzEzYzY5OTQxNzNmY2E5ZTJjOTFiZDcyMTM0ZjhhODUxZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.Lg_MN6BCbBaqEXnYwd9Bt5JdTnc080o74E-963a8scU">
> The integration is now running again on HA 2025.8

Hi, can you be a little bit more clear on the procedure ? Thanks

osigge, 2025-08-11:
> i am using this integration which uses modbus for communication with heat pumps: https://github.com/pail23/stiebel_eltron_isg_component

The stiebel integration already released a fixed version. https://github.com/pail23/stiebel_eltron_isg_component/releases/tag/2025.8

razvan1despa, 2025-08-16:
Guys,

This morning i have updated to HA CORE 2025.8.1 and run into the HUAWEI Issues - similar error. I ahve removed the Integration and re-add - error is the same. I have just noticed that 2025.8.2 is available. I have updated and restart HA. After the update, re-add Huawei integration worked just fine imediatelly.

LotharWoman, 2025-08-20:
Hello, i have since 3 days new 1.5.6 on HA 2025.8.2. it runs 3 days perfect, but now i have lost connection and after this failure in log:
Logger: pymodbus.logging
Quelle: runner.py:156
Erstmals aufgetreten: 17. August 2025 um 19:19:50 (355 Vorkommnisse)
Zuletzt protokolliert: 19:14:26

ERROR: expected transaction 22708 but got 22707, IGNORING.
ERROR: expected transaction 22709 but got 22708, IGNORING.
ERROR: expected id 1 but got 16, IGNORING.
ERROR: expected transaction 22712 but got 22710, IGNORING.
ERROR: expected id 16 but got 1, IGNORING.

valimaties, 2025-08-20:
@LotharWoman I think is better to open a new issue for these errors, as I see there is nothing to do with the title of this issue here. 
I think there is something else.

thchin94, 2025-08-21:
> [1.5.6](https://github.com/wlcrs/huawei_solar/releases/tag/1.5.6) version is working only with HA 2025.8.2 (or greater). If anyone uses older version of HA core, then install an older version of the integration (1.5.5 or smaller). Don't install 1.5.6 on HA **< 2025.8.2**

This fixed the issue. Update HA core to 2025.8.2 and integration reloaded and healthy. Thanks!

Also thanks wlcrs for the ongoing patches and updates on the integration.

wlcrs, 2025-08-29:
For those running the 2025.9.b0 beta version (@mnimtz, @cmatte):

I've done a soft-release of 1.5.7, meaning that it's available in HACS if you select it manually (Use the 'redownload' function), but is not automatically suggested as an available upgrade.

cmatte, 2025-08-29:
@wlcrs it won't let it be downloaded even manually as it states it requires 2025.9.0 or newer, guessing HACS doesn't recognize 0b0 👀

[image]

wlcrs, 2025-08-29:
Not what I expected to happen... I've created a beta release 1.5.7 beta 1 which marks 2025.9.b0 as the minimum required HA. Does that work for you?

cmatte, 2025-08-30:
Attaching the log for the connection. Thanks in advance!

[home-assistant_2025-08-30T19-43-29.141Z.log](https://github.com/user-attachments/files/22060873/home-assistant_2025-08-30T19-43-29.141Z.log)

cmatte, 2025-08-30:
Attaching the log for the connection attempt failure, now cleaned-up from other noise :) Thanks in advance!

[home-assistant_2025-08-30T19-43-29.141Z.log](https://github.com/user-attachments/files/22060881/home-assistant_2025-08-30T19-43-29.141Z.log)