wlcrs/huawei_solar #1072: [Bug]: Error to load package in HA 2025.08.2 / 16.1

santanar00, 2025-08-15:
### Describe the issue

Hi guys!

After I updated the Home Assistant system to version 2025.08.2 and the operating system to version 16.1, I can no longer load the integration.

**Inverter model:** SIW300H M060 L1

**HA versions:**
Core 2025.8.2
Supervisor 2025.08.1
Operating System 16.1
Front-end 20250811.0

**Error log**
Registrador: homeassistant.util.package
Fonte: util/package.py:159
Ocorreu pela primeira vez: 16:03:04 (3 ocorrências )
Último registro: 16:03:06

Unable to install package huawei-solar==2.4.4: × No solution found when resolving dependencies: ╰─▶ Because huawei-solar==2.4.4 depends on pymodbus>=3.8.2,<3.8.5 and pymodbus==3.9.2, we can conclude that huawei-solar==2.4.4 cannot be used. And because you require huawei-solar==2.4.4, we can conclude that your requirements are unsatisfiable.

I would be very grateful if you could include a fix for the next version of the integration. Thank you very much.

### Describe your Huawei Solar Setup

Inverter Type: SIW300H M060 L1
Inverter Firmware version: V200R001C00SPC130
Huawei Solar integration version: 1.5.5

### How do you connect to the inverter?

Via the SDongle, wireless connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
Unable to install package huawei-solar==2.4.4: × No solution found when resolving dependencies: ╰─▶ Because huawei-solar==2.4.4 depends on pymodbus>=3.8.2,<3.8.5 and pymodbus==3.9.2, we can conclude that huawei-solar==2.4.4 cannot be used. And because you require huawei-solar==2.4.4, we can conclude that your requirements are unsatisfiable.
```

### Please confirm the following:

- [x] I'm running the latest release of Home Assistant.
- [x] I'm running the latest release of this integration.
- [x] I did not find an existing issue describing this problem.
- [x] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [x] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

Gadeo, 2025-08-15:
Yes, here also.

codetrembler, 2025-08-15:
Same Problem here.
There were changes for core 8.2 with the version of pymodbus.
Found this: https://github.com/redpomodoro/fronius_modbus/issues/51

Please take a look at it.
After a rollback to 8.1 everything is back online.

Keep up the good work!
Thank you.