wlcrs/huawei_solar #1073: [Feature Request]: Provide a register in the inverter that signals the current backup mode.

Bjoern112, 2025-08-16:
### Describe your feature request

**Background**

The inverter communicates with the Backup Box during backup operation.

For monitoring and automations it would be very helpful if this state were available via a Modbus register.

**Limitation**

Querying such a register is only possible as long as EMMA is reachable.
Normally EMMA is powered from the grid – in case of a grid outage it would go offline and Modbus TCP would no longer be available.
Only if EMMA is powered via the 12 V input with an external power supply that is connected to a UPS (which itself is connected to the backup power circuit), EMMA remains online during backup mode and can provide the register data via Modbus TCP.

**Proposal**

Check if there is already an internal register in the inverter that signals the backup status (active/inactive).

If available: add it to the [Wiki register list](https://github.com/wlcrs/huawei_solar/wiki).

Home Assistant integration: optionally expose an entity such as “Backup Mode Active”.

**Benefits**

Reliable detection of grid/backup switching.

Enables automations and alerts.

Better transparency for operators.

### Proper usage

- [x] I confirm that this is not a bug report or support request
- [x] I confirm that this feature request is within the stated scope of the integration

wlcrs, 2025-08-17:
I don't appreciate this LLM-generated gibberish. Do you understand yourself what you are asking?

Read the PDFs provided by Huawei to understand what is possible, or heck: even ask an AI to do it for you. But what you did now is just wasting everyone's time.

closed 2025-08-17 by wlcrs

Bjoern112, 2025-08-17:
Thanks for your feedback.
My question is simply about the registers for the Off-Grid / Backup Mode of the SUN2000.
I’m not sure where Huawei provides the correct documentation, that’s why I asked here.