wlcrs/huawei_solar #1087: [Bug]: Addone doesn't start after update to version 1.5.7

TrackerTheCode, 2025-08-29:
### Describe the issue

I got today a update notification in my HA. I started the update procedure of my Huawei Solar addon.
After the restart the addon doesn't start anymore i I got some error messages:

[image]

I checked the installed version and it cofuses me that the version 1.6.7 ist installed:

[image]

latest available release is displayed currently 1.5.6 
Whats went wrong and what can I do, to get working again?

### Describe your Huawei Solar Setup

Inverter Type: 
Inverter Firmware version: 
sDongle present: Yes / No
sDongle Type: sDongleA-05 (WiFi / Ethernet) / SDongleA-03 (4G)
sDongle Connectivitiy: WiFi / Ethernet / 4G
sDongle Firmware: 
Power meter present: three phase / single phase / no
Optimizers Present: Yes / No
Battery: LUNA2000-SO xxkWh / LUNA2000-S1 xxkWh (Released 2024) / LG RESU xxkWh / None
Battery Firmware version: 
Huawei Solar integration version: 

### How do you connect to the inverter?

Please select your connection method

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
[33m2025-08-29 18:03:34.112 WARNING (ImportExecutor_0) [py.warnings] /usr/local/lib/python3.13/site-packages/google/__init__.py:2: UserWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html. The pkg_resources package is slated for removal as early as 2025-11-30. Refrain from using this package or pin to Setuptools<81.
  __import__('pkg_resources').declare_namespace(__name__)
[0m
[31m2025-08-29 18:04:24.961 ERROR (MainThread) [aiohttp.server] Error handling request from 2a00:6020:4ea6:7200:5194:e9a9:7c88:267b
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_protocol.py", line 510, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_app.py", line 569, in _handle
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 92, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 83, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 26, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 86, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 242, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 41, in headers_middleware
    response = await handler(request)
[... 132 lines cut ...]
    return await manager.async_get_integration_with_requirements(domain)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/requirements.py", line 170, in async_get_integration_with_requirements
    await future
  File "/usr/src/homeassistant/homeassistant/requirements.py", line 159, in async_get_integration_with_requirements
    await self._async_process_integration(integration, done)
  File "/usr/src/homeassistant/homeassistant/requirements.py", line 182, in _async_process_integration
    await self.async_process_requirements(
        integration.domain, integration.requirements
    )
  File "/usr/src/homeassistant/homeassistant/requirements.py", line 264, in async_process_requirements
    self._raise_for_failed_requirements(name, missing)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/requirements.py", line 292, in _raise_for_failed_requirements
    raise RequirementsNotFound(integration, [req])
homeassistant.requirements.RequirementsNotFound: Requirements for huawei_solar not found: ['huawei-solar==2.4.6'].[0m
```

### Please confirm the following:

- [x] I'm running the latest release of Home Assistant.
- [x] I'm running the latest release of this integration.
- [x] I did not find an existing issue describing this problem.
- [x] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [x] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

Blackmannl, 2025-08-29:
did you read this? "This release is INCOMPATIBLE with HA 2025.08 and earlier This release is REQUIRED for HA 2025.09 and later"

c-mholm, 2025-08-29:
I have Core 2025.8.3 and it broke my config too. I reverted back to 1.5.6 and all is good again. Going to wait with upgrading until it stops working.

TrackerTheCode, 2025-08-29:
> did you read this? "This release is INCOMPATIBLE with HA 2025.08 and earlier This release is REQUIRED for HA 2025.09 and later"

I noticed it aftwards.... and that was to late. The interessting question for me is (I just waering the HA user glasses), why this version was promoted to me as a new update in the regular settings section from HA? This is version is not compatible and was presented as the new version ... this is misleading and ends in not working system.... for me a bad user experience

closed 2025-08-29 by TrackerTheCode