wlcrs/huawei_solar #1088: [Bug]: Batter configuration is not active

bkuehnel795, 2025-08-30:
### Describe the issue

Integration works, only Batterie configuration does not show any active entries:

[image]

Integration is configured w/ enhanced rights, but there is no user or password request during the configuration requested. Configuration reports sucessful installation. 

### Describe your Huawei Solar Setup

Inverter Type: SUN2000-17K-MB0
Inverter Firmware version: V200R023C10SPC209

sDongle present: No
EMMA A02
EMMA Firmware: SmartHEMS V100R024C10SPC100
Power meter present: three phase / EMMA A02

Optimizers Present: No

Battery: LUNA2000-S1 20kWh 
Battery Firmware version:  V200R023C10SPC110
Huawei Solar integration version: 

### How do you connect to the inverter?

Via the `SUN2000-<serial_no>` WiFi

### Upload your Diagnostics File

[config_entry-huawei_solar-01JVHTQ80YMSKN2KSRZW9RKS2E.json](https://github.com/user-attachments/files/22056882/config_entry-huawei_solar-01JVHTQ80YMSKN2KSRZW9RKS2E.json)

### Upload your relevant debug logs

```text
integration 'huawei_solar' at custom_components/huawei_solar/diagnostics.py, line 31: "pymodbus_version": version("pymodbus"), (offender: /usr/local/lib/python3.13/site-packages/importlib_metadata/__init__.py, line 817: return os.listdir(self.root or '.')), please create a bug report at https://github.com/wlcrs/huawei_solar/issues
2025-08-30 08:52:43.398 WARNING (MainThread) [custom_components.huawei_solar.config_flow] Skipping slave 2 with model SmartGuard-63A-T0. Only SUN2000 inverters are supported as secondary slaves
2025-08-30 08:52:43.878 WARNING (MainThread) [custom_components.huawei_solar.config_flow] Skipping slave 8 with model SmartLoad. Only SUN2000 inverters are supported as secondary slaves
2025-08-30 08:52:44.775 ERROR (MainThread) [homeassistant.components.sensor] Platform huawei_solar does not generate unique IDs. ID NS2431424900_storage_unit_1_battery_pack_3_working_status already exists - ignoring sensor.batterie_1_pack_3_arbeitsstatus
2025-08-30 08:52:55.228 ERROR (MainThread) [custom_components.huawei_solar] Error fetching NS2431424900_config_data_update_coordinator data: Could not update NS2431424900 values: Got error while reading from register 47415 with length 4: ExceptionResponse(dev_id=3, function_code=131, exception_code=3)
```

### Please confirm the following:

- [x] I'm running the latest release of Home Assistant.
- [x] I'm running the latest release of this integration.
- [x] I did not find an existing issue describing this problem.
- [x] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

d3zit, 2025-09-02:
Same issue here

vwtuner, 2025-09-09:
[image]

Don’t work any think in 1.6.0b2

bkuehnel795, 2025-09-09:
One question, this is a pull only connection?
If e.g. the Backup decharge level is set in HA, Now change takes place in FusionSolar.

wlcrs, 2025-09-09:
@vwtuner I fixed the error in the de.json

Please note that screenshots like the one that you're posting are not very helpful. I need proper debug logs if you want me to diagnose the root cause of your issue. I've documented how to get those debug logs in the 'bug report' issue template.

vwtuner, 2025-09-09:
[image]

Thanks it works great 

You must new configuration!