wlcrs/huawei_solar #1089: [Bug]: 500 Internal Server Error Server got itself in trouble

tonimg87, 2025-08-30:
### Describe the issue

HA version 2025.8.3

This is in the logs of HA, last versión in both:

Unable to install package huawei-solar==2.4.6: × No solution found when resolving dependencies: ╰─▶ Because huawei-solar==2.4.6 depends on pymodbus>=3.11.0,<4 and pymodbus==3.9.2, we can conclude that huawei-solar==2.4.6 cannot be used. And because you require huawei-solar==2.4.6, we can conclude that your requirements are unsatisfiable.

The connection stopped working when updating the Huawei Solar version. When deleting it and reinstalling the latest version from scratch, this appears in the log, and when "adding integration" it gives this error:
[Bug]: 500 Internal Server Error Server got itself in trouble

### Describe your Huawei Solar Setup

Inverter Type: huawei 4000
Inverter Firmware version: 
sDongle present:  No
sDongle Type:
sDongle Connectivitiy: 
sDongle Firmware: 
Power meter present: ingle phase
Optimizers Present: No
Battery: None
Battery Firmware version: 
Huawei Solar integration version: 2.4.6

### How do you connect to the inverter?

Please select your connection method

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
Unable to install package huawei-solar==2.4.6: × No solution found when resolving dependencies: ╰─▶ Because huawei-solar==2.4.6 depends on pymodbus>=3.11.0,<4 and pymodbus==3.9.2, we can conclude that huawei-solar==2.4.6 cannot be used. And because you require huawei-solar==2.4.6, we can conclude that your requirements are unsatisfiable.
```

### Please confirm the following:

- [x] I'm running the latest release of Home Assistant.
- [x] I'm running the latest release of this integration.
- [x] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [x] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

faPU1909, 2025-08-30:
Same problem in my side :\

algopt, 2025-08-30:
Same issue, rolled back and it works again.

tmycgn, 2025-09-02:
Update: I no longer have the error with running the latest HA and integration version. I think the error disappeared after I completely! restarted HA OS (inkl. all Addons/Docker Container). Perhaps someone else with the same server error could test this too and confirm it for the others.