wlcrs/huawei_solar #1091: [Bug]: Updated to day to 1.5.7 and stopped working

lfloureiro, 2025-09-03:
### Describe the issue

After updating to new version the integration is not coming up anymore.

"O fluxo de configuração não pôde ser carregado: 500 Internal Server Error Server got itself in trouble"

basicly means it cannot load the configuraiton flux

[image]

### Describe your Huawei Solar Setup

Inverter Type: 
Inverter Firmware version: 
sDongle present: Yes / No
sDongle Type: sDongleA-05 (WiFi / Ethernet) / SDongleA-03 (4G)
sDongle Connectivitiy: WiFi / Ethernet / 4G
sDongle Firmware: 
Power meter present: three phase / single phase / no
Optimizers Present: Yes / No
Battery: LUNA2000-SO xxkWh / LUNA2000-S1 xxkWh (Released 2024) / LG RESU xxkWh / None
Battery Firmware version: 
Huawei Solar integration version: 

### How do you connect to the inverter?

Please select your connection method

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
Registo: homeassistant.util.package
Origem: util/package.py:159
Primeira ocorrência: 11:14:59 (3 ocorrências)
Último registo: 11:14:59

Unable to install package huawei-solar==2.4.6: × No solution found when resolving dependencies: ╰─▶ Because huawei-solar==2.4.6 depends on pymodbus>=3.11.0,<4 and pymodbus==3.9.2, we can conclude that huawei-solar==2.4.6 cannot be used. And because you require huawei-solar==2.4.6, we can conclude that your requirements are unsatisfiable.

Registo: homeassistant.setup
Origem: setup.py:278
Primeira ocorrência: 11:14:59 (1 ocorrência)
Último registo: 11:14:59

Setup failed for custom integration 'huawei_solar': Requirements for huawei_solar not found: ['huawei-solar==2.4.6'].
```

### Please confirm the following:

- [ ] I'm running the latest release of Home Assistant.
- [ ] I'm running the latest release of this integration.
- [ ] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

janstolte, 2025-09-03:
You should not use this integration on HA versions below 2025.8. Since 2025.8.3 is the latest GA version, you should roll back to using 1.5.6 of Huawei Solar.
Huawei Solar 1.5.7 requires 2025.9 and vice versa. Source: https://github.com/wlcrs/huawei_solar/releases/tag/1.5.7

I had the same issue you described and successfully rolled back, maintaining all functionality.

ogghi, 2025-09-03:
> EKENGOESWILD left a comment (wlcrs/huawei_solar#1091)
> [https://github.com/wlcrs/huawei_solar/issues/1091#issuecomment-3249733692]
>
> How to do a rollback in HACS? :)
>
> Reply to this email directly, view it on GitHub
> [https://github.com/wlcrs/huawei_solar/issues/1091#issuecomment-3249733692],
> or unsubscribe

Go to HACS in the menu. Then select huawei addon. 
3 dots menu right upper corner, download again.
Here select the desired version.
Restart HA afterwards

Cheers

Gesendet mit dem Nachrichtendienst von Infomaniak [https://ik.me], The
Ethical Cloud.