wlcrs/huawei_solar #1099: [Bug]: cannot connect to Inveter

smoki3, 2025-09-07:
### Describe the issue

As the other report is closed, here a new one with logs.

Since 1.5.7 I can not add my inverter any more. The EMMA itself is working with slave Id 0 but no other devices can be connected 

### Describe your Huawei Solar Setup

Inverter Type: SUN2000 6KTL M1
Inverter Firmware version: 172
sDongle present: EMMA

sDongle Connectivitiy: WiFi 
sDongle Firmware: 102
Power meter present: three phase / single phase / no
Optimizers Present: Yes 
Battery: LUNA2000-SO 10kW
Battery Firmware version: 336
Huawei Solar integration version: 1.5.7

### How do you connect to the inverter?

Via the SDongle, wireless connection

### Upload your Diagnostics File

[config_entry-huawei_solar-01K4GC8KWK4TDVA0JPESY0TZE7.json](https://github.com/user-attachments/files/22198276/config_entry-huawei_solar-01K4GC8KWK4TDVA0JPESY0TZE7.json)
Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
2025-09-07 21:23:28.696 DEBUG (MainThread) [pymodbus.logging] Connecting to 192.168.10.50:502.
2025-09-07 21:23:28.697 DEBUG (MainThread) [pymodbus.logging] Connecting comm
2025-09-07 21:23:28.700 DEBUG (MainThread) [pymodbus.logging] Connected to comm
2025-09-07 21:23:28.702 DEBUG (MainThread) [huawei_solar.modbus] Waiting for 1500 milliseconds after connection before performing operations
2025-09-07 21:23:28.806 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 30000 with length 65 from server 0
2025-09-07 21:23:28.806 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x1 0x0 0x0 0x0 0x6 0x0 0x3 0x75 0x30 0x0 0x41
2025-09-07 21:23:28.826 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x1 0x0 0x0 0x0 0x85 0x0 0x3 0x82 0x53 0x6d 0x61 0x72 0x74 0x48 0x45 0x4d 0x53 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x4e 0x53 0x32 0x35 0x36 0x31 0x31 0x32 0x30 0x39 0x37 0x34 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x53 0x6d 0x61 0x72 0x74 0x48 0x45 0x4d 0x53 0x20 0x56 0x31 0x30 0x30 0x52 0x30 0x32 0x35 0x43 0x30 0x30 0x53 0x50 0x43 0x31 0x30 0x32 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 extra data: 
2025-09-07 21:23:28.826 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x1 0x0 0x0 0x0 0x85 0x0 0x3 0x82 0x53 0x6d 0x61 0x72 0x74 0x48 0x45 0x4d 0x53 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x4e 0x53 0x32 0x35 0x36 0x31 0x31 0x32 0x30 0x39 0x37 0x34 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x53 0x6d 0x61 0x72 0x74 0x48 0x45 0x4d 0x53 0x20 0x56 0x31 0x30 0x30 0x52 0x30 0x32 0x35 0x43 0x30 0x30 0x53 0x50 0x43 0x31 0x30 0x32 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2025-09-07 21:23:28.827 DEBUG (MainThread) [pymodbus.logging] decoded PDU function_code(3 sub -1) -> ReadHoldingRegistersResponse(dev_id=0, transaction_id=0, address=0, count=0, bits=[], registers=[21357, 24946, 29768, 17741, 21248, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 20051, 12853, 13873, 12594, 12345, 14132, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 21357, 24946, 29768, 17741, 21280, 22065, 12336, 21040, 12853, 17200, 12371, 20547, 12592, 12800, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], status=1retries=0) 
2025-09-07 21:23:28.829 DEBUG (MainThread) [huawei_solar.huawei_solar] Sleeping for 0.048477 seconds before making next call
2025-09-07 21:23:28.880 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 30222 with length 20 from server 0
2025-09-07 21:23:28.881 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x2 0x0 0x0 0x0 0x6 0x0 0x3 0x76 0xe 0x0 0x14
2025-09-07 21:23:28.900 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x2 0x0 0x0 0x0 0x2b 0x0 0x3 0x28 0x45 0x4d 0x4d 0x41 0x2d 0x41 0x30 0x32 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 extra data: 
2025-09-07 21:23:28.901 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x2 0x0 0x0 0x0 0x2b 0x0 0x3 0x28 0x45 0x4d 0x4d 0x41 0x2d 0x41 0x30 0x32 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2025-09-07 21:23:28.901 DEBUG (MainThread) [pymodbus.logging] decoded PDU function_code(3 sub -1) -> ReadHoldingRegistersResponse(dev_id=0, transaction_id=0, address=0, count=0, bits=[], registers=[17741, 19777, 11585, 12338, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], status=1retries=0) 
2025-09-07 21:23:28.902 INFO (MainThread) [custom_components.huawei_solar.config_flow] Successfully connected to inverter SmartHEMS with SN NS2561120974
2025-09-07 21:23:28.904 DEBUG (MainThread) [huawei_solar.huawei_solar] Sleeping for 0.047565 seconds before making next call
2025-09-07 21:23:28.953 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 30000 with length 65 from server 2
2025-09-07 21:23:28.954 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x3 0x0 0x0 0x0 0x6 0x2 0x3 0x75 0x30 0x0 0x41
2025-09-07 21:23:28.969 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x3 0x0 0x0 0x0 0x85 0x2 0x3 0x82 0x53 0x55 0x4e 0x32 0x30 0x30 0x30 0x2d 0x36 0x4b 0x54 0x4c 0x2d 0x4d 0x31 0x0 0x30 0x31 0x30 0x37 0x34 0x33 0x31 0x35 0x2d 0x30 0x30 0x36 0x0 0x0 0x54 0x41 0x32 0x33 0x33 0x30 0x34 0x31 0x36 0x32 0x38 0x39 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x31 0x30 0x37 0x34 0x33 0x31 0x35 0x2d 0x30 0x30 0x36 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x56 0x31 0x30 0x30 0x52 0x30 0x30 0x31 0x2d 0x30 0x32 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x56 0x31 0x30 0x30 0x52 0x30 0x30 0x31 0x43 0x30 0x30 0x53 0x50 0x43 0x31 0x37 0x32 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 extra data: 
2025-09-07 21:23:28.970 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x3 0x0 0x0 0x0 0x85 0x2 0x3 0x82 0x53 0x55 0x4e 0x32 0x30 0x30 0x30 0x2d 0x36 0x4b 0x54 0x4c 0x2d 0x4d 0x31 0x0 0x30 0x31 0x30 0x37 0x34 0x33 0x31 0x35 0x2d 0x30 0x30 0x36 0x0 0x0 0x54 0x41 0x32 0x33 0x33 0x30 0x34 0x31 0x36 0x32 0x38 0x39 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x31 0x30 0x37 0x34 0x33 0x31 0x35 0x2d 0x30 0x30 0x36 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x56 0x31 0x30 0x30 0x52 0x30 0x30 0x31 0x2d 0x30 0x32 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x56 0x31 0x30 0x30 0x52 0x30 0x30 0x31 0x43 0x30 0x30 0x53 0x50 0x43 0x31 0x37 0x32 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2025-09-07 21:23:28.971 DEBUG (MainThread) [pymodbus.logging] decoded PDU function_code(3 sub -1) -> ReadHoldingRegistersResponse(dev_id=0, transaction_id=0, address=0, count=0, bits=[], registers=[21333, 20018, 12336, 12333, 13899, 21580, 11597, 12544, 12337, 12343, 13363, 12597, 11568, 12342, 0, 21569, 12851, 13104, 13361, 13874, 14393, 0, 0, 0, 0, 12337, 12343, 13363, 12597, 11568, 12342, 0, 0, 0, 0, 22065, 12336, 21040, 12337, 11568, 12800, 0, 0, 0, 0, 0, 0, 0, 0, 0, 22065, 12336, 21040, 12337, 17200, 12371, 20547, 12599, 12800, 0, 0, 0, 0, 0, 0], status=1retries=0) 
2025-09-07 21:23:28.972 DEBUG (MainThread) [huawei_solar.huawei_solar] Sleeping for 0.048735 seconds before making next call
2025-09-07 21:23:29.022 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 30071 with length 1 from server 2
2025-09-07 21:23:29.023 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x4 0x0 0x0 0x0 0x6 0x2 0x3 0x75 0x77 0x0 0x1
2025-09-07 21:23:29.039 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x4 0x0 0x0 0x0 0x5 0x2 0x3 0x2 0x0 0x2 extra data: 
2025-09-07 21:23:29.039 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x4 0x0 0x0 0x0 0x5 0x2 0x3 0x2 0x0 0x2
2025-09-07 21:23:29.040 DEBUG (MainThread) [pymodbus.logging] decoded PDU function_code(3 sub -1) -> ReadHoldingRegistersResponse(dev_id=0, transaction_id=0, address=0, count=0, bits=[], registers=[2], status=1retries=0) 
2025-09-07 21:23:29.041 DEBUG (MainThread) [huawei_solar.huawei_solar] Sleeping for 0.048972 seconds before making next call
2025-09-07 21:23:29.091 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37200 with length 1 from server 2
2025-09-07 21:23:29.091 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x5 0x0 0x0 0x0 0x6 0x2 0x3 0x91 0x50 0x0 0x1
[... cut ...]
2025-09-07 21:25:36.717 DEBUG (MainThread) [custom_components.huawei_solar.number] No battery detected on slave 0. Skipping energy storage number entities
```

### Please confirm the following:

- [x] I'm running the latest release of Home Assistant.
- [x] I'm running the latest release of this integration.
- [x] I did not find an existing issue describing this problem.
- [x] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [x] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

smoki3, 2025-09-07:
> You configuration mentions only slave 0, which is the EMMA. Other devices have other slave IDs, which you need to configure properly during setup.

As soon I try to configure the slave ID 2 (Inveter) I run into this error which you can see in the logs. So the config flow can not finished. That the reason why you don't see it in the json

wlcrs, 2025-09-07:
Those logs only show that there is no reaction from slave 2. This is a configuration issue, not a bug.

I cannot assist you any further.

closed 2025-09-09 by wlcrs