wlcrs/huawei_solar #1116: [Feature Request]: Allow to control "Active power control mode" in EMMA

JensenNick, 2025-09-13:
### Describe your feature request

After upgrading my SDongle to EMMA I'm having issues controlling the "Active power control mode" when trying to prevent exporting to grid (when price is negative). In Fusion Solar this setting is now done under the parameter settings for EMMA, but I don't see this option under EMMA in Home Assistant - here it is still a "general option" to which you have to assign to an inverter. Doing it this way does not take effekt in EMMA / FusionSolar. So it seems like the setting in EMMA is "overruling" this attempt to change this settings.

In the Modbus Definitions for EMMA (page 16) there is the "Limited feed-in" with the appropriate modes for "Active power control". Can this be activated / made accessible in the integration?

[image]

I've had to disable the automatic function "Negative Tariff Optimisation" in EMMA because this setting does not work with my Charger - so looking to control this via this integration.

### Proper usage

- [x] I confirm that this is not a bug report or support request
- [x] I confirm that this feature request is within the stated scope of the integration

wlcrs, 2025-09-27:
Please test with the develop-branch to validate if my implementation works as expected.

JensenNick, 2025-10-03:
> Please test with the develop-branch to validate if my implementation works as expected.

Will do, and report back.

commit 4abfc8b1 (2025-09-27): Fix Active power control mode services when EMMA is present

Fixes #1116